merbanan/rtl_433 #2524: M5stack

Koekebeest, 2023-06-06:
Hello,

I wonder if this can be used on a M5stack core v2 with a 433mhz receiver/transmitter?

zuckschwerdt, 2023-06-06:
A better fit could be some Flipper-Zero software. But the Arduino port of rtl_433 should support some simpler transceivers.

Koekebeest, 2023-06-07:
Hm, interesting. Thank you for your time and information! :)

Koekebeest, 2023-06-13:
> A better fit could be some Flipper-Zero software. But the Arduino port of rtl_433 should support some simpler transceivers.

Which software are you referring to if I may ask? Will that work on the m5core2?

gdt, 2023-09-26:
This was a question and has run its course; closing to reduce issue clutter.  Please feel free to open a discussion or ask on the mailinglist.

closed 2023-09-26 by gdt