merbanan/rtl_433 #2527: RSSI seems screwy

buteomont, 2023-06-08:
I have three [Mighty Mule FM231 Driveway Alerts](https://www.mightymulestore.com/Mighty-Mule-Driveway-Alarm-p/fm231.htm) located at various locations on my (long) driveway.  I receive two of them on a Raspberry Pi 4 located at the house, and the third one on a Raspberry Pi 3B+ located at the far end of the driveway. The gate at the end of the driveway also has a (modified) FM341 receiver that also picks up the signal from the sensor and activates the gate opener. I feed the sensor data from both RTL_433 programs to Home Assistant via MQTT.

There's two issues.  First, the two sensors picked up by the RTL_433 at the house _always_ have the exact same RSSI value. The values change dramatically (which is also a strange thing, see the second issue), but the screwy part is that that they are always the exact same value for these two sensors even though they are in different directions and at different distances with respect to the RTL receiver.  The match is much too perfect for it to be coincidence. The graph below shows 3 days worth of signal for **both** sensors, and you can see that it looks like just one sensor because the two are exactly congruent with one another.
[image]

Why would these two sensors show up on RTL_433 with the exact same RSSI?

The other issue is the variance in signal strength.  The RTL device at the gate end of the driveway doesn't always pick up the signal when a car goes by, even though the gate receiver always does pick it up and opens the gate. I'd estimate that there is about a 40% miss rate. Looking at the graph, the misses seem to always occur when the RSSI report is -12db.  The distance from the sensor to the RTL device is about the same as the distance to the gate opener receiver, albeit in a different direction.  I have changed the RTL device with another brand, checked the antenna for resonance, and even replaced the sensor transmitter. 
Nothing helps.
[image]

What can I do to increase the hit rate on this?

zuckschwerdt, 2023-06-08:
With clear line-of-sight your signals could maybe report at roughly the same level. I suspect the big variance in levels is due to the auto gain in the RTLSDR. There might be some interference that occasionally drives the gain down -- you would then see the signal at a lower level. The -12 dB is the default cut-off for rtl_433 -- you don't want signals near that as it will stop working ;)

Try to use the `-Y autolevel` to allow a wider range of accepted levels. And maybe experiment with fixed gain settings to get a true (but relative) dB reading. Also try logging the `-M noise` ouput to monitor the band.

buteomont, 2023-06-16:
I tried the `-Y autolevel` and nothing changed, as far as I can tell.  I will try setting a fixed gain. What would be an appropriate starting gain value?  (`rtl_433 -V` shows `rtl_433 version unknown inputs file rtl_tcp RTL-SDR SoapySDR`)

zuckschwerdt, 2023-06-16:
The `-M level -M noise` should give you a good picture of the channel. What do you get for noise levels?
Try to start with gain somewhere in the middle maybe (range should be 0 – 49.6) and check the RSSI. With `-Y autolevel` an RSSI of e.g. -15 should still be fine (for the weakest sender), but you must not get too close to -0 (for the strongest sender).

buteomont, 2023-06-16:
Noise levels on both of the congruent signals ranges from -21 to -27.  The noise on the one at the end of the driveway ranges from -17 to -24.   I am still using the autolevel param, so I guess this might change when I try a fixed gain.

The snr for the two congruent signals goes from 9 to 26.  Every entry pair is identical down to 5 decimal places.  I just don't see how this can be a chance coincidence. The only thing these two devices have in common is the type of device and the fact that they are both being handled by the same RTL receiver.

Here is a small sample of the MQTT output, sorted by SNR value:

```
rtl_433/buteoha/devices/DrivewayAlert/snr 	11.80456
rtl_433/buteoha/devices/PowellRidgeAlert/snr 11.80456
rtl_433/buteoha/devices/DrivewayAlert/snr 	12.36572
rtl_433/buteoha/devices/PowellRidgeAlert/snr 12.36572
rtl_433/buteoha/devices/DrivewayAlert/snr 	12.59637
rtl_433/buteoha/devices/PowellRidgeAlert/snr 12.59637
rtl_433/buteoha/devices/DrivewayAlert/snr 	13.0103
rtl_433/buteoha/devices/PowellRidgeAlert/snr 13.0103
rtl_433/buteoha/devices/DrivewayAlert/snr 	14.41612
rtl_433/buteoha/devices/PowellRidgeAlert/snr 14.41612
rtl_433/buteoha/devices/DrivewayAlert/snr 	15.3252
rtl_433/buteoha/devices/PowellRidgeAlert/snr 15.3252
rtl_433/buteoha/devices/DrivewayAlert/snr 	15.39912
rtl_433/buteoha/devices/PowellRidgeAlert/snr 15.39912
rtl_433/buteoha/devices/DrivewayAlert/snr 	15.4072
rtl_433/buteoha/devices/PowellRidgeAlert/snr 15.4072
rtl_433/buteoha/devices/DrivewayAlert/snr 	15.7682
rtl_433/buteoha/devices/PowellRidgeAlert/snr 15.7682
rtl_433/buteoha/devices/DrivewayAlert/snr 	16.10128
rtl_433/buteoha/devices/PowellRidgeAlert/snr 16.10128
rtl_433/buteoha/devices/DrivewayAlert/snr 	16.14399
rtl_433/buteoha/devices/PowellRidgeAlert/snr 16.14399
rtl_433/buteoha/devices/DrivewayAlert/snr 	16.20086
rtl_433/buteoha/devices/PowellRidgeAlert/snr 16.20086
rtl_433/buteoha/devices/DrivewayAlert/snr 	16.25259
rtl_433/buteoha/devices/PowellRidgeAlert/snr 16.25259
rtl_433/buteoha/devices/DrivewayAlert/snr 	16.33679
rtl_433/buteoha/devices/PowellRidgeAlert/snr 16.33679
rtl_433/buteoha/devices/DrivewayAlert/snr 	16.47138
rtl_433/buteoha/devices/PowellRidgeAlert/snr 16.47138
```

gdt, 2023-09-26:
I think this is a question that has been answered, and thus should not remain open.  If you want to articulate that there is  a bug, please do.  Otherwise, please close.

zuckschwerdt, 2023-10-21:
I remember other discussions (on the list I think) about Mighty Mule sensors. I don't remember ever finding out what determines good reception. Maybe this points to an intrinsic problem with that sensor or some quirk of modulation.

buteomont, 2023-10-21:
For the record and future searches, I increased the sample rate (and thereby the bandwidth) and now I get about a 95% hit rate on the gate sensor. I don't remember right now to what value I set it,  but I believe it was about 50% more than the default.