merbanan/rtl_433 #2528: Decode signal from electric toothbrushes (e.g. Braun Oral-B Triumph)

sheilbronn, 2023-06-13:
Hello,

Has anybody tried to decode the signals form a Braun electric toothbrush, e.g. a Braun Oral-B Triumph 5000 Smartguide?
(The Braun device type would 3737 or 3738 or could be a similar one as a starter).

Google'ing has yielded no results. Just wanted to make sure I'm missing something and also would like to document any findings here.

How could I **find the frequency the toothbrush uses** to send the signal to the smartguide display?
The German manual has no reference to a frequency. Would it be 433Mhz for any toothbrushes? 
Any hints would be very much appreciated.

(PS: Newer Braun Oral-B electric toothbrushes might have Bluetooth but this is not what I'm looking for)

zuckschwerdt, 2023-06-13:
If you look inside there should be a CC2500 which is a 2,4 GHz transceiver.

closed 2023-06-13 by zuckschwerdt

tomtastic, 2023-10-26:
For future curious people, the FCC ID doc for SmartGuide model 3762 (also using the CC2500) : https://fccid.io/USQ3762/Test-Report/Test-Report-1451094
The brush transmits in four possible channels within the `2400 - 2483,5 MHz band`:
* 2403MHz
* 2425MHz
* 2427MHz
* 2452MHz

sheilbronn, 2023-10-26:
Here is the publicly available data from the certification:

- 2.4 GHz ISM transmitter
- Uses 4 channels (2403/2425/ 2427/2452 MHz) which are changed sequentially.
- Sequence repeats after 36 channel changes. 
- Single packet is transmitted on all 4 channels at least one time before information will be updated.
- Method of frequency generation: Integrated PLL system to achieve self adaptive carrier frequency to antenna resonant frequency
- Type of modulation: **Binary Frequency Shift Keying (2-FSK)** for details refer to the specification of the front end IC
- Duty cycle of the transmitter: Duty cycle: in practical use approx. 18% while transmitting at each frequency 
- Total length of one active transmission period: approx. 2 minutes per user, typically not more than 4 users per hour. 
- Toothbrush transmits only when motor is running and for 1 sec after motor has stopped or for 1 sec when toothbrush is put on the charger. Also when a button is pressed or when toothbrush is in setup mode.

I guess it is difficult to find an SDR device for the 2.4 GHz band... ?

tomtastic, 2023-10-27:
I have a capable SDR, but the Braun 3762 I have is not working so I can’t capture a signal.

sheilbronn, 2023-10-27:
> I have a capable SDR, but the Braun 3762 I have is not working so I can’t capture a signal. 

Which type of SDR can Do this? Thx.

tomtastic, 2023-10-27:
Lots of them will, from all the LimeSDRs, to the HackRF One, the ADALM-PLUTO, etc.