merbanan/rtl_433 #2529: Ecowitt WN32B not detected / decoded

antamy, 2023-06-14:
I'm new to RTL_433, so apologies upfront if I'm not conveying the correct information.  This is similar to issue #2303, but seems different enough that I'm creating a new issue.

The device is an Ecowitt WN32B 915MHz (no barometric pressure).  I applied the code changes in #2303, but they don't have any effect. 

I do see this in the output:

```
rtl_433 -f 915.05M
...
codes     : {170}aaaaaaaaaaaaaa2dd4d122663affff915c000000000
```

which contains the correct data using the decode info in `fineoffset.c` 

```
aa 2d d4 d1 22 66 3a ff ff 91 5c
         MI IT TT HH PP PP CC XX
```
Running with `-vv` doesn't show show `fineoffset_WH25_callback` anywhere, so I assume the code in `fineoffset.c` is not being invoked.  I'm not sure what / where to change to enable this device to be detected.

Any ideas?

Thanks

ProfBoc75, 2023-06-14:
hi @antamy 

With the changes from #2303 it works with your sample. 

```
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : @0.060813s
model     : Fineoffset-WH32                        ID        : 18
Battery   : 1            Temperature: 19.6 C       Humidity  : 60 %          Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : @0.104606s
model     : Fineoffset-WH32                        ID        : 18
Battery   : 1            Temperature: 19.6 C       Humidity  : 60 %          Integrity : CRC
```

I guess your file fineoffset.c is not properly updated

antamy, 2023-06-14:
> I guess your file fineoffset.c is not properly updated

Thanks for looking at this.  I haven't done any programming for years, and getting my head round all the tools and environment is still a work in progress.  My issue was that the `make install` actually didn't work (it gave an error on updating man) and I assumed that the man piece could be ignored.  Turns out that the other steps never ran, so I was using an older version of rtl_433.  Running `sudo make install` fixed my issue.

That said, if I hadn't encountered the problem, I wouldn't have learned as much about rtl_433 as I did last night trying to figure it out!

closed 2023-06-14 by antamy