merbanan/rtl_433 #2531: Add support for Landis+Gyr G5 Integrated FOCUS AXe Power Meter

MidwestTechnologies, 2023-06-14:
The power company installed new power meters recently so I decided to embark on yet another rtl_433 project.  I've been going it alone for more than a week and decided to have the experts take a stab at it.  I've not been able to come up with anything resembling useful data.

FCC ID: R7PEG1R1S6
https://fccid.io/R7PEG1R1S6

I believe mine is using mode 1 (see page 3 of the test report) at 9.6 kbps (104.17 uS per pulse).

Test report: https://fccid.io/R7PEG1R1S6/Test-Report/Test-Report-2789309

If this unit is sending the LAN ID, this should be 0x900F418E as printed on the meter sticker.  This has been what I have been searching for to no avail.

Sample: [g024_916.15M_250k.cu8.zip](https://github.com/merbanan/rtl_433/files/11746721/g024_916.15M_250k.cu8.zip)

Close (within about 10 seconds) to the time this sample was recorded, my meter was  reading 182 kW delivered, 00000 recorded, "KW" 6.336 kW, 245.031 volts, and "IKW" 0.897.  I can't get all of the readings at the time of transmission because have to wait for the meter to scroll through.

Throwing this into bit bench, I've gotten close to the classic sync word of  0xD391 but found 0xD351 and 0xD371 at times.

GFSK or FSK may be in use according to the information found in the links above.

Bit bench for the above sample: <a href="http://triq.net/bitbench#c=55%2055%2055%2055%2055%2055%2040%200F%20FF%202A%2055%2050%2005%2089%2006%205F%20F7%20FD%20FF%207F%20DF%20F7%20FC%20FF%2004%20DE%2012%2014%2043%2000%2043%20B0%2004%20C1%2001%20D3%2051%202D%20A1%2004%20DE%2012%2014%2043%2040%2040%2016%2044%2021%2060%204A%2016%20BC%20FD%2007%2059%2017%20DC&f=hh%20ID%3Ahh%20b%20CH3d%20TEMP_C%3A12d%20HUM%3Ad%20CRC%3A8h%20%7C%208h%2016h%2016h%20&a=Preamble&m=&r=true&cw=4"> here </a>

There may be two or more packets in one transmission.  The above transmission was recorded without an antenna connected.  Let me know if different sample rates are needed.  I have recorded samples at 1M with worse results.  Thanks.

ProfBoc75, 2023-06-15:
Hi @MidwestTechnologies 

Can you try more messages with :
```
rtl_433 -f 916M -X "n=landis,m=FSK_PCM,s=104,l=104,r=20000" -s 1024k
```

MidwestTechnologies, 2023-06-15:
Bit bench using above settings: <a href="http://triq.net/bitbench?c=f0%206d%200a%20fd%20ff%20ff%20ff%2027%20%5B%23%20no%20wind%5D&c=f0%206d%200a%20fd%20ff%20ff%20ff%2027%20%5B%23%20no%20wind%5D&c=f0%206d%200a%20fd%20fe%208f%20ff%205a%20%5B%23%20weak%20wind%5D&c=f0%206d%200a%20fd%20fe%20ef%20ff%200f%20%5B%23%20weak%20wind%5D&c=f0%206d%200a%20fd%20f4%20af%20ff%2058%20%5B%23%20strong%20wind%5D&c=f0%206d%200a%20fd%20f7%20ef%20ff%201b%20%5B%23%20strong%20wind%5D&c=f0%206d%200a%20fd%20ff%20ff%20ff%2027%20%5B%23%20no%20wind%5D&c=f0%206d%200a%20fd%20ff%20ff%20ff%2027%20%5B%23%20no%20wind%5D&c=f0%206d%200a%20fd%20ff%20ff%20ff%2027%20%5B%23%20no%20wind%5D&c=f0%206d%200a%20fd%20ff%20ff%20ff%2027%20%5B%23%20no%20wind%5D&f=hh%20hh%20hh%20hhh%20WIND%3A%20~8d%20STUFF%3A%20h%20hh%20CRC%3A%20#c=d555555555555003ffca955401624197fdff7fdff7fdff3fc13385776a4011bc011842f40c4b68413385776a50100511084415856f3f41d1c5b50&f=hh%20hh%20hh%20hhh%20WIND%3A%20~8d%20STUFF%3A%20h%20hh%20CRC%3A%20&a=Sync&m=&r=true&cw=4">here</a>

MidwestTechnologies, 2023-06-16:
Here's a sample too @ 1024k : 
[g002_916M_1024k.cu8.zip](https://github.com/merbanan/rtl_433/files/11772455/g002_916M_1024k.cu8.zip)

MidwestTechnologies, 2023-08-03:
Progress has been slow these last couple of months.  I have analyzed the data and I'm not getting anywhere with it.  These meters work in a mesh network and repeat transmissions from other meters.  Most of the meters that I have observed have LAN addresses printed on their stickers thar begin with 0x900F but I have observed one meter that began with 0x90EE.

krvmk, 2023-08-24:
Have you seen this project?https://github.com/BitBangingBytes/gr-smart_meters
They seem to have got further with dissecting and decoding the packets for the Gridstream Rf protocol using gnuradio. I'm not sure how easily that would translate into rtl_433.

MidwestTechnologies, 2023-08-24:
I did look that over a couple of months ago when I got into this.  There is a lot of good information there and I was able to take that sync word (or much of it) and apply it to rtl_433.  Currently, I am using sync 0xaaaaaaaaaaaaa007ff.  I could probably add 0x952a to that sync word because all messages have that in common.

The trouble is, I don't see the LAN IDs for the two meters on my property in any of the data.  I even tried the GPS algorithm found on Recessim and could not get a solution that contained my coordinates or coordinates that are even close to me.  The rest of the data in my decodes don't really match up well to the Recessim formats.

I expect at least some of these packets to contain the ID and power consumption at a minimum.  Some likely contain repeated data with the ID of the station that is being repeated and its consumption since this is a mesh network.

I would much rather use rtl_433 to do this because it allows me to decode many different sensors on one SDR.  I will need to run rtl_433 with one radio and the Recessim package with another radio otherwise.

Thanks for your help.

krvmk, 2023-08-26:
I would also rather use rtl_433 instead of gnuradio, but I'm still trying to work out parity between the recessim package and the output from rtl_433. Can you share what are using as a flex decoder?

merbanan, 2023-08-26:
@MidwestTechnologies can we get access to more signal recordings? And try with some other sample rates also. From the spreadsheet file it looks like the demodulation is unstable because the amount of bits are varying (209-213).

merbanan, 2023-08-26:
One more thing, the signal is slightly clipped. Adjusting the gain should result in a more clear demodulated signal. Maybe that will be enough to recover the bits properly.

MidwestTechnologies, 2023-08-26:
@krvmk  certainly...here is the complete command I used including the flex...

rtl_433 -d 1 -f 914.915M -s 1024k -M level -Y minmax -F csv:log_landis.csv -X "name=landis,m=FSK_PCM,s=104,l=104,r=20000,tolerance=4,preamble={72}0xaaaaaaaaaaaaa007ff" -R -149 -F kv

Again, I should have added 0x952a to the end of sync above.  LAN IDs in my neighborhood all so far begin with 0x90 and should be 4 bytes long (i.e. 90xxxxxx) according to the meter sticker.  There is a second nine digit ID, specific to the power company, on the sticker.  None of my decodes match these numbers, either.  

@merbanan  I can get you more samples but, in addition to 1024k, I have tried 250k, 1000k, and 1200k.  I still saw the variance that you mentioned.  I tried 1400k and 2400k as well and rtl_433 was no longer able to decode.  If you have any suggested sample rates, please let me know.

Thank you all for your guidance.

MidwestTechnologies, 2023-08-26:
@merbanan Just now saw you comment on gain.  I tried without any antenna hooked up as one of these Landis meters is only 20 feet from my radio.  I also tried a 10 dB attenuator in line with my outdoor antenna.  With the attenuator, I still got variance in the decodes.  Without the antenna connected, I got no decodes in a couple of hours.  I think one thing I could have done differently is let rtl_433 run longer without the antenna.  I'm not sure how often these meters transmit and it could take time to hear my meter on the frequency I'm monitoring.  Any suggestions are appreciated.

krvmk, 2023-08-26:
Since the Landis meters use FHSS, I use the max sample rate for the rtl-sdr of 2560k. I've had good luck seeing signals around 904M instead of 915M for my personal meter, since the rate is relatively narrow compared to the 902-928 frequency range of the meter.

merbanan, 2023-08-26:
The bandwidth and sample rate is 1to1ish on the rtl-sdr. So too high sample rate and the band is so wide that we get in disturbances. Anyway @MidwestTechnologies I guess that you could try with 1000k and 1200k then. Regarding FHSS we should eventually see some signals around the observed band. The demodulator can not handle to too many strong signals at the same time.

Anyway if you run the following:
rtl_433 -Y minmax -s 1024k -r signals/LFG/g002_916M_1024k.cu8 -W test.sr
we can see that the signal is almost completely framed. It looks like we are loosing like 5 bits at the end.

But the pulse recovery is riddled with short pulses (lets call them tiny pulses).
[image]
If we add -vvvv we get the pulse table output that is used for bit recovery.

Pulse data: 185 pulses
[  0] Pulse:    0, Gap:    6, Period:    6
[  1] Pulse:    1, Gap:    1, Period:    2
[  2] Pulse:    2, Gap:    1, Period:    3
[  3] Pulse:    1, Gap:    1, Period:    2
[  4] Pulse:    2, Gap:    1, Period:    3
[  5] Pulse:    2, Gap:    1, Period:    3
[  6] Pulse:    1, Gap:    2, Period:    3
[  7] Pulse:    1, Gap:    1, Period:    2
[  8] Pulse:    2, Gap:    4, Period:    6
[  9] Pulse:    1, Gap:    2, Period:    3
[ 10] Pulse:    1, Gap:    1, Period:    2
[ 11] Pulse:    1, Gap:    1, Period:    2
[ 12] Pulse:    2, Gap:    6, Period:    8
[ 13] Pulse:    2, Gap:    1, Period:    3
[ 14] Pulse:    2, Gap:    3, Period:    5
[ 15] Pulse:    2, Gap:    2, Period:    4
[ 16] Pulse:    1, Gap:    1, Period:    2
[ 17] Pulse:    1, Gap:    1, Period:    2
[ 18] Pulse:    2, Gap:    1, Period:    3
[ 19] Pulse:    2, Gap:    1, Period:    3
[ 20] Pulse:    1, Gap:    1, Period:    2
[ 21] Pulse:    2, Gap:    1, Period:    3
[ 22] Pulse:    2, Gap:    4, Period:    6
[ 23] Pulse:    1, Gap:    1, Period:    2
[ 24] Pulse:    2, Gap:    3, Period:    5
[ 25] Pulse:    2, Gap:    1, Period:    3
[ 26] Pulse:    4, Gap:    2, Period:    6
[ 27] Pulse:    1, Gap:    4, Period:    5
[ 28] Pulse:    1, Gap:    2, Period:    3
[ 29] Pulse:    1, Gap:    1, Period:    2
[ 30] Pulse:    4, Gap:    1, Period:    5
[ 31] Pulse:  218, Gap:  105, Period:  323
[ 32] Pulse:    1, Gap:    1, Period:    2
[ 33] Pulse:  107, Gap:  104, Period:  211
[ 34] Pulse:  108, Gap:    1, Period:  109
[ 35] Pulse:    1, Gap:  103, Period:  104
[ 36] Pulse:    1, Gap:    1, Period:    2
[ 37] Pulse:  108, Gap:  103, Period:  211
[ 38] Pulse:    1, Gap:    1, Period:    2
[ 39] Pulse:  108, Gap:  107, Period:  215
[ 40] Pulse:  106, Gap:  105, Period:  211
[ 41] Pulse:    1, Gap:    1, Period:    2
[ 42] Pulse:  106, Gap:   39, Period:  145
[ 43] Pulse:    1, Gap:   65, Period:   66
[ 44] Pulse:  108, Gap:  107, Period:  215
[ 45] Pulse:  106, Gap:  107, Period:  213
[ 46] Pulse:  106, Gap:  108, Period:  214
[ 47] Pulse:  105, Gap:  108, Period:  213
[ 48] Pulse:   85, Gap:    1, Period:   86
[ 49] Pulse:   19, Gap:  108, Period:  127
[ 50] Pulse:  107, Gap:  107, Period:  214
[ 51] Pulse:  106, Gap:  107, Period:  213
[ 52] Pulse:  106, Gap:  107, Period:  213
[ 53] Pulse:  106, Gap:  107, Period:  213
[ 54] Pulse:  106, Gap:  105, Period:  211
[ 55] Pulse:    1, Gap:    1, Period:    2
[ 56] Pulse:  108, Gap:  105, Period:  213
[ 57] Pulse:  108, Gap:  105, Period:  213
[ 58] Pulse:  106, Gap:  108, Period:  214
[... 103 lines cut ...]
[162] Pulse:    1, Gap:  406, Period:  407
[163] Pulse:    1, Gap:    1, Period:    2
[164] Pulse:  107, Gap:  168, Period:  275
[165] Pulse:    1, Gap:  256, Period:  257
[166] Pulse:  107, Gap:  320, Period:  427
[167] Pulse:  106, Gap:  535, Period:  641
[168] Pulse:    1, Gap:    1, Period:    2
[169] Pulse:  103, Gap:    2, Period:  105
[170] Pulse:    1, Gap:  533, Period:  534
[171] Pulse:  105, Gap:  321, Period:  426
[172] Pulse:  106, Gap:  108, Period:  214
[173] Pulse:  107, Gap:  211, Period:  318
[174] Pulse:  108, Gap:  105, Period:  213
[175] Pulse:  108, Gap:  106, Period:  214
[176] Pulse:  214, Gap:  212, Period:  426
[177] Pulse:  642, Gap:  105, Period:  747
[178] Pulse:  108, Gap:  533, Period:  641
[179] Pulse:  320, Gap:  106, Period:  426
[180] Pulse:  108, Gap:  425, Period:  533
[181] Pulse:  321, Gap:  214, Period:  535
[182] Pulse:  104, Gap:  109, Period:  213
[183] Pulse:  105, Gap:  108, Period:  213
[184] Pulse:  213, Gap:  215, Period:  428

Either we get a cleaner signal or we manipulate the pulse table and do a pulse merge when we get the tiny pulses. Maybe @zuckschwerdt can elaborate if it would be feasible to dynamically per r_device merge pulses and gaps shorter then lets say 10% of device->short_width. That should help with noisy signals and signals with a low bandwidth.

krvmk, 2023-08-27:
I have a capture of a good signal from rtl_433 and using the gnuradio package, I know what the output should be. However I captured it at the higher sample rate and can see the signal on triq. Above and below about 600k I can see what I would assume is harmonics. Is there a decimate feature that I can give it to isolate 2560k down to 1000k if I give it the frequency?

zuckschwerdt, 2023-08-27:
We don't have such a tool (shift, decimate). But perhaps you could build it with gnuradio?

merbanan, 2023-08-28:
This sample is clean and we can use the flex decoder to pick up the sync words.

build/src/rtl_433 -Y minmax -s 250k -r g00*_904.4M_250k.cu8 -X "name=landis,m=FSK_PCM,s=104,l=104,r=20000,tolerance=10,preamble={28}0xaaaa005"

```
{278}ff2a557005d14a403374575c4037f48956d0104cc10047b055b7224516d45100d45084
{279}ff2a55005d14a403375431ec037f48956dc704cc10044705543725517d571054bb494
{278}ff2a557005d14a403375852f4037f48956dd104cc10050705581364a1784b10050f594
```
I checked the bits and at least one file matched up. This does not match gr-smart meters though. There is a code block called deframer:
https://github.com/BitBangingBytes/gr-smart_meters/blob/maint-3.10/lib/Deframer_impl.cc#L100
and the wiki mentions start and stop bits.

If we take ff2a557005d14a40337 we get the following bits:

```
      111 1111 1
0 0101 0100 1 [54] bit reverse [2A]
0 1010 1011 1 [AB] ~ [D5]
0 0000 0000 1 [00] ~ [00]
0 1110 1000 1 [E8] ~ [17]
0 1001 0100 1 [94]
0 0000 0001 1 [01]
0 0110 111
```
The starting 0 bit and ending 1 bit holds for this section and probably the rest. And matching against the wiki we start with 2A and then D5. After that we have the length in 2 bytes. 0x17 is 23 and 23*10 + CRC and some headers could be around 278 bits.

So this info should now be implemented into a decoder.

MidwestTechnologies, 2023-08-29:
I can confirm @merbanan  that using the 0 start bit and 1 stop bit, I am now able to decode one of my meter's IDs.  I used one of my saved decodes and manually deleted the start and stop bits then reflected the 8 bits in between and got the ID.

krvmk, 2023-08-29:
I was hoping that decode_uart would handle the 8N1 decoding with the flex decoder, but it doesn't seem to work like I thought it would.

merbanan, 2023-08-29:
Works for me:

rtl_433 -Y minmax -s 250k -r signals/LFG/g00*_904.4M_250k.cu8 -X "name=landis,m=FSK_PCM,s=104,l=104,r=20000,tolerance=10,decode_uart,preamble={36}0xaaaa005ff"

And the decoded data matches the supplied files.

MidwestTechnologies, 2023-08-29:
Mine is working off the radio with the following:
rtl_433 -d 1 -f 904.4M -R -149 -Y minmax -s 1000k  -M level -F csv:log_landis_new.csv -F kv -X "name=landis,m=FSK_PCM,s=104,l=104,r=20000,tolerance=10,decode_uart,preamble={72}0x555555555555400fff"

krvmk, 2023-08-29:
Mine is working live now. I think I had messed up the preamble. It doesn't seem like it is picking up the signal as easily as gnuradio.

rct, 2023-08-29:
I have a similar, but not the same meter.  My meter label says `Focus RXR`.   However the FCC ID: R7PEG1R1S2 says *Landis+Gyr Technology, Inc. Gridstream RF Enhanced Integrated Focus AX Meter EG1R1S2*  - https://fccid.io/R7PEG1R1S2.

The FCC ID of your meter is ..S6 rather than ...S2.   My utility is PSEGLI.  I'm in Long Island, NY, USA. 

I haven't tried to receive anything from it yet because I wasn't sure where to start.  Also I was hoping some day the utility would have a way to get the built in Zigbee radio enabled. 

If it helps, I can capture and contribute signals. Any recommendations for capturing signals welcome. How often does it transmit?  How long does it take to frequency hop through it's range and come back to the same frequency?

krvmk, 2023-08-29:
I believe the Gridstream protocol is fairly standard across the Landis+Gyr meter line, with some variations with v4 vs v5. Based on what I have captured thus far, the meters will often broadcast updates simultaneously on multiple frequencies in the 902-928 MHz range, so tuning anywhere in that range for long enough should be sufficient. 

While rtl_433 is able to convert the bits into a datastream, I've not seen any indicators of how to decode the power usage that it is sending up to the provider.

merbanan, 2023-08-29:
Well we need a proper decoder based on the information from here:
https://wiki.recessim.com/view/Landis%2BGyr_GridStream_Protocol
and then we need to analyse the unknown fields.

Before that happens it would be of interest to test some sample rates to see what settings give more received messages.

krvmk, 2023-08-29:
The hardest part I had in getting it initially decoding with gnuradio was finding the right value for the CRC check, since it seems like bad signal is fairly common with a frequency hopping mesh network like they use. Each provider uses their own code, and I used this [https://wiki.recessim.com/view/Gr-smart_meters_Setup_Guide#Determining_your_power_providers_CRC](https://wiki.recessim.com/view/Gr-smart_meters_Setup_Guide#Determining_your_power_providers_CRC) to find mine.
I had several dozen codes initially captured that apparently had bad data, as the tool was not able to find a match. Once I got enough good packets and found the CRC init value, I could at least validate the data was not corrupted. 

Since that value will vary, can that be passed to a proper decoder somehow? Or would it be a matter that it would have to figure it out on its own after seeing X amount of packets captured?

rct, 2023-08-29:
> I believe the Gridstream protocol is fairly standard across the Landis+Gyr meter line, with some variations with v4 vs v5. 

On the face, my meter says `Model 40-1565 Series IV`  so maybe v4?

> Based on what I have captured thus far, the meters will often broadcast updates simultaneously on multiple frequencies in the 902-928 MHz range, so tuning anywhere in that range for long enough should be sufficient.

So how long is "long enough" -- do I need to wait some number of minutes or hours before deciding to try something else?

Thanks

krvmk, 2023-08-29:
I usually get something after 5-10 minutes or so, but it is hard to tell for certain how much is coming from my meter vs the neighborhood. Since it is a mesh, all the meters will relay traffic between each other to get back to the company.

MidwestTechnologies, 2023-08-29:
I've been sitting on 904.4 MHz for some time now.  My meter IDs show up about once every half hour average.  That may not be typical of every situation.  However, I am getting my neighbors' meters about once every 5 minutes or so.  I had to turn my gain down quite a lot to get good decodes at 1000k sample rate.

krvmk, 2023-08-29:
In theory, if your meter id is in the stream, then it is probably coming from your meter, so if you add "match={32}0x" and your meter id to the flex decoder, that should limit it. 

Some caution about posting flex rows online  as according to the recessim site, some power companies use geolocation data to generate the meterid.

merbanan, 2023-08-29:
For the record, data from this thread:

```
./reveng -w 16 -s 298076437a807f22b517900600c15003360a0f1a00e14d 29807685bc807f22b5c7900600c4508527155f1d50ba49 298076d41d807f22b501900600bc50db22145b14804508
width=16  poly=0x1021  init=0xe623  refin=false  refout=false  xorout=0x0000  check=0x9fc1  residue=0x0000  name=(none)

```

and then data from https://wiki.recessim.com/

```
./reveng -w 16 -s 29F0611C3EF03B9FFBAE98401416309B47 29F0611C3EF03B9FFBBA9804B51650C505 29F0611C3EF03B9FFBCA9808441600413C
width=16  poly=0x1021  init=0x5fd6  refin=false  refout=false  xorout=0x0000  check=0x139e  residue=0x0000  name=(none)

```

merbanan, 2023-08-29:
So we have 2 different inits 0xe623/0x5fd6, so far.

merbanan, 2023-08-29:
Anyway regarding the init value we will add a table in the decoder and then check against all known values.

merbanan, 2023-08-29:
https://www.landisgyr.com/webfoo/wp-content/uploads/2012/12/PS_RFMesh-CI-Endpoints-12-21.pdf

"In addition to kWh, kW, time-of-use and
voltage readings, the RF endpoint reports
load profile and up to 5-minute interval data
for billing, engineering and customer service
applications."

So I guess that most values that can been seen on the meter display can also be seen over radio.

merbanan, 2023-08-29:
https://www.landisgyr.com/webfoo/wp-content/uploads/2012/12/PS_GridstreamRFC7500Collector.pdf

And this document has some signal specific parameters.

krvmk, 2023-08-30:
Some additional init values are also noted here: [https://github.com/BitBangingBytes/gr-smart_meters/blob/maint-3.10/README.md](https://github.com/BitBangingBytes/gr-smart_meters/blob/maint-3.10/README.md) 

CRC = 0xD553
CRC = 0x45F8
CRC = 0x5FD6
CRC = 0x62C1
CRC = 0x23D1
CRC = 0x2C22
CRC = 0x142A

MidwestTechnologies, 2023-08-30:
I confirm with Recessim that the packet type 2A subtype 55 length 233 contains uptime.   From my log...
{312}2a55002330fffffffffffffe900f418e005200**4fd4ce**a40b900f418e01000f080311c17e70c706

0x4fd4ce corresponds to 5,231,822 seconds or 60.55 days which is precisely the length of time since we had our last outtage.  I imagine the entire counter takes up 32 bits.

krvmk, 2023-08-30:
With the s=104,l=104 timing, it appears that it only works for the 9600b/s transmissions. I'm seeing about s=28,l=28 for the 19200b/s datarates, but I'm still looking for a clean signal capture. Based on what I'm seeing with the decoder framework, there is not any way to support multiple timings in the same decoder.

merbanan, 2023-08-30:
> Based on what I'm seeing with the decoder framework, there is not any way to support multiple timings in the same decoder.

You need one definition per rate but you can call the same decoder code. It will work well as long as the signals are decoded properly.

krvmk, 2023-09-02:
My first attempt at a decoder. I did not add the possible GPS decoding or unit conversion for uptime. I also did not map any of the unknown values. I did add a -v output for CRC failure in order to determine any new CRC init values. There is also -vv output for dumping the decoded bytes that passed CRC.

MidwestTechnologies, 2023-09-04:
I'm still looking at the unknowns and trying to decipher what is going on.  While the timer mentioned on Recessim resets after 167.99 seconds, the timers in my neighborhood appear to reset every 60 sec.  Here are two captures from the same meter, one from 15:45:18 and another from 15:46:18.  

{216}2ad5001729900ee111900ee245368801000f080304**0323**11603a62
{216}2ad5001729900ee111900ee2455a8801000f080304**031a**1140e0a8

0x323 = 803 / 100 = 8.03 sec
0x31a = 794 / 100 = 7.94 sec

So the timer must have reset at 10 seconds after the minute.

merbanan, 2023-09-04:
The timers could be timestamps used to derive power values. If you compare the bitstreams you will see another difference and that is 36 vs 5a. It could be that this is the encoded meter reading. If you include this counter you can derive a power envelope from a kWh reading.

krvmk, 2023-09-05:
I believe the wiki referred to it as a "counter" but not sure about the scale, as it is likely less than 1kwh, since it goes up faster than the display. Would probably need to track it over an extended period and compare it with the display and what the provider recorded.

merbanan, 2023-09-05:
I found one meter that seems to use the Gridstream Protocol.

https://fccid.io/R7PEG1R1S2/User-Manual/Manual-1457951

If you read the manual you can find out what data that can be transmitted over the radio. Line voltage, current and phase angle for example. But it is not clear what is transmitted automatically and what needs to be requested.

merbanan, 2023-09-05:
But to be able to further analyze things we need long sequences of messages and corresponding readings of the meter values.

zuckschwerdt, 2023-09-06:
"rows" can be arbitrarily long (we called that "row spill" in the commit), but must always be used as `uint8_t *` never as `bitrow_t`

MidwestTechnologies, 2023-09-08:
I believe that within the 2 bytes following the timer in the 2A D5 messages are the number of hops that were required to relay the message, specifically the low nibble of the first byte.

The first example here is from two meters known to be within about 50 feet of each other.
{208}2ad5001629900f41e2900f428414b801000f08030c59**1100**386b

So in this case, 1 hop.

The second example shows two meters that are known to be ~1650 ft (~500m) apart.
256}2ad5001c22900f43ed900f418e840101001208030520302d8480096c**1b90**d4c7

So there  are 11 hops in this example.  I have seen consistently, when the same meters are paired, that low nibble of the first byte has the same value.  The high nibble of the first byte might be number of tries.  I've never seen it above 4.

I thought maybe the high nibble of the second byte had usage data per time from the timer in the preceding 2 bytes.  But even when my usage reported from the utility website is 0, there is still a a value greater than 0 in the nibble.  The low nibble of the second byte is always 0 from my observations.

One of my meters has a single load on it which is off for extended periods of time.  I am very interested in seeing if I can figure out consumption from that meter.  It is counter-intuitive to me that these meters seem more concerned about relaying uptime than usage but I continue to investigate.

250k sample rate works best for me.  

Many thanks to those who have contributed.

krvmk, 2023-09-08:
Interesting theory. The hops nibble could likely be proven if you are able to compare D5 packet from a source meter and then a relayed D5 packet with the same source in a different location.

Are you staying with the 904.4MHz frequency or are you hopping around the spectrum?

merbanan, 2023-09-08:
The meters can report historical consumption. The messages with a 253 bytes large payload  sounds like a candidate for that.

From the wiki we have this list of packets:

https://wiki.recessim.com/w/images/9/9e/D5-Packets-Table-F03B9FFB.txt

If I take one of the messages from that list and add your data we get this list:

```
2a d5 001c 22 900f43ed 900f418e 84 |0101001208030520302d8480|   096c 1[b]9[0] d4c7

2a d5 0016 29 900f41e2 900f4284 14 |b801000f0803|               0c59 1[1]0[0] 386b
2A D5 001D 29 F0611C3E F03B9FFB 51 |91080091504CAD0129BD448152| 1350 1[7]4[0] 8FAD
```

To confirm the hop theory you should be able to log packets from a remote meter and then as you move closer to that meter the hop counter should go down. To get consumption we need to figure out which payload/packet between the || that correspond to that. I am quite sure that the data around only handles the transport layer data of the GridStream protocol.

But we need lots of messages to be able to figure out how it is represented.

merbanan, 2023-09-09:
If meters can route messages for other meters then there needs to be a max jump counter. Otherwise a message can be relayed forever.

MidwestTechnologies, 2023-09-09:
@krvmk There is a meter about 200 ft closer to the far meter that I have identified but I have not seen the far meter and the "nearer" meter communicate yet.  I am letting the log fill up and looking for changes.  I'll report back when I get more information.  I continue to monitor on the 904.4 MHz frequency only.

@merbanan I have not observed the D5 1D packet yet at my location.  However, the D5 47 has been seen perhaps once per hour.  It has quite a bit of information in it but I haven't correlated anything yet to what the utility website is reporting.  The D5 1C packet is seen too but very rarely.  I agree on your point about an upper limit to hopping.  In my mind, I imagine the meters closest to the collection points doing a lot of the forwarding work.

krvmk, 2023-09-09:
All of the meters here do not appear to be geolocated. However, I do see some traffic from a single node that is, and it doesn't appear to be tied to a structure when I look at it on Google Maps. I'm wondering if they have neighborhood relays. I have seen some type 55 packets from that device that has an uptime of over two years, when the rest of the neighborhood has had an outage in the last 2-3 months or less. That ID also seems to account for almost half of the traffic that I collected.

MidwestTechnologies, 2023-09-09:
I've been looking for something like [this](https://www.landisgyr.com/webfoo/wp-content/uploads/2012/12/PS_GridstreamRFC6500Collector.pdf) mounted on a pole in my neighborhood, but have not seen one.  This likely reports its coordinates.

krvmk, 2023-09-09:
It may be difficult with the narrow band, but in theory if it is a hops counter, you should be able to see an increment of the counter with the relay from node A to node B to node C.

0rbitlouder, 2023-09-11:
Would  seeing the guts of one of these  help you at all. From what I can tell is one Processor For the data collection.
And one processor. for what appears  to be three radios circuits two together  and one isolated  from  the other  two. They can be set to up date in any number  of time formats 5 10 15 30 up to every  hour  depending on how they  are set up with the software. Zigbee is the software  company  that communicates with them. They are set up by the manufacturer to automatically setup with the buyers peer to peer network Protocol when powered on. Out of the box. I've got some catching up with  you guys I'm sure you all know this and Mention it somewhere in the comments. I'm gonna go back and read. Oh one other thing  there  are two   advanced Low voltage slow dischargeing 4Capacitors Connected two one of the of the radios it is possible sents  the company clams that they  can still up date even dearing long outages it's very possible they charge off an incoming signal. About  every  ten telephone poles there is a box about  as big as old VCR with 3 antennas on them that kinda Confirms my 3 radio hypothesis.. They are fascinating. Advanced and have a lot of Tech inside of them for more than what  you would expect from just a power meter. I got some reading  to catch up on this is so rad I'm not the only  one doing  this .sm:)es

JVital2013, 2023-09-17:
First off, thank to you all for your work on this so far! I have a "Gridstream RF Enhanced Integrated Focus AX" meter, FCC ID R7PEG1R1S2. I tried getting data out of it 2 years ago but was unable to get very far. With the data you provided, I was able to get a new init value for my meter, from PPL Electric in Pennsylvania. It's **0x47F7**.

I've opened a PR on @krvmk's fork with the new init, which seems to reliably work. It does look like the wanaddress column always has an extra 00 byte at the end, but otherwise the data looks OK so far.

Like all of you, I'd love to be able to get the actual meter readings as well. I captured about 9 hours of data with the following command:

```bash
rtl_433 -f 904.4M -s 250k -X "name=landis,m=FSK_PCM,s=104,l=104,r=20000,tolerance=10,decode_uart,preamble={36}0xaaaa005ff" -F csv:log_landis.csv
````
CSV download: [log_landis.csv](https://github.com/merbanan/rtl_433/files/12642939/log_landis.csv)

**Once I got the init, from the above data, I had @krvmk's decoder run for just over an hour and got this data:**
[log_decoded.csv](https://github.com/merbanan/rtl_433/files/12642943/log_decoded.csv).

I hope this data helps the decoder progress - thanks again!

MidwestTechnologies, 2023-09-20:
It's nice to see the contributions here.  I thought I would give an update on my three weeks of observing this data.  I will say these meters are much more powerful than my ERT-SCM protocol using gas meter.  I have trouble receiving gas meters that are more than 750 ft away.  I am able to receive a Landis meter that is 1650 ft away without issue.  This powerful signal needs to bridge the gap in sparsely populated areas to get to the next meter or collector.

The 2A D5 0047 packet contains some interesting data.  There is another timer in there which may have something to do with peak demand time.  It will increment for a while and then decrement for some unknown reason.  There is also some data toward the tail end that changes even when my meter that has one load on it shows 0 kWh per 15 min demand.  

`2a d5 0047 51 900ee2ce 900f418e 30 64f02c1a [00026b24] 0050d2c3 a40b 0101 fe900f418e00 900f418e 0001032500000000000000000000 [00045948] 000f080300f220302d8480 1065 1100 5ca0`

`2a d5 0047 51 900ee2ce 900f418e 50 64f02c3d [00026c82] 0050d2e6 a40b 0101 fe900f418e00 900f418e 0001032500000000000000000000 [000497c8] 000f080300f220302d8480 068e 1150 da`

The above two packets were sent 35 sec apart.  The 4 bytes after the timestamp contain this timer.  0x26b24 = 158500.  Divide by 10 and you get 15,850.  Do the same for the next example and you get 15,885.  The difference is 35.  Unpredictably this data decrements or resets.  The range of this data so far observed at my location has been minimum 7118 to maximum 20206.  The second set of data is also interesting.  I've noticed a range of 44,000 to 983,000.  Even when demand on a meter is 0, these values change.  I've also noticed two different meters with the same timer reading.

I'm also looking over the 001C and 001D packets.  I have not seen a single 001D packet from either of my meters in three weeks. 

 I'm seeing consistency with those alleged "hop counter"bytes at the end.  Routing between the two meters in my example above always contain 11 in the first byte with the second byte changing.  I wish I had more to report as far as reading consumption.  Even on my meter that reports 0 kWh to the web per 15 min interval, I've not seen anything that stops changing.

krvmk, 2023-09-20:
Is it possible that there is a residual usage by the meter itself that the system automatically subtracts?

rct, 2023-09-24:
A few questions:

- What's the best thing(s) for me to be doing in terms of collecting/analyzing data to help with this effort?
- What should I be doing to make sure I capture more messages from my meter? 

Right now I'm running @krvmk's fork.   I'm seeing data from a number of meters in my neighborhood, but **I don't see my meter** in the output.  However, when I run the flex decoder **I do** see my meter's ID in the captured packets. 

I'm listening on 904.4mhz, using a 250K sample rate. Should I be using different values?

I have a script cycling through different gain values, running for 20 minutes on each gain value.  Here's the rough number of decodes from @krvmk's decoder.  The gain value is in the file name

```
4       out-gridstream-0.0-20230923.txt   # auto
5       out-gridstream-0.0-20230924.txt

0       out-gridstream--1.0-20230923.txt  # gain of 0.0
0       out-gridstream--1.0-20230924.txt

0       out-gridstream-1.5-20230923.txt
0       out-gridstream-1.5-20230924.txt
0       out-gridstream-4.0-20230923.txt
0       out-gridstream-4.0-20230924.txt
0       out-gridstream-6.5-20230923.txt
0       out-gridstream-6.5-20230924.txt
0       out-gridstream-9.0-20230923.txt
0       out-gridstream-9.0-20230924.txt
1       out-gridstream-11.5-20230923.txt
0       out-gridstream-11.5-20230924.txt
1       out-gridstream-14.0-20230923.txt
0       out-gridstream-14.0-20230924.txt
1       out-gridstream-16.5-20230923.txt
0       out-gridstream-16.5-20230924.txt
1       out-gridstream-19.0-20230923.txt
2       out-gridstream-19.0-20230924.txt
1       out-gridstream-21.5-20230923.txt
8       out-gridstream-21.5-20230924.txt
6       out-gridstream-24.0-20230923.txt
7       out-gridstream-24.0-20230924.txt
1       out-gridstream-29.0-20230923.txt
3       out-gridstream-29.0-20230924.txt
2       out-gridstream-34.0-20230923.txt
1       out-gridstream-34.0-20230924.txt
4       out-gridstream-42.0-20230923.txt
17      out-gridstream-42.0-20230924.txt
1       out-gridstream-43.0-20230923.txt
7       out-gridstream-43.0-20230924.txt
3       out-gridstream-45.0-20230923.txt
2       out-gridstream-45.0-20230924.txt
3       out-gridstream-47.0-20230923.txt
6       out-gridstream-47.0-20230924.txt
0       out-gridstream-49.0-20230923.txt
3       out-gridstream-49.0-20230924.txt
```

Not sure if this is clue for improving capture or not, the flex decoder seems to capture (the start) of more messages than @krvmk's decoders, but looking at the length byte, many of the received messages seem to be short of the correct length.

MidwestTechnologies, 2023-09-25:
@rct glad you were able to decode some messages.  These meters are relatively very powerful and I have confirmed reception of a meter from 1650 ft away.  I tried no antenna for a while and didn’t have any luck.  Then I tried a rubber duck antenna and started mainly getting messages from my own meters.  I tried adjusting the gain, too.  The 19.7 dB gain tended to work well for me for picking up mainly my own meters.  I have since moved back to my outside antenna at 50 feet and can still get my own meters.  I now keep AGC enabled and 250k sample rate.  I would like to get 1 Msps working but so far no luck.  I think I have over 25,000 log entries now with maybe 10% of those from my own meters.

That’s a good idea on the script to change the gain.  You might try the no antenna config — it might work for your situation. 

Right now, one of my goals is to collect each type of message from at least one of my meters.  I have yet to have decoded the 2A D5 1D packet for one of my meters.  Another goal is to look for the patterns in the other packets already decoded.

gdt, 2023-10-03:
@krvmk It seems obvious to me that a decoder with a checksum is desired, for a protocol with a checksum.

Is the "one you wrote" now in the sources?  If not, is there an open PR?   I would like to move tickets to closed while capturing the work in a place that llives on.

[35 more replies]