merbanan/rtl_433 #2534: Viking 02038 wrong temperature

tuomaz, 2023-06-16:
I have a Viking AB sensor 02038, very similar to this one https://www.lohelectronics.se/hemautomation/433mhz/sensorer-1110/tradlos-temp-luftfukt-for-inne-utomhusbruk

It's identified as Fineoffset-WH5. When the thermometer shows 29 degrees rtl_433 reports -11, a difference of 40. Humidity is decoded correct.

I guess this is similar to https://github.com/merbanan/rtl_433/issues/1691

Can someone point me to the "working solution out of the box" ?

zuckschwerdt, 2023-06-16:
Yes, the WH5 has a different temperature calculation which does not fit other similar devices (e.g. Agimex Rosenborg 66796)
Try this: `-R 18:no-wh5`

tuomaz, 2023-06-16:
Thanks, I will try this. It will disable everything except "Fine Offset Electronics, WH2, WH5, Telldus Temperature/Humidity/Rain Sensor" ?

zuckschwerdt, 2023-06-16:
Yes using just some `-R 123` will disable all other protocols. If you don't want that use: `-R -1 -R 18:no-wh5` (`-R 0` is "disable all" and conversely `-R -1` is "enable all")

tuomaz, 2023-06-16:
It worked, thanks. Why is this working? May I enable some other protocols that I need? Which should I not disable?

zuckschwerdt, 2023-06-16:
Usually you'd want to only enable protocols of the sensors you own. But you can keep the whole list enabled to also grab random sensors of your neighbours and e.g. TPMS.

fredrik-tuomas-lv, 2023-06-19:
What confuses me is the `:no-wh5` part, does this mean to exclude the wh5 decoding from device protocol 18? Is this feature documented somewhere? I wan to learn. If I want to exclude "Fine Offset Electronics" from protocol 18, which name to use?

zuckschwerdt, 2023-06-19:
No, this a special syntax and only used for excluding the WH5 quirk.

fredrik-tuomas-lv, 2023-06-20:
Thanks for all your answers. This issue may be closed.

tuomaz, 2023-06-20:
What confuses me is the :no-wh5 part, does this mean to exclude the wh5 decoding from device protocol 18? Is this feature documented somewhere? I wan to learn. If I want to exclude "Fine Offset Electronics" from protocol 18, which name to use?

tuomaz, 2023-06-20:
Thanks for all your answers. This issue may be closed.