merbanan/rtl_433 #2535: EMOS E6016 not working?

djmalin, 2023-06-16:
Hello, I have a problem with starting this weather station, I don't know what I'm doing wrong?
My configuration:
`# This is an empty template for configuring rtl_433. mqtt information will be
# automatically added. Create multiple files ending in '.conf.template' to
# manage multiple rtl_433 radios, being sure to set the 'device' setting. The
# device must be set before mqtt output lines.
# https://github.com/merbanan/rtl_433/blob/master/conf/rtl_433.example.conf

output mqtt://${192.168.1.xx}:${1883},user=${xxx},pass=${wetererexx},retain=${retain}
report_meta time:iso:usec:tz

# To keep the same topics when switching between the normal and edge versions,
# use this output line instead.
# output mqtt://${host}:${port},user=${username},pass=${password},retain=${retain},devices=rtl_433/9b13b3f4-rtl433/devices[/type][/model][/subtype][/channel][/id],events=rtl_433/9b13b3f4-rtl433/events,states=rtl_433/9b13b3f4-rtl433/states

# Uncomment the following line to also enable the default "table" output to the
# addon logs.
# output kv

# Disable TPMS sensors by default. These can cause an overwhelming number of
# devices and entities to show up in Home Assistant.
# This list is generated by running:
# rtl_433 -R help 2>&1 | grep -i tpms | sd '.*\[(\d+)\].*' 'protocol -/run.sh'
#    [59]  Steelmate TPMS
#    [60]  Schrader TPMS
#    [82]  Citroen TPMS
#    [88]  Toyota TPMS
#    [89]  Ford TPMS
#    [90]  Renault TPMS
#    [95]  Schrader TPMS EG53MA4, PA66GF35
#    [110]  PMV-107J (Toyota) TPMS
#    [123]* Jansite TPMS Model TY02S
#    [140]  Elantra2012 TPMS
#    [156]  Abarth 124 Spider TPMS
#    [168]  Schrader TPMS SMD3MA4 (Subaru)
#    [180]  Jansite TPMS Model Solar
#    [186]  Hyundai TPMS (VDO)
#    [201]  Unbranded SolarTPMS for trucks
#    [203]  Porsche Boxster/Cayman TPMS
protocol -59
protocol -60
protocol -82
protocol -88
protocol -89
protocol -90
protocol -95
protocol -110
protocol -123
protocol -140
protocol -156
protocol -168
protocol -180
protocol -186
protocol -201
protocol -203
protocol 217
protocol 214`

Log:
`s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
/tmp/rtl_433_heredoc: line 1: # This is an empty template for configuring rtl_433. mqtt information will be
# automatically added. Create multiple files ending in '.conf.template' to
# manage multiple rtl_433 radios, being sure to set the 'device' setting. The
# device must be set before mqtt output lines.
# https://github.com/merbanan/rtl_433/blob/master/conf/rtl_433.example.conf
output mqtt://${192.168.1.xx}:${1883},user=${xxx},pass=${xexseses},retain=${retain}
report_meta time:iso:usec:tz
# To keep the same topics when switching between the normal and edge versions,
# use this output line instead.
# output mqtt://${host}:${port},user=${username},pass=${password},retain=${retain},devices=rtl_433/9b13b3f4-rtl433/devices[/type][/model][/subtype][/channel][/id],events=rtl_433/9b13b3f4-rtl433/events,states=rtl_433/9b13b3f4-rtl433/states
# Uncomment the following line to also enable the default "table" output to the
# addon logs.
# output kv
# Disable TPMS sensors by default. These can cause an overwhelming number of
# devices and entities to show up in Home Assistant.
# This list is generated by running:
# rtl_433 -R help 2>&1 | grep -i tpms | sd '.*\[(\d+)\].*' 'protocol -/run.sh'
#    [59]  Steelmate TPMS
#    [60]  Schrader TPMS
#    [82]  Citroen TPMS
#    [88]  Toyota TPMS
#    [89]  Ford TPMS
#    [90]  Renault TPMS
#    [95]  Schrader TPMS EG53MA4, PA66GF35
#    [110]  PMV-107J (Toyota) TPMS
#    [123]* Jansite TPMS Model TY02S
#    [140]  Elantra2012 TPMS
#    [156]  Abarth 124 Spider TPMS
#    [168]  Schrader TPMS SMD3MA4 (Subaru)
#    [180]  Jansite TPMS Model Solar
#    [186]  Hyundai TPMS (VDO)
#    [201]  Unbranded SolarTPMS for trucks
#    [203]  Porsche Boxster/Cayman TPMS
protocol -59
protocol -60
protocol -82
protocol -88
protocol -89
protocol -90
protocol -95
protocol -110
protocol -123
protocol -140
protocol -156
protocol -168
protocol -180
protocol -186
protocol -201
protocol -203
protocol 217
protocol 214
: bad substitution
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped`

zuckschwerdt, 2023-06-16:
The HEREDOC substitution fails. Does your password maybe contain special characters like e.g. `$` or `@`?
Try to review why that configuration template doesn't work. Maybe trim it down to test things.

djmalin, 2023-06-16:
> The HEREDOC substitution fails. Does your password maybe contain special characters like e.g. `$` or `@`?
> Try to review why that configuration template doesn't work. Maybe trim it down to test things.

my password only consists of letters and numbers

closed 2023-06-16 by zuckschwerdt