merbanan/rtl_433 #2541: Security+ 2.0 Decoder not working

dasrue, 2023-06-22:
I'm having trouble getting the Security+2.0 decoder to work with my remotes. In the src/devices/secplus_v2.c file I have used that information to build a flex decoder, and run it using `-g 1 -X "n=vI3,m=OOK_PCM,s=230,l=230,t=40,r=10000,g=7400"`. 
This gets me codes from the remotes as multiple messages each with only 1 row. If I combine 2 messages together to 2 rows and give it to the decoder using the -y option it works. I have tried increasing the reset time to 1 second in the flex decoder, but it still always wants to decode as 2 messages. Any ideas why this would be happening?

zuckschwerdt, 2023-06-22:
Can you check a capture on the actual distances between packets? It should be only about 500 µs

dasrue, 2023-06-22:
Using `-M time:usec` with the same -X decoder shows that there is just over 100ms between the packets

zuckschwerdt, 2023-06-22:
Capture a sample with either `-S known` or save a longer stretch with `-w myfile_250k_433.92M.cu8 -T 20`
Then look at the timings in https://triq.org/pdv/

dasrue, 2023-06-22:
The output from PDV suggests a gap of 10004.0us. If I take the decoded bits from PDV and put them into the security+2.0 decoder using -y it works.
The output I get from PDV is

| Pulses  | 132× 223.9 ±24.0 µs | 27× 472.7 ±20.0 µs |                    |                      |
|---------|---------------------|--------------------|--------------------|----------------------|
| Gaps    | 129× 273.1 ±22.0 µs | 27× 520.7 ±20.0 µs | 2× 10004.0 ±0.0 µs |                      |
| Periods | 110× 497.0 ±18.0 µs | 39× 745.1 ±16.0 µs | 7× 994.3 ±14.0 µs  | 2× 10352.0 ±140.0 µs |
| Timings | 260× 248.4 ±46.0 µs | 54× 496.7 ±48.0 µs | 1× 196.0 ±0.0 µs   | 3× 10004.0 ±0.0 µs   |

zuckschwerdt, 2023-06-22:
What you'd want would be `-X "n=vI3,m=OOK_PCM,s=230,l=230,t=40,r=12000,g=900"` but I guess that's beyond what rtl_433 expects as sensible timing values.

dasrue, 2023-06-23:
I'm still getting split up packets using that decoder. I have sent you an email with my capture if that helps anything

zuckschwerdt, 2023-06-23:
The packet distances I see are ~70 ms, i.e. 70000 µs, we would need a reset limit of e.g. `r=71000` -- generally around 10000 is the maximum, it won't work.

Assembling packets that far spaced is not something rtl_433 can do. You would need to capture them one by one and then parse in a second step. Sorry :/

dasrue, 2023-06-23:
Ok, I have had a play with it. Based on `src/devices/blueline.c` I have added a context to the decoder which saves the data halves between invocations of the decoder with a timestamp. This seems to fix the issue, but I'm not sure if that would be an acceptable fix? My change is [here](https://github.com/dasrue/rtl_433_secplusfix/commit/174b18c51701e36851f5949825221e313f777e6e)

zuckschwerdt, 2023-06-24:
Looks good, but we really want decoders to be stateless. Open a PR so it can help others, but we won't merge right now.

gdt, 2023-10-15:
This is part of #2641.

@dasrue It would be helpful to have a PR that adds the stateful processing for this decoder, even if we don't merge it because of doctrine for now.  Having a concrete example to think about would be useful, and an open PR is usable by other people.

gdt, 2023-10-15:
I dropped `discuss` because that's what #2641 is about.

gdt, 2023-10-17:
@dasrue Can you explain how this is different from #2314?   Is this really a dup of that, too?

gdt, 2024-06-04:
Dup of PR #2314

closed 2024-06-04 by gdt