merbanan/rtl_433 #2542: rtl_433 -h or rtl_433 -R help  prints to stderr instead stdout

devZer0, 2023-06-24:
hello, 

i found that the help page(s) of rtl_433 prints to stderr by default.

is this intentional ?

normally tools print to stdout what's not considered an error

russinnes, 2023-09-25:
By piping messages for the user to stderr, the "real" output can stay with stdout, allowing it do be piped ( | ) easily into another program for further processing.

gdt, 2023-09-25:
Printing the output of -h/-R to stderr is a bug.  When the user asks for it, it belongs on stdout and the program should exit with status 0.   When it is printed because of bad or missing aguments on the command line, it should go to stderr and the program should exit with a non-zero status.

zuckschwerdt, 2023-09-25:
Help to stdout might be expected, I see one doing `rtl_433 -R | grep Breser`. But we also output help in error cases where stdout might be used otherwise, e.g. `rtl_433 -R 999 -w OOK:-`
This seems though to get right.

gdt, 2023-09-25:
As I understand the norms, if it is printed because of an error, leading to non-zero exit, stderr is appropriate.  If it is printed because the user asked for it to be printed, and this does not lead to non-zero exit, stdout is appropriate.

If the user asks for help output and something else, then either:
  - they both go on stdout; they get what they want, or what they deserve :-(
  - asking for help and something else should be an error.  I lean to this.  This would be like any other invalid args.

zuckschwerdt, 2023-09-25:
The trouble here is that we'd need to fit the chain of print functions with an additional flag where it should go. Not great. But basically we could change all those print functions to be something more like fprint and take the fd as first arg.

commit b7f0b405 (2023-10-02): Change help texts to print to stdout (closes #2542)
commit e2f8d627 (2023-10-02): Change help texts to print to stdout (closes #2542)