merbanan/rtl_433 #2543: protocol information missing in output

devZer0, 2023-06-24:
shouldn't there protocol version/number information printed for EVERY datagram being decoded  when using -M protocol switch ?

i wonder why it's missing for 1 out of 4 sensors (Fineoffset-WH2 - see below)

i would expect an "18" here.

# ./rtl_433 -F csv -M protocol
rtl_433 version nightly branch  at 202304272301 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/root/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Use "-F log" if you want any messages, warnings, and errors in the console.
time,msg,codes,protocol,model,button,id,channel,battery_ok,temperature_C,<snipp>
Found Fitipower FC0012 tuner
Exact sample rate is: 250000.000414 Hz
Allocating 15 zero-copy buffers
2023-06-24 17:31:07,,,,Fineoffset-WH2,,255,,,30.200,CRC,,36,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
2023-06-24 17:31:27,,,31,TFA-TwinPlus,,29,1,1,27.500,CHECKSUM,,44,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,

zuckschwerdt, 2023-06-24:
Only for the "plain" protocols. If there is any state or setup (like the `no-wh5`)  then it won't get a number.

devZer0, 2023-06-24:
sorry, but i don't understand.

please understand that i ask for this from an end user perspective with very few insights into the code.

WH2 is protocol 18 and i can add -R 18 for filtering, so why is it hidden ?

zuckschwerdt, 2023-06-25:
Fine Offset WH2 has quirks, it is stateful and not just a protocol. There is no "the protocol 18" every `-R 18` is a new stateful decoder and those have no fixed number.

gdt, 2023-09-26:
So are we going to:
  - close this as invalid
  - adjust the docs to explain
  - accept that if you enable a decoder by -R 18 ,then it is protocol 18, by definition, and include it, even though that doe not fully specify the decoder?
  - introduce a sub-protocol that is used to disambiguate the previous?

(The plan of leaving the issue open is not on the above list!)

zuckschwerdt, 2023-10-21:
The proper solution would be to split off Rosenborg-66796 as it's own decoder and bump down the Fineoffset-WH5 priority. Decoders would be static then. And flex decoders never get a fixed protocol number.

gdt, 2024-06-04:
I'm calling this a bug; IMHO decoders should have protocol numbers, and people should be able to enable/disable them.  That may mean adjusting the semantics of what a protocol number is, or some other cleanup.