merbanan/rtl_433 #2544: RFE:  add protocol number per default for easier matching device/protocol from list

devZer0, 2023-06-24:
wouldn't it be helpful to print protocol numerical value per default or when "-v" is in place ?

it's really hard to add a "-R <device>" filter, as the device/protocol-number needs to be found in the large list of devices and the model name printed in rtl_433 output does not always easily match the device/model name in -R help list. 

for example "Fineoffset-WH2" is not found in that help list, as it's "[18]  Fine Offset Electronics, WH2,"

i have often struggled with this and there were devices i needed to search for long for the protocol version numer.

i and wondered, how i could match this easier - and eventually came across "-M protocol" switch - but that also was not easy to find. 

i guess this would be very helpful to unexperienced users or newbies

gdt, 2023-09-26:
What is helpful to one is clutter to another.   The usage text clearly shows -M, which lists protocol.

I would suggest that you read all the docs and figure out where would be good to point this out, where it's already talking about how to disable everything but the decoders you need, and submit a PR.   It's likely just a small parenthetical comment to give the hint, like  "(Use -M protocol to see what protocol a decode is from.)", which should be enough to point someone to read the -M documentation.  (I have a bias to optimize for people that read the documentation, and to avoid clutter to put info everywhere to help people that don't.)

I'll leave this open for a while in case others think that we should default -M protocol to on.

gdt, 2024-06-04:
it has been "a while" since september.

closed 2024-06-04 by gdt