merbanan/rtl_433 #2545: Docker Issues

gcormier, 2023-06-26:
First, I know this doesn't really belong here at all, but I'm totally stumped and I believe @hertzg is as well.

https://github.com/hertzg/rtl_433_docker/issues/83

In short, rtl_433 works perfectly fine - but when running inside a container, the blueline sensor stops working, while another sensor works.

This means
- [x] MQTT is working fine
- [x] USB device is exposed/working fine
- [x] Docker is working fine

All I can think of is some sort of gain setting or other environment configuration that the docker environment is not picking up versus a regular Ubuntu environment. I'm reaching in the dark for this issue.

Any thoughts from the experts here on additional troubleshooting steps would be appreciated.

zuckschwerdt, 2023-06-26:
That sounds really strange. The blueline decoder is complicated (and stateful) but you should see the same results wether it's run directly or in a container.

You can:
- check versions on both with `-V`
- enable error log (`-F log`) and/or raise verbosity (`-vv`)
- daisy chain both rtl_433, e.g. `-F rtl_tcp` on the container, then `-d rtl_tcp` outside. (or maybe the other way around)
- grab samples (`-S all`) and inspect if the transmissions are there and can be decoded.
- inspect the live waterfall with `-F rtl_tcp` and some rtl_tcp enabled UI.

gcormier, 2023-06-26:
The output of this command differs

`./rtl_433 -R 176:12080 -R 40 -vv`

## Container "hertzg/rtl_433:nightly-debian"
```
greg@elvis:~/bay$ docker compose up rtl433
[+] Building 0.0s (0/0)                                                                         
[+] Running 1/0
 ✔ Container rtl433  Created                                                               0.0s 
Attaching to rtl433
rtl433  | rtl_433 version 22.11 (2022-11-19) inputs file rtl_tcp RTL-SDR with TLS
rtl433  | Use -h for usage help and see https://triq.org/ for documentation.
rtl433  | Trying conf file at "rtl_433.conf"...
rtl433  | Trying conf file at "/root/.config/rtl_433/rtl_433.conf"...
rtl433  | Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
rtl433  | Trying conf file at "/etc/rtl_433/rtl_433.conf"...
rtl433  | [Protocols] Registered 2 out of 245 device decoding protocols
rtl433  | [Input] The internals of input handling changed, read about and report problems on PR #1978
rtl433  | [SDR] Found 1 device(s)
rtl433  | [SDR] trying device 0: Realtek, RTL2838UHIDIR, SN: 00000001
rtl433  | Detached kernel driver
rtl433  | Found Rafael Micro R820T/2 tuner
rtl433  | [SDR] Using device 0: Realtek, RTL2838UHIDIR, SN: 00000001, "Generic RTL2832U OEM"
rtl433  | Exact sample rate is: 250000.000414 Hz
rtl433  | [SDR] Sample rate set to 250000 S/s.
rtl433  | [Input] Bit detection level set to 0.0 (Auto).
rtl433  | [SDR] Tuner gain set to Auto.
rtl433  | [Input] Reading samples in async mode...
rtl433  | [SDR] rtlsdr_set_center_freq 433920000 = 0
rtl433  | [SDR] Tuned to 433.920MHz.
rtl433  | [acquire_thread] acquire_thread enter...
rtl433  | Allocating 15 zero-copy buffers
^CGracefully stopping... (press Ctrl+C again to force)
Aborting on container exit...
[+] Stopping 1/1
 ✔ Container rtl433  Stopped
```

## Local build pulled June 26
```
greg@elvis:~/rtl_433/build/src$ ./rtl_433 -R 176:12080 -R 40 -vv
rtl_433 version nightly-4-g767e5387 branch master at 202306081919 inputs file rtl_tcp RTL-SDR SoapySDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/greg/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
[Protocols] Registered 2 out of 245 device decoding protocols
[Input] The internals of input handling changed, read about and report problems on PR #1978
[SDR] Found 1 device(s)
[SDR] trying device 0: Realtek, RTL2838UHIDIR, SN: 00000001
Detached kernel driver
Found Rafael Micro R820T tuner
[SDR] Using device 0: Realtek, RTL2838UHIDIR, SN: 00000001, "Generic RTL2832U OEM"
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
[SDR] Sample rate set to 250000 S/s.
[Input] Bit detection level set to 0.0 (Auto).
[SDR] Tuner gain set to Auto.
[Input] Reading samples in async mode...
[SDR] Tuned to 433.920MHz.
Allocating 15 zero-copy buffers
^CSignal caught, exiting!
[rtl_433] stopping...
```

The container has `[SDR] rtlsdr_set_center_freq 433920000 = 0` in the log, the local copy does not.

So I guess I'm suspecting the container itself now and the build version.

Note that `-V` isn't useful in the container unfortunately - https://github.com/hertzg/rtl_433_docker/issues/69#issuecomment-1566076916

gcormier, 2023-06-26:
```
greg@elvis:~/bay$ docker run --device /dev/bus/usb hertzg/rtl_433 -R 176:12080 -v
rtl_433 version 22.11 (2022-11-19) inputs file rtl_tcp RTL-SDR with TLS
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/root/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 1 out of 223 device decoding protocols
Found 1 device(s)

trying device  0:  Realtek, RTL2838UHIDIR, SN: 00000001
Detached kernel driver
Found Rafael Micro R820T/2 tuner
Using device 0: Generic RTL2832U OEM
Exact sample rate is: 250000.000414 Hz
Sample rate set to 250000 S/s.
Bit detection level set to 0.0 (Auto).
Tuner gain set to Auto.
Reading samples in async mode...
Tuned to 433.920MHz.
Allocating 15 zero-copy buffers
Pulse data: 183 pulses
[  0] Pulse:  159, Gap:  147, Period:  306
[  1] Pulse:  156, Gap:  148, Period:  304
[  2] Pulse:  155, Gap:  148, Period:  303
[  3] Pulse:  156, Gap:  147, Period:  303
[  4] Pulse:  104, Gap:   50, Period:  154
[  5] Pulse:  103, Gap:   51, Period:  154
[  6] Pulse:   56, Gap:   97, Period:  153
[  7] Pulse:  103, Gap:   51, Period:  154
[  8] Pulse:   56, Gap:   97, Period:  153
[  9] Pulse:   57, Gap:   97, Period:  154
[ 10] Pulse:   56, Gap:   97, Period:  153
[ 11] Pulse:   57, Gap:   97, Period:  154
[ 12] Pulse:   56, Gap:   97, Period:  153
[ 13] Pulse:   57, Gap:   97, Period:  154
[ 14] Pulse:  103, Gap:   50, Period:  153
[ 15] Pulse:   57, Gap:   97, Period:  154
[ 16] Pulse:   56, Gap:   98, Period:  154
[ 17] Pulse:   56, Gap:   97, Period:  153
[ 18] Pulse:  103, Gap:   50, Period:  153
[ 19] Pulse:  104, Gap:   50, Period:  154
[ 20] Pulse:   56, Gap:   97, Period:  153
[ 21] Pulse:  104, Gap:   50, Period:  154
[ 22] Pulse:   56, Gap:   97, Period:  153
[ 23] Pulse:   57, Gap:   97, Period:  154
[ 24] Pulse:   56, Gap:   97, Period:  153
[ 25] Pulse:  103, Gap:   51, Period:  154
[ 26] Pulse:   57, Gap:   96, Period:  153
[ 27] Pulse:   57, Gap:   97, Period:  154
[ 28] Pulse:  103, Gap:   51, Period:  154
[ 29] Pulse:  103, Gap:   50, Period:  153
[ 30] Pulse:   56, Gap:   97, Period:  153
[ 31] Pulse:  104, Gap:   50, Period:  154
[ 32] Pulse:   57, Gap:   96, Period:  153
[ 33] Pulse:   57, Gap:   97, Period:  154
[ 34] Pulse:  103, Gap:   51, Period:  154
[ 35] Pulse:   56, Gap:   97, Period:  153
[ 36] Pulse:   57, Gap:   97, Period:  154
[ 37] Pulse:   56, Gap:   97, Period:  153
[ 38] Pulse:   56, Gap:   98, Period:  154
[ 39] Pulse:   56, Gap:   97, Period:  153
[ 40] Pulse:  103, Gap:   50, Period:  153
[ 41] Pulse:   57, Gap:   97, Period:  154
[ 42] Pulse:   56, Gap:   97, Period:  153
[ 43] Pulse:  104, Gap:   50, Period:  154
[ 44] Pulse:   57, Gap:   97, Period:  154
[ 45] Pulse:  103, Gap:   50, Period:  153
[ 46] Pulse:   57, Gap:   97, Period:  154
[... 488 lines cut ...]
[162] Pulse:  103, Gap:   50, Period:  153
[163] Pulse:   56, Gap:   98, Period:  154
[164] Pulse:   56, Gap:   97, Period:  153
[165] Pulse:  103, Gap:   51, Period:  154
[166] Pulse:   56, Gap:   97, Period:  153
[167] Pulse:  104, Gap:   50, Period:  154
[168] Pulse:   56, Gap:   98, Period:  154
[169] Pulse:   55, Gap:   98, Period:  153
[170] Pulse:   56, Gap:   97, Period:  153
[171] Pulse:  103, Gap:   51, Period:  154
[172] Pulse:   56, Gap:   98, Period:  154
[173] Pulse:   56, Gap:   97, Period:  153
[174] Pulse:   56, Gap:  103, Period:  159
[175] Pulse:  104, Gap:   50, Period:  154
[176] Pulse:   56, Gap:   97, Period:  153
[177] Pulse:  103, Gap:   51, Period:  154
[178] Pulse:   56, Gap:   98, Period:  154
[179] Pulse:  103, Gap:   50, Period:  153
[180] Pulse:  103, Gap:   51, Period:  154
[181] Pulse:   56, Gap:   97, Period:  153
[182] Pulse:   56, Gap: 2501, Period: 2557
Pulse data: 1 pulses
[  0] Pulse:   62, Gap: 2501, Period: 2563
^CSignal caught, exiting!
```

gcormier, 2023-06-27:
Here is the output of `rtl_433 -X "name=blueline,modulation=OOK_MC_ZEROBIT,short=500,long=1000,r=58000"` locally and in the container.

[container.txt](https://github.com/merbanan/rtl_433/files/11885191/container.txt)

[local.txt](https://github.com/merbanan/rtl_433/files/11885192/local.txt)

gcormier, 2023-06-27:
Captured data on host

`rtl_433 -w rtl433data`

Then
```
greg@elvis:~/rtl_433/build/src$ docker run -v ./:/greg hertzg/rtl_433 -r /greg/rtl433data -R 176:12080
rtl_433 version 22.11 (2022-11-19) inputs file rtl_tcp RTL-SDR with TLS
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/root/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 1 out of 223 device decoding protocols [ ]
Test mode active. Reading samples from file: /greg/rtl433data
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 

time      : @19.285257s
model     : Blueline-PowerCost
id        : 12080
gap       : 4757
Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 

time      : @19.389055s
model     : Blueline-PowerCost
id        : 12080
gap       : 4757
Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 

time      : @19.492855s
model     : Blueline-PowerCost
id        : 12080
gap       : 4757
Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 

time      : @51.082691s
model     : Blueline-PowerCost
id        : 12080
flags     : 1f
Battery   : 1
temperature_C: 25.884
Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 

time      : @51.186481s
model     : Blueline-PowerCost
id        : 12080
flags     : 1f
Battery   : 1
temperature_C: 25.884
Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 

time      : @51.290276s
model     : Blueline-PowerCost
id        : 12080
gap       : 4281
Integrity : CRC
```

gcormier, 2023-06-27:
## Capture data inside container
`docker run --device /dev/bus/usb -v ./:/greg hertzg/rtl_433 -w /greg/rtl433containerdata`

## Analyze data on host
Nothing found for blueline. Accurite data was found.

```
greg@elvis:~/rtl_433/build/src$ rtl_433 -r rtl433containerdata -R 176:12080
rtl_433 version 21.05-137-g0097048a branch master at 202111141542 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/greg/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 1 out of 200 device decoding protocols [ ]
Test mode active. Reading samples from file: rtl433containerdata
greg@elvis:~/rtl_433/build/src$
```

```
greg@elvis:~/rtl_433/build/src$ rtl_433 -r rtl433containerdata 
rtl_433 version 21.05-137-g0097048a branch master at 202111141542 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/greg/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 169 out of 200 device decoding protocols [ 1-4 8 11-12 15-17 19-23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 177-197 199 ]
Test mode active. Reading samples from file: rtl433containerdata
baseband_demod_FM: low pass filter for 250000 Hz at cutoff 25000 Hz, 40.0 us
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : @8.898400s
model     : Acurite-Tower id        : 4131
channel   : A            Battery   : 1             Temperature: 25.3 C
Humidity  : 70 %         Integrity : CHECKSUM
... etc ...
```

hertzg, 2023-06-27:
Looking at potential differences it's could be the `librtlsdr`. It's a longshot but could you try older versions of `alpine`, could be worth checking debian one as well 

| Image | librtlsdr version|
| --- | --- |
| hertzg/rtl_433:latest-alpine-3.18 | [0.8.0-r0](https://pkgs.alpinelinux.org/package/v3.18/main/x86_64/librtlsdr) |
| hertzg/rtl_433:latest-alpine-3.17 |  [0.6.0-r3](https://pkgs.alpinelinux.org/package/v3.17/main/x86_64/librtlsdr) |
|hertzg/rtl_433:latest-debian-bookworm |  [0.6.0-4](https://packages.debian.org/bookworm/librtlsdr-dev) |
|hertzg/rtl_433:latest-debian-bullseye | [0.6.0-3](https://packages.debian.org/bullseye/librtlsdr-dev) |

gcormier, 2023-06-27:
BINGO.

`hertzg/rtl_433:latest-alpine-3.17` is working.

gcormier, 2023-06-27:
Ubuntu 20.04 LTS and alpine-3.17 use librtlsdr 0.6.0, works fine.

`hertzg/rtl_433:latest` is using 0.8.0, does not work with blueline.

gdt, 2023-09-26:
This seems like a support request and not a bug report.   If it turns out that librtlsdr 0.8.0 fails on bare metal, please open a new issue.

closed 2023-09-26 by gdt

zuckschwerdt, 2023-10-04:
Related to #2570