merbanan/rtl_433 #2550: NATS support

xhci, 2023-06-30:
Is [NATS](https://nats.io) support something that would be desirable for RTL 433? I see that there is InfluxDB and MQTT support currently.

I would be willing to create a PR adding this if anyone is interested.

gdt, 2023-09-26:
It seems unlikely as it does not seem to be a mainstream approach in use by the people that make up the rtL433 community -- I had never heard of it before, as a random data point.   Certainly a python script `examples/rtl_433_nas.py` that reads syslog and sends data would be reasonable to merge, as it wouldn't add complexity to anyone not using this.

closed 2023-09-26 by gdt

xhci, 2023-09-26:
Thanks for the reply and some direction on this. Based on your feedback, I think a Python script would be a great way to go for this.

I will try to devote some time to that soon.