merbanan/rtl_433 #2551: rtl_433 crashes after sveral days

ifuchs, 2023-07-03:
I have been running the following command:
`rtl_433 -D restart -F "mqtt://192.168.3.136:1883,user=homeassistant,pass=geeCheemoh7uhaiz8uizaisiexahiem0ESoa5leemae8Pheighah1echaht6eic0,retain=1"
rtl_433 version nightly-33-gf775d1ac branch master at 202304221552 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/pi/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
MQTT: Publishing MQTT data to 192.168.3.136 port 1883
MQTT: Publishing device info to MQTT topic "rtl_433/raspberrypi/devices[/type][/model][/subtype][/channel][/id]".
MQTT: Publishing events info to MQTT topic "rtl_433/raspberrypi/events".
MQTT: Publishing states info to MQTT topic "rtl_433/raspberrypi/states".
Use "-F log" if you want any messages, warnings, and errors in the console.
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Allocating 15 zero-copy buffers
usb_claim_interface error -6
`
It invariably crashes after several days and can be restarted without any other action. Why is this crashing and how can I keep it running?

zuckschwerdt, 2023-07-06:
libusb error `6` is `LIBUSB_ERROR_BUSY`. This might indicate that the RTL-SDR dongle is stuck somehow. Perhaps an undervoltage glitch or maybe overtemperature or just bad luck?

fallenpegasus, 2023-09-17:
Most SDRs glitch every few days, more often when they get hot.  And libusb is brittle.  And rtl_433 doesnt try to recover from libusb errrors, because how could it?

I just run my rtl_433 under the control of supervisord.

ifuchs, 2023-09-17:
I've not used supervisord. My RPI0w command is:

rtl_433 -D restart -F "mqtt://192.168.1.115:1883,user=homeassistant,pass=XXX,retain=1";pushover-cli "rtl433 stopped on Pi0W".

This runs rtl_4433 and sends me a notification if the command ends. Unfortunately, sometimes it stops working and writes an error message to the console but does not end so I never get a notification. Two questions, if I use supervisord, do I just put this complete command in to the .conf file and second will it be any better at detecting when rtl_433 has ceased working but has not ended?

zuckschwerdt, 2023-09-17:
There is a subtle bug when shutting down (the not working) receiver and rtl_433 can get stuck before reaching the `-D restart` action. I still have to debug that or rewrite the logic :/

gdt, 2024-06-04:
Referenced bug fixed, and no recent test with up-to-date code.  Please feel free to add evidence that there is still a bug and we can reopen.

closed 2024-06-04 by gdt