merbanan/rtl_433 #2552: Capture frequency accuracy

frevoli, 2023-07-04:
Hello community,

When I run a capture centred at 433.5MHz with a 1000KHz bandwidth I am able to save data to an ook file which is received between 433-434MHz.

If I transmit with a known source for example at 433.9MHz, this is written to the ook file and the output specifies the centre frequency as 433.5 which is correct, but it always gets the received frequency wrong. It generally shows a figure around the centre frequency.

How can I get the actual frequency of the transmitter from the ook file?

Thanks in advance!
Matt

ProfBoc75, 2023-07-04:
Hi @frevoli 

try with "-M level" and you should see the frequency.

[image]

ASK = OOK

frevoli, 2023-07-04:
Thanks Prof!
I'll give that a go, and report back.
Cheers

frevoli, 2023-07-04:
No change I'm afraid.  

If I run...
"rtl_433 -f 433.5M -R 0 -M level -w test.ook"

Then I transmit a burst of data on 433.7MHz.
When I view the contents of "test.ook" it says...

;freq1 433519008
;centerfreq 433500000
;sample rate 250000 Hz

Freq1 is not correct. It should surely say 433.7MHz.

Or am I misunderstanding how to interpret an ook file?

Thanks again.
Matt

ProfBoc75, 2023-07-04:
You have the good interpret, the freq1 is the ook signal you received and the centerfreq is the dongle frequency you have with the -f 433.5M , notice that into the same ook file, you may have other freq1 if you received several ook signals. 2 lines above freq1 you should have :
;received xxx_timestamp_xxx
;ook zzz pulses
;freq1 ....
;centerfreq ....  hz
;samplerate .... hz
.. db
.. db 
.../...
; pulse_duration gap_duration 
; pulse_duration gap_duration ... 
.../...
;end
;received  ... next timestamp ...
;ook ... 

So between "received" and "end" this is one burst of data.

I tested from my side with a Nexus-TH temp/hum sensor , ook modulation, depends on the sample rate at 250K it's reported freq1 at 434 MHz and using sample rate at 1000K it's reported freq1 434.2 MHz, with the same id also, so I know this is the same sensor properly decoded by rtl_433, so strange gap between the metrics and related to sample rate ...

frevoli, 2023-07-05:
Hello, yes I have the other output like you...

received xxx_timestamp_xxx
;ook zzz pulses
;freq1 ....
;centerfreq .... hz
;samplerate .... hz
.. db
.. db
.../...
; pulse_duration gap_duration
; pulse_duration gap_duration ...
.../...
;end
;received ... next timestamp ...
;ook ...

I just don't know why my freq1 value does not reflect the true tx frequency.

I don't know how to fix this. It is important for my application.

Any suggestions would be greatly appreciated.

zuckschwerdt, 2023-07-05:
The frequency seen is inaccurate and just a rough estimate based on the instant phase angle.
We have plans to someday optionally add FFT if your compute has the power for that and you want more accuracy.

frevoli, 2023-07-05:
Christian, is there any way I can get the true frequency of the source from the ook file?

The true frequency seems to be present when viewed in a terminal using -M level, so presumably this is possible?

Thanks again

zuckschwerdt, 2023-07-05:
Just to confirm: you see different frequency values in the terminal (KV output) and in the .ook? Can you upload a sample file zip where this happens?

frevoli, 2023-07-05:
This is ok, but can you help me with any solution how I can get the true frequency of the source?

Thank you

zuckschwerdt, 2023-07-05:
Analyze the .cu8 file with an FFT an pick the dominant bucket. Or drop on https://triq.org/pdv/ and eyeball the frequency.

frevoli, 2023-07-05:
Thank you.

Generating cu8 files will be an additional memory overhead for my application, but I'll give that a try.

Cheers!

gdt, 2023-07-05:
How did you calibrate the dongle?  Some are very close,  but two of mine are 53 and 71 ppm off.

frevoli, 2023-07-05:
I didn't calibrate the dongle. How do I do this?

frevoli, 2023-07-05:
> Analyze the .cu8 file with an FFT an pick the dominant bucket.

Christian, could you advise me how to specifically carry this out?

Thanks again

zuckschwerdt, 2023-07-05:
It's a programming task. Pick any FFT demo you like, e.g. https://www.fftw.org/fftw3_doc/Complex-One_002dDimensional-DFTs.html#Complex-One_002dDimensional-DFTs

jithivk, 2023-09-06:
How can I capture signal between a frequency band, if I want to capture active signals between 450M and 460M is that possible

zuckschwerdt, 2023-09-06:
> How can I capture signal between a frequency band, if I want to capture active signals between 450M and 460M is that possible

No, with RTL-SDR you usually go for 1 MHz bands. Use `rtl_433 -s 1024k -f 450M` then 451M and so on.

jithivk, 2023-09-06:
Any other app that can capture and record iq data from that band

jithivk, 2023-09-06:
> You can use rtl-sdr to do it also. But rtl_433 supports frequency hopping
> and signal recording when a signal is detected. So if you combine -s 2000k
> with -H and -S unknown you can continuously scan the range and then save
> recordings of when a transmission is detected. The alternative is to get
> hardware that can record the whole range.

Can I use other sdr like LimeSdr with larger bandwidth to record active
signals especially fsk signal have any recommendations using gnu radio

gdt, 2023-09-24:
This is no longer a bug report.

closed 2023-09-24 by gdt