merbanan/rtl_433 #2554: Home Assistant auto discovery broken due to bad topic subscription

mill1000, 2023-07-06:
Looks like the Home Assistant auto discovery (rtl_433_mqtt_hass.py) is broken. Possibly introduced by #2289.

The topic subscriptions that are created for the sensors are missing the proper topic prefix so no values are received.
New subscription
[image]

Previous subscription
[image]

mill1000, 2023-07-06:
Here's a working config using the current main.

```
> rtl_433_mqtt_hass.py -d -T rtl_433/devices[/type][/model][/subtype][/channel][/id]
```

rtl_433
```
> rtl_433 -C si -F "mqtt://10.100.1.6:1883,devices=rtl_433/devices[/type][/model][/subtype][/channel][/id],events=rtl_433/test/events,states=rtl_433/test/states" -F log -F kv -M level
```

Not ideal since I had to put an extra level in the events and states topics

zuckschwerdt, 2023-07-06:
@unverbraucht can you check this please?

unverbraucht, 2023-07-29:
Looking into it.

unverbraucht, 2023-08-11:
I could reproduce the issue, looking into it now.

PR #2593 (2023-08-11): Fix #2554 (HA topics broken)
This fixes #2554 . In the original code I had not considered that the command line arg is a topic suffix, so we need to prefix the `/rtl_433/HOSTNAME`.
closed 2023-08-11 by zuckschwerdt