merbanan/rtl_433 #2562: Add additional sensors to Cavius

sindalschmidt, 2023-07-12:
# Adding Door/Window sensor to Cavius decoder
I have submitted a pull request to add sample test files that I have captured. I have also added them here: [Cavius-door-sensor-captures.zip](https://github.com/merbanan/rtl_433/files/12031304/Cavius-door-sensor-captures.zip)

Currently the Cavius decoder only supports Smoke, Heat and Water alarms. The Cavius lineup also has door/window sensors. I have attempted to modify the cavius.c decoder myself without luck and are now looking for help.

In the files attached I have made 2 captures; one when opening the door and one when closing the door. If someone can point me in the right direction, I will be able to modify the cavius.c and test right away locally. Any help is much appreciated, as I have no experience with radio signals.

ProfBoc75, 2023-07-12:
Hi @sindalschmidt 

Your 2 captures can't be decoded, the signal is too strong.
Can you try again, please. reduce the gain and/or not too close to the sensor, at 1 meter not less, and/or remove the antenna from the rtl dongle.

But for what I can see using Pulseview, it's not a cavius coding. too small message compare to existing capture into rtl_433 test.

Modulation is still FSK_PCM but pulse is around 420 µs instead of 206 µs.

Both Open and Close are distinct signals with different values.

SteveCooling, 2023-07-13:
> @SteveCooling are you able to assist here? I really appreciate your initial commit of supporting Cavius smoke/heat/water alarms. I rely on it. Thank you!

Sorry, there's nothing I can do at this point. Glad my contribution is working for you, though.

sindalschmidt, 2023-07-13:
> Can you try again, please.

Thank you for assisting me! I have made two new captures with the antenna removed.
[Cavius-door-sensor-captures2.zip](https://github.com/merbanan/rtl_433/files/12039161/Cavius-door-sensor-captures2.zip)

According to the cavius.c file, the Cavius radio signals are Manchester encoded.

> try this :
> 
> ```
> rtl_433 -F 868.68M -X "n=cavius,m=FSK_PCM,s=415,l=415,r=1500"
> ```

I do get some output, but it doesn't seem to line up with me opening and closing the door.

```
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2023-07-13 14:08:15
model     : cavius       count     : 1             num_rows  : 1             rows      :
len       : 1            data      : 8
codes     : {1}8
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2023-07-13 14:08:29
model     : cavius       count     : 1             num_rows  : 1             rows      :
len       : 4            data      : f
codes     : {4}f
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2023-07-13 14:08:37
model     : cavius       count     : 1             num_rows  : 1             rows      :
len       : 1            data      : 8
codes     : {1}8
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2023-07-13 14:08:41
model     : cavius       count     : 1             num_rows  : 1             rows      :
len       : 12           data      : fff
codes     : {12}fff
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2023-07-13 14:08:52
model     : cavius       count     : 1             num_rows  : 1             rows      :
len       : 1            data      : 8
codes     : {1}8
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2023-07-13 14:08:52
model     : cavius       count     : 1             num_rows  : 1             rows      :
len       : 1            data      : 8
codes     : {1}8
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2023-07-13 14:08:53
model     : cavius       count     : 1             num_rows  : 1             rows      :
len       : 1            data      : 8
codes     : {1}8
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
```

Thanks again for helping!

SteveCooling, 2023-07-13:
Just a quick note, It might well be that the door sensors are using a completely different packet format. I don't think Cavius would want the door sensors waking up their sleeping (battery powered) smoke sensors every time you walk through a door.

sindalschmidt, 2023-07-13:
Good point, that makes sense. I hope my captures can help point us in the right direction then. I'm definitely getting larger samples when collecting upon opening and closing the door. In between there are small captures, so I can definitely tell them apart. They also line up quite well right after opening/closing the door.

zuckschwerdt, 2023-07-13:
FSK 400 µs looks about right. But the reset needs to be longer: `rtl_433 -f 868.68M -X "n=cavius,m=FSK_PCM,s=415,l=415,r=2500"`
You should see codes of 100 or more bits length so `{100}`or more with 14 or so chars code.

ProfBoc75, 2023-07-13:
Hi , 

Here, my findings:
Only 3 bytes changed between open & close, and always same bytes.
- Byte 6 and 14 don't have the same value
- But the byte 13 is 0x25 when Open and 0x24 when Closed.

  |   |   |   |   |   |   |   | x |   |   |   |   |   |   | x | x
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --
 Byte |   |00 |01 |02 |03 |04 |05 |06 |07 |08 |09 |10 |11 |12 |13 |14
 Data |   |PP |PP |PP |PP |PP |PP |UU |II |II |II |II |II |II |SS |CC
Door-sensor-OPEN_g022_868.68M_250k | open | AA | AA | AA | D3 | 15 | 27 | FF | DC | 3C | E3 | 12 | 65 | 33 | 25 | F8
Door-sensor-CLOSE_g025_868.68M_250k | close | AA | AA | AA | D3 | 15 | 27 | FE | DC | 3C | E3 | 12 | 65 | 33 | 24 | EC
  |   |   |   |   |   |   |   | x |   |   |   |   |   |   | x | x
OPEN_g031_868.68M_250k | open | AA | AA | AA | D3 | 15 | 27 | FE | DC | 3C | E3 | 12 | 65 | 33 | 25 | EB
CLOSE_g036_868.68M_250k | close | AA | AA | AA | D3 | 15 | 27 | FD | DC | 3C | E3 | 12 | 65 | 33 | 24 | D9

Data Layout :

PP: preamble{48} 6 bytes : 0xAA AA AA D3 15 27
UU:  Unknown{8} 1 byte : lower nibble changed between Open & Close
II : Unknown{48} 6 bytes : Fixed value 0xDC 3C E3 12 65 33
SS :  {8} 1 byte : State Open 0x25 / Close , 0x24 , so lower nibble changed between states.
CC: {8} 1 byte : CRC-8 from byte 6 to 14 (so without preamble)  , poly 0x07, init 0x00.

I looked to the Cavius guides, the sensor should provide also low battery signal from these byte, I can't say which one, you must test with dc supply.
So remaining some work to get the figures.

sindalschmidt, 2023-07-16:
> You should see codes of 100 or more bits length so `{100}`or more with 14 or so chars code.

Thank you for the suggestion! Unfortunately it's not as expected, but I do see changes when opening and closing the door. At 22:10:42 I open the door. At 22:10:54 I close the door. But the length is not as expected. I have tried adjusting the reset a bit without luck.
```
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2023-07-16 22:10:37
model     : cavius       count     : 1             num_rows  : 1             rows      :
len       : 3            data      : e
codes     : {3}e
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2023-07-16 22:10:40
model     : cavius       count     : 1             num_rows  : 1             rows      :
len       : 3            data      : e
codes     : {3}e
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2023-07-16 22:10:41
model     : cavius       count     : 1             num_rows  : 1             rows      :
len       : 1            data      : 0
codes     : {1}0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2023-07-16 22:10:42
model     : cavius       count     : 1             num_rows  : 1             rows      :
len       : 44           data      : fffffffffff
codes     : {44}fffffffffff
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2023-07-16 22:10:42
model     : cavius       count     : 1             num_rows  : 1             rows      :
len       : 34           data      : ffffffffc
codes     : {34}ffffffffc
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2023-07-16 22:10:46
model     : cavius       count     : 1             num_rows  : 1             rows      :
len       : 20           data      : 00000
codes     : {20}00000
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2023-07-16 22:10:48
model     : cavius       count     : 1             num_rows  : 1             rows      :
len       : 1            data      : 8
codes     : {1}8
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2023-07-16 22:10:54
model     : cavius       count     : 1             num_rows  : 1             rows      :
len       : 22           data      : fffffc
codes     : {22}fffffc
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2023-07-16 22:10:54
model     : cavius       count     : 1             num_rows  : 1             rows      :
len       : 16           data      : ffff
codes     : {16}ffff
```
Let me know if I should do more captures, if that's helpful. Thank you for taking your time with me.

sindalschmidt, 2023-07-16:
> Only 3 bytes changed between open & close, and always same bytes.

Very interesting findings! Thank you so much for your contribution. How can I implement this to capture these events; either by modifying the cavius decoder, build a new one or use a flex decoder?

sindalschmidt, 2023-07-16:
I have confirmed with a source that it is sending using FSK.

ProfBoc75, 2023-07-16:
Hi @sindalschmidt

About the captures, your cu8 files are still not good enough to be decoded by rtl_433.
I was able to decode them only manually and from Pulseview graphics of the signal.

Try to capture with sample rate at 1000k, this should be the default at 869Mhz so do not put -s 250k option, shift the frequency to 869MHz to avoid the DC offset / to correct the IQ:

```
rtl_433 -f 869M -X "n=cavius,m=FSK_PCM,s=415,l=415,r=2500" -S unknown 
```

You can play with the gain :
0 = auto , then 1 to 50 , it will depend on the RTL dongle, from +1 to +49db of gain.
```
rtl_433 -g 0 -f 869M -X "n=cavius,m=FSK_PCM,s=415,l=415,r=2500"
rtl_433 -g 20 -f 869M -X "n=cavius,m=FSK_PCM,s=415,l=415,r=2500"
```

You can play with the FSK pulse detector mode -Y :
```
rtl_433 -g 0 -f 869M -X "n=cavius,m=FSK_PCM,s=415,l=415,r=2500" -Y autolevel
rtl_433 -g 0 -f 869M -X "n=cavius,m=FSK_PCM,s=415,l=415,r=2500" -Y classic
```

About the device decoder:
- you can use the current cavius.c, add the new demodulation settings and add a new decoding part, lot of devices are working this way.
- or you can create a new device decoder. It will create a new device number.
- flex decoder is good to start, to test, but it could be nice to get the CRC 8 decoding/control and then avoid false positive.
- if flex decoder is ok for you, you can create a .conf file and get the sensor be decoded as others, but CRC will not be checked, something like this:

```
rtl_433 -g 0 -f 869M -X "n=cavius,m=FSK_PCM,s=415,l=415,r=2500,preamble=d31527,get=@104:{8}:state:[0x25:Open 0x24:Close" -Y autolevel
```

sindalschmidt, 2023-07-17:
> Try to capture with sample rate at 1000k, this should be the default at 869Mhz so do not put -s 250k option, shift the frequency to 869MHz to avoid the DC offset / to correct the IQ:

I have received confirmation from the manufacturer, that the device sends using 868.65Mhz with FSK. I therefore kept it at 868.65Mhz. Let me know if I should still use 869Mhz to get better captures for you.

The below are captures from when opening the door at around 11:42:53 which generates captures number 10 to 16. At around 11:43:08 I close the door, which seem to generate capture number 21 to 26.
[868.65M_1000k.zip](https://github.com/merbanan/rtl_433/files/12067954/868.65M_1000k.zip)

### rtl_433 -f 868.65M -p 9 -X "n=cavius,m=FSK_PCM,s=415,l=415,r=2500" -S unknown
```
[Protocols] Registered 193 out of 223 device decoding protocols [ 1-4 8 11-12 15-17 19-23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119-121 124-128 130-149 151-161 163-168 170-175 177-197 199 201-215 217-223 ]
[SDR] Found 1 device(s)
[SDR] trying device  0:  Realtek, RTL2838UHIDIR, SN: 00000001
Detached kernel driver
Found Fitipower FC0012 tuner
[SDR] Using device 0: Generic RTL2832U OEM
Exact sample rate is: 1000000.026491 Hz
Allocating 15 zero-copy buffers
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2023-07-17 11:42:43
model     : cavius       count     : 1             num_rows  : 1             rows      :
len       : 1            data      : 0
codes     : {1}0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2023-07-17 11:42:43
model     : cavius       count     : 1             num_rows  : 1             rows      :
len       : 2            data      : 0
codes     : {2}0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2023-07-17 11:42:43
model     : cavius       count     : 1             num_rows  : 1             rows      :
len       : 1            data      : 0
codes     : {1}0
*** Saving signal to file g007_868.65M_1000k.cu8 (42789 samples, 131072 bytes)
*** Saving signal to file g008_868.65M_1000k.cu8 (160958 samples, 393216 bytes)
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2023-07-17 11:42:46
model     : cavius       count     : 1             num_rows  : 1             rows      :
len       : 10           data      : 000
codes     : {10}000
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2023-07-17 11:42:46
model     : cavius       count     : 1             num_rows  : 1             rows      :
len       : 10           data      : 000
codes     : {10}000
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2023-07-17 11:42:48
model     : cavius       count     : 1             num_rows  : 1             rows      :
len       : 29           data      : 00000000
codes     : {29}00000000
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2023-07-17 11:42:48
model     : cavius       count     : 1             num_rows  : 1             rows      :
len       : 1            data      : 0
codes     : {1}0
*** Saving signal to file g009_868.65M_1000k.cu8 (160961 samples, 393216 bytes)
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2023-07-17 11:42:51
model     : cavius       count     : 1             num_rows  : 1             rows      :
len       : 27           data      : 0000000
codes     : {27}0000000
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2023-07-17 11:42:52
model     : cavius       count     : 1             num_rows  : 1             rows      :
len       : 2            data      : c
codes     : {2}c
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2023-07-17 11:42:53
[... cut ...]
codes     : {23}fffffe
*** Saving signal to file g028_868.65M_1000k.cu8 (388163 samples, 786432 bytes)
```

### Gain and pulse detector
Playing around with these unfortunately only continues to produce these meaningless (at least to me) messages already shown above. I can't seem to get messages large enough (>=100). I appreciate your suggestions and yes, I would very much like help with the expanding the cavius.c coding, once we have something to aim for here. Hopefully the new captures can give some insights.

ProfBoc75, 2023-07-17:
Yes it's better to not be at the exact frequency, to remove the dc offset that could be introduce by rtl dongle or signal.
If you use SDRSharp , you may see a spike at the tuned frequency in the middle of the screen and in the rain fall , this is the dc offset that will alter the signal.

With -f 869M, the sample rate at 1000khz will covert from 868.5 Mhz to 869.5 Mhz so including your signal at 868.65 Mhz.

I will check later your captures.

ProfBoc75, 2023-07-18:
Hi @sindalschmidt 

Back to your cu8 cap files:
- g001, 2, 4, 5, 7, 18, 26 = Noise = trash can
- g003, 6, 8, 9, 10, 17, 19, 28 = another sensor 1, not recognized by rtl_433 but could be, modulation is fsk_pcm, short=long=15, always same value {127/128}: aaaaaaaaaa84c37c76557412fe829d
- g025, 27 = another sensor 2, not recognized by rtl_433, FSK_PCM, short=long=10
- g011 -- > g016, g020, 22, 23, 24 = look like your previous captures for Cavius
- g021 is dirty with a mix of 2 signals with Cavius + other sensor 1.

About the Cavius captures, it's still clipping (too loud) and can't be decoded. 
Try more distance, remove antenna, and lower gain (try step by step, start at lower gain with -g 1, then -g 2 ... up to -g 50, but 50 will be too strong too).

```
rtl_433 -g 1 -f 869M -X "n=cavius,m=FSK_PCM,s=415,l=415,r=2500,preamble=d31527,bits>=100,get=@104:{8}:state:[0x25:Open 0x24:Close" -S unknown
rtl_433 -g 2 -f 869M -X "n=cavius,m=FSK_PCM,s=415,l=415,r=2500,preamble=d31527,bits>=100,get=@104:{8}:state:[0x25:Open 0x24:Close" -S unknown
rtl_433 -g 3 -f 869M -X "n=cavius,m=FSK_PCM,s=415,l=415,r=2500,preamble=d31527,bits>=100,get=@104:{8}:state:[0x25:Open 0x24:Close" -S unknown
```

About the clipping:
Using PulseView application to see it (need first to convert cu8 to sr file like this ```rtl_433 -r g011...cu8 -w g011...sr```, you will see the 2 information I and Q, yours , they look like square signal and should be sinusoidal, this means that it's too strong, and clipping the signal.

Clipping / square (your Cavius cu8 files): then FSK or ASK is not decoded.
Zoom in
[image]
Zoom out
[image]

Good sinusoidal signal (expected): FSK is decoded here 
Zoom in
[image]
Zoom out
[image]

User guide here: https://triq.org/rtl_433/ANALYZE.html

sindalschmidt, 2023-07-22:
Hi @ProfBoc75 

I apologize for the delay in responses. I really appreciate your help and the detailed responses. I have taken new captures with the antenna removed and the base (of the antenna) hanging down from my metal cabinet, as well as capturing with rtl_433 with a gain of -20.

I have installed the software PulseView and from what I can see of these captures, the I and Q are not clipping with these captures: 
[g112-g142.zip](https://github.com/merbanan/rtl_433/files/12137159/g112-g142.zip)

```
rtl_433 -g -20 -f 869M -X "n=cavius,m=FSK_PCM,s=415,l=415,r=2500,preamble=d31527,bits>=100,get=@104:{8}:state:[0x25:Open 0x24:Close" -S unknown
rtl_433 version 22.11-34-gb605a3d2 branch master at 202212302231 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/thomas/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...

New defaults active, use "-Y classic -s 250k" for the old defaults!

[Protocols] Registered 193 out of 223 device decoding protocols [ 1-4 8 11-12 15-17 19-23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119-121 124-128 130-149 151-161 163-168 170-175 177-197 199 201-215 217-223 ]
[SDR] Found 1 device(s)
[SDR] trying device  0:  Realtek, RTL2838UHIDIR, SN: 00000001
Detached kernel driver
Found Fitipower FC0012 tuner
[SDR] Using device 0: Generic RTL2832U OEM
Exact sample rate is: 1000000.026491 Hz
Allocating 15 zero-copy buffers
*** Saving signal to file g112_869M_1000k.cu8 (42780 samples, 131072 bytes)
*** Saving signal to file g113_869M_1000k.cu8 (42784 samples, 131072 bytes)
*** Saving signal to file g114_869M_1000k.cu8 (42787 samples, 131072 bytes)
*** Saving signal to file g115_869M_1000k.cu8 (58420 samples, 131072 bytes)
*** Saving signal to file g116_869M_1000k.cu8 (42780 samples, 131072 bytes)
*** Saving signal to file g117_869M_1000k.cu8 (42779 samples, 131072 bytes)
*** Saving signal to file g118_869M_1000k.cu8 (42782 samples, 131072 bytes)
*** Saving signal to file g119_869M_1000k.cu8 (42780 samples, 131072 bytes)
*** Saving signal to file g120_869M_1000k.cu8 (42781 samples, 131072 bytes)
*** Saving signal to file g121_869M_1000k.cu8 (185051 samples, 393216 bytes)
*** Saving signal to file g122_869M_1000k.cu8 (185371 samples, 393216 bytes)
*** Saving signal to file g123_869M_1000k.cu8 (185296 samples, 393216 bytes)
*** Saving signal to file g124_869M_1000k.cu8 (185260 samples, 393216 bytes)
*** Saving signal to file g125_869M_1000k.cu8 (185376 samples, 393216 bytes)
*** Saving signal to file g126_869M_1000k.cu8 (290592 samples, 655360 bytes)
*** Saving signal to file g127_869M_1000k.cu8 (185301 samples, 393216 bytes)
*** Saving signal to file g128_869M_1000k.cu8 (42791 samples, 131072 bytes)
*** Saving signal to file g129_869M_1000k.cu8 (58317 samples, 131072 bytes)
*** Saving signal to file g130_869M_1000k.cu8 (42781 samples, 131072 bytes)
*** Saving signal to file g131_869M_1000k.cu8 (42781 samples, 131072 bytes)
*** Saving signal to file g132_869M_1000k.cu8 (42784 samples, 131072 bytes)
*** Saving signal to file g133_869M_1000k.cu8 (185051 samples, 393216 bytes)
*** Saving signal to file g134_869M_1000k.cu8 (693353 samples, 1441792 bytes)
*** Saving signal to file g135_869M_1000k.cu8 (185273 samples, 393216 bytes)
*** Saving signal to file g136_869M_1000k.cu8 (328741 samples, 786432 bytes)
*** Saving signal to file g137_869M_1000k.cu8 (314250 samples, 655360 bytes)
*** Saving signal to file g138_869M_1000k.cu8 (520010 samples, 1048576 bytes)
*** Saving signal to file g139_869M_1000k.cu8 (347346 samples, 786432 bytes)
*** Saving signal to file g140_869M_1000k.cu8 (185358 samples, 393216 bytes)
*** Saving signal to file g141_869M_1000k.cu8 (42782 samples, 131072 bytes)
*** Saving signal to file g142_869M_1000k.cu8 (46687 samples, 131072 bytes)
```

From **g121** to **g127** I open the door. From **g133** to **g140** I close the door. I hope these works – otherwise let me know and I will keep trying my best to improve the captures.

Thank you.

ProfBoc75, 2023-07-23:
Hi @sindalschmidt 

It's better, but I guess the tuner FC0012 is guilty here and not good at this frequency.
I was facing same kind of issues where this tuner is not working properly for a weather station at 868 MHz and it's working fine with a R820T/R828D tuner.

Can't help more on this rf side, may be increase again the sample rate?
I can help on the coding part, but if rtl_433 is not able to understand the raw signal, it will be useless until we find the reason why. 

@zuckschwerdt @merbanan , need your help here, please. I found the decoding data layout and crc , but the signal itself , we can't get the FSK decoding despite the tuning with the gain and sample rate. any idea ? next steps ? Thanks.

zuckschwerdt, 2023-07-23:
In g125 the gain is still too high and we get clipping. E.g. g135 seems a little better with not too much clipping. Best advice I have currently: reduce gain or increase distance and also maybe try to get a little closer to center so it's easier to inspect, perhaps `-f 868.93M`

sindalschmidt, 2023-07-23:
Hi @ProfBoc75 and @zuckschwerdt 

I have dismounted the sensor and moved around the house with it. I have made some captures and tried finding the most optimal ones. The two captures attached here are taken with the antenna removed and 4 walls between the base of the antenna (antenna removed) and the sensor. Looking with pulseview there are definitely something in FSK to look at. I hope it's not just noise that I've captured – it is much more difficult to distinguish the captures. 

g224 I "close" the sensor and g246 I "open" the sensor.
[g224-g246.zip](https://github.com/merbanan/rtl_433/files/12139822/g224-g246.zip)

```
rtl_433 -g -10 -f 868.93M -X "n=cavius,m=FSK_PCM,s=415,l=415,r=2500,preamble=d31527,bits>=100,get=@104:{8}:state:[0x25:Open 0x24:Close" -S unknown -v -R 179
rtl_433 version 22.11-34-gb605a3d2 branch master at 202212302231 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.

New defaults active, use "-Y classic -s 250k" for the old defaults!

[Protocols] Registered 2 out of 223 device decoding protocols [ 179 ]
[SDR] Found 1 device(s)
[SDR] trying device  0:  Realtek, RTL2838UHIDIR, SN: 00000001
Detached kernel driver
Found Fitipower FC0012 tuner
[SDR] Using device 0: Generic RTL2832U OEM
Exact sample rate is: 1000000.026491 Hz
[SDR] Sample rate set to 1000000 S/s.
[Input] Bit detection level set to 0.0 (Auto).
[SDR] Tuner gain set to -9.900000 dB.
[Input] Reading samples in async mode...
[SDR] Tuned to 868.930MHz.
Allocating 15 zero-copy buffers
[Baseband] low pass filter for 1000000 Hz at cutoff 200000 Hz, 5.0 us
*** Saving signal to file g208_868.93M_1000k.cu8 (222537 samples, 524288 bytes)
*** Saving signal to file g209_868.93M_1000k.cu8 (58454 samples, 131072 bytes)
*** Saving signal to file g210_868.93M_1000k.cu8 (42780 samples, 131072 bytes)
*** Saving signal to file g211_868.93M_1000k.cu8 (42779 samples, 131072 bytes)
*** Saving signal to file g212_868.93M_1000k.cu8 (42782 samples, 131072 bytes)
*** Saving signal to file g213_868.93M_1000k.cu8 (42782 samples, 131072 bytes)
*** Saving signal to file g214_868.93M_1000k.cu8 (42781 samples, 131072 bytes)
*** Saving signal to file g215_868.93M_1000k.cu8 (58602 samples, 131072 bytes)
*** Saving signal to file g216_868.93M_1000k.cu8 (42779 samples, 131072 bytes)
*** Saving signal to file g217_868.93M_1000k.cu8 (42779 samples, 131072 bytes)
*** Saving signal to file g218_868.93M_1000k.cu8 (42783 samples, 131072 bytes)
*** Saving signal to file g219_868.93M_1000k.cu8 (46687 samples, 131072 bytes)
*** Saving signal to file g220_868.93M_1000k.cu8 (42782 samples, 131072 bytes)
*** Saving signal to file g221_868.93M_1000k.cu8 (42788 samples, 131072 bytes)
*** Saving signal to file g222_868.93M_1000k.cu8 (42785 samples, 131072 bytes)
*** Saving signal to file g223_868.93M_1000k.cu8 (42781 samples, 131072 bytes)
*** Saving signal to file g224_868.93M_1000k.cu8 (59766 samples, 131072 bytes)
*** Saving signal to file g225_868.93M_1000k.cu8 (42779 samples, 131072 bytes)
*** Saving signal to file g226_868.93M_1000k.cu8 (35540 samples, 131072 bytes)
*** Saving signal to file g227_868.93M_1000k.cu8 (42788 samples, 131072 bytes)
*** Saving signal to file g228_868.93M_1000k.cu8 (42791 samples, 131072 bytes)
*** Saving signal to file g229_868.93M_1000k.cu8 (42781 samples, 131072 bytes)
*** Saving signal to file g230_868.93M_1000k.cu8 (42797 samples, 131072 bytes)
*** Saving signal to file g231_868.93M_1000k.cu8 (192600 samples, 393216 bytes)
*** Saving signal to file g232_868.93M_1000k.cu8 (42789 samples, 131072 bytes)
*** Saving signal to file g233_868.93M_1000k.cu8 (42783 samples, 131072 bytes)
*** Saving signal to file g234_868.93M_1000k.cu8 (42785 samples, 131072 bytes)
*** Saving signal to file g235_868.93M_1000k.cu8 (42780 samples, 131072 bytes)
*** Saving signal to file g236_868.93M_1000k.cu8 (58596 samples, 131072 bytes)
*** Saving signal to file g237_868.93M_1000k.cu8 (42781 samples, 131072 bytes)
*** Saving signal to file g238_868.93M_1000k.cu8 (42781 samples, 131072 bytes)
*** Saving signal to file g239_868.93M_1000k.cu8 (42787 samples, 131072 bytes)
*** Saving signal to file g240_868.93M_1000k.cu8 (42779 samples, 131072 bytes)
*** Saving signal to file g241_868.93M_1000k.cu8 (46687 samples, 131072 bytes)
*** Saving signal to file g242_868.93M_1000k.cu8 (42782 samples, 131072 bytes)
*** Saving signal to file g243_868.93M_1000k.cu8 (42793 samples, 131072 bytes)
*** Saving signal to file g244_868.93M_1000k.cu8 (42780 samples, 131072 bytes)
*** Saving signal to file g245_868.93M_1000k.cu8 (42793 samples, 131072 bytes)
*** Saving signal to file g246_868.93M_1000k.cu8 (58586 samples, 131072 bytes)
*** Saving signal to file g247_868.93M_1000k.cu8 (42793 samples, 131072 bytes)
*** Saving signal to file g248_868.93M_1000k.cu8 (42779 samples, 131072 bytes)
*** Saving signal to file g249_868.93M_1000k.cu8 (42782 samples, 131072 bytes)
*** Saving signal to file g250_868.93M_1000k.cu8 (42780 samples, 131072 bytes)
```

sindalschmidt, 2023-07-23:
I have made a lot more captures at different frequencies, but most notable some with sample rate at 250k, as noted in the Cavius decoder, which should be more reliable. In the attached file I have selected all captures with what seems solid FSK readings; especially g019_868.67M_250k.sr and g021_868.67M_250k.sr.
[more captures.zip](https://github.com/merbanan/rtl_433/files/12139874/more.captures.zip)

zuckschwerdt, 2023-07-24:
Note that recordings are `.cu8`, the `.sr` are converted for screen display only. (You could also use the .cu8 on https://triq.org/spectrogram-next/ directly for a similar display.)

sindalschmidt, 2023-07-24:
Alright, thank you for pointing that out. Here are the actual recordings in `.cu8`

[captures.zip](https://github.com/merbanan/rtl_433/files/12152303/captures.zip)

zuckschwerdt, 2023-07-25:
There are two different signals in the recordings: at 868.93 M, fast with wide deviation and at 868.64 M, slow with narrow deviation -- I suspect the latter the one we are looking for. (My suggestion of `-f 868.93M` was an error reading the freqs, you are right to use `-f 868.70M`)

It's interesting how different the signal looks in 1000k (a straight line of FSK) vs the 250k (somewhat freq instable).

And it looks like the `-f 868.70M -s 250k` recoding (`g003_868.7M_250k.cu8`) is somewhat readable by rtl_433.

ProfBoc75, 2023-07-25:
> 

Hi @zuckschwerdt : yes, the second signal is a Wireless MBus signal well decoded by rtl_433 if you add the -Y minmax. 

[image]

For the rest, I don't know enough how the signal is decode by rtl_433 to get the FSK PCM , but something around the sampling rate and the frequency of the signal alter the result. may be the sensitivity of the tuner FC0012 and SNR ?
I don't know if @sindalschmidt  can use another kind of rtl dongle like the RTL-SDR v3 or any RTL with R820T/R828D/R860 tuner ?

sindalschmidt, 2023-08-05:
> I don't know if @sindalschmidt can use another kind of rtl dongle like the RTL-SDR v3 or any RTL with R820T/R828D/R860 tuner ?

Sorry for the late reply. Yes, the wireless mbus has been around for a while. I suspect a neighbor somewhere. Unfortunately I do not have another dongle. Can you explain a little about how you were able to manually tell the "Open" and "Close" signals apart, but rtl_433 is not able to? I'm curious of this. The dongle has no issues decoding the Cavius smoke, heat and water sensors that I have.

> And it looks like the `-f 868.70M -s 250k` recoding (`g003_868.7M_250k.cu8`) is somewhat readable by rtl_433.

Can we work from here and do some fine tuning?

ProfBoc75, 2023-08-05:
> Can you explain a little about how you were able to manually tell the "Open" and "Close" signals apart, but rtl_433 is not able to?

Hi @sindalschmidt : 

The signal is transmit in an analog way, here it's a FM signal for frequency modulation.

Then we need to transform this analog signal into a digital signal,  the FSK decoding, FSK for frequency-shift keying. 
In fact, there are 2 frequencies, one for blank (lower frequency) , the other one for the pulse (higher frequency).
Then, depending on the digital coding, the pulse can be 1 and blank 0.

So, I looked into your cu8 files, made by rtl_433, which contains lot of information.

- The I & Q signals, they are the analog rf signal : I/Q signaling refers to the use of two sinusoids that have the same frequency and a relative phase shift of 90°
- The AM signal / analog , from I & Q  --> here the amplitude do not change, so it's simple line.
- The FM signal / analog , from I & Q   --> I looked here, you have the 0 and 1 you can guess
- The Frame information / digital = start / stop of the signal transmit
- The ASK (for Amplitude Shift Keying) information / digital = the AM signal converted into digital information based on Amplitude changes.
- The FSK (for Frequency Shift Keying) information / digital = the FM signal converted into digital information based on Frequency changes.

Usually we just need to look at the FSK or ASK already decoded line (done by rtl_433), but here, you have nothing, so I looked into the FM line, where we can guess the figures. See my previous screenshots.
You understand that since rtl_433 is not able to decode the FSK, it can not be able to show you any result.
 
Then, carefully, for each file, I wrote down the 0 and 1 into an excel table (can be done from [bitbench web site](https://triq.net/bitbench)), convert the binary information into hexa data and I analyzed the result.

Between the 2 states, you have very few bit / byte changes, and I found which bit / byte changed.
About the data layout, it's a classic sync AA AA then the preamble 0xD31527, then the useful information like ID / State / CRC (CRC to validate the data integrity).

gdt, 2023-10-15:
Where are we on this?  Is there a path to a PR?

gdt, 2024-06-04:
I've marked this inactive to make it easier to ignore.