merbanan/rtl_433 #2563: Matching Gate Remote with Rtl_433 detection and Arduino Rc-Switch

M-C-Dev, 2023-07-13:
I'm trying to do copy my gate remote signal to improve my knowledge of rf. 
I've read documentation of signals, and how bit trace can be taken from a waveform, but i'm still a fresher so pardon me if I use wrong terms and correct me.
I have 2 remotes, for the same gate. Those 2 aren't among known drivers of rtl's library, but i've scanned the signals with:

`rtl_433.exe -A -R 0`

These are the results:
-REMOTE1
[https://triq.org/pdv/#AAB024040804E409C014906BFC808081809081918080808090819190808081809080808180809255+AAB024040104E409C014906BFC808081809081918080808090819190808081809080808180809355](url)

-REMOTE2
[https://triq.org/pdv/#AAB00C040104E809C81554674C809255+AAB024040504E809C81554674C808081809081918080808090819190808081809080808180809255+AAB021040104E809C81554674C808081809081918080808090819190808081809080808355](url)

So, observing results, this must be the binary sequence:
11110101111010011110111110

I desume that the "pulselength" value could be the short pwm, so around 1250. This is probably wrong, but I don't know how to calculate it exactly.

With these info, I've put up an Arduino with cheap and common FS1000A 433 MHz RF transmitter module.
Then I've uploaded following easy sketch: 

```
RCSwitch mySwitch = RCSwitch();

void setup() {

  Serial.begin(9600);
  
  // Transmitter is connected to Arduino Pin #10  
  mySwitch.enableTransmit(10);
  
  // Optional set protocol (default is 1, will work for most outlets)
  // mySwitch.setProtocol(2);

  // Optional set pulse length.
    mySwitch.setPulseLength(1245);
  
  // Optional set number of transmission repetitions.
 //   mySwitch.setRepeatTransmit(10);
  
}

void loop() { 
    
  //mySwitch.send("1111010111101001111011111");//this is what's I supposed to send...
   mySwitch.send("0000101000010110000100000");//this is opposite of previous to match the signal output...
// delay(500);
}
```

And this is what I got from rtl_433 detection:

[https://triq.org/pdv/#AAB024040904F00EB016EC9384818181819081908181818190819090818181819081818181818255+AAB024040104F00EB016EC9384818181819081908181818190819090818181819081818181818355](url)

Now unfortunately, this isn't able to open the gate, and I'm quite puzzled about the report.
I got 2 main questions:

1) why I need to send opposite binary sequence to match the output of rtl?
2) my binary always ends with 1. If I put the 0 at the very end of my binary I got this report that clearly adds an unwanted 1 at the end:
[https://triq.org/pdv/#AAB025040904EC0EB016EC938481818181908190818181819081909081818181908181818181908255+AAB025040104EC0EB016EC938481818181908190818181819081909081818181908181818181908355](url)

I'm stuck from days, I really need help here.
Thanks in advance for support!

zuckschwerdt, 2023-07-13:
This is a question for that library and not rtl_433.

rtl_433 uses a "1" to represent a fast or short pulse and a "0" for a slow or long pulse. This might is a sane and practical mapping but no way a standard.

You need to look at the signal more closely. Compare to e.g. https://triq.org/rtl_433/PULSE_FORMATS.html
The signal is not PWM. It looks somewhat like broken MC/DMC but to replicate you just need to care about sending the data from the gaps too.

Best to use PCM 1252 µs to look at signal, it should be even easier to comprehend then: 0/1 is off/on.

gdt, 2023-09-26:
Seems like an answered question and should be closed.