merbanan/rtl_433 #2568: WH1080 stopped being received by rtl_433

FosterSM, 2023-07-15:
At 9.30 am this morning my WH1080 Fineoffset weather stations stopped being received by my rtl_433 receivers.
The interesting thing is I have two separate systems, two computers, two receivers, two weather stations.
One station is used for testing the other is the live system. The system has a number of other temperature/humidity sensors from various manufacturers they are all being received fine. My inclination is to pull the battery on the test WH1080 to see if a power reset will clear the problem but I thought I should see if anyone has any idea what may be the problem. I have noted that the red led on the test wh1080 is still flashing from time to time so I think signals are being sent.
Thanks.

FosterSM, 2023-07-15:
One further point I have unplugged the test rtl_433 receiver and replugged it and restarted the rtl_433 software it but this had no effect.

FosterSM, 2023-07-15:
Now very confused @12pm both WH1080's started being received again I thought it might be an uptime issue but the main station is powered by rechargable batteries and charged by a solar panel and has been running since last november, the test station has normal batteries and has only been working for just over a week. Rule 39 states that there is no such thing as a coincidence but ...
One other thought I had was that the signal may be being blocked some how , but again the test setup is only a few metres from its reciever and the signal should be quite stong. The main station is about 15 metres from its receiving aerial and I have other sensors that are twice that distance and they have been fine. Could the transmit times some how got syncronised and blocked each other and have now drifted apart again, unlikley given the mark/space ratio of the transmitted signals but...

zuckschwerdt, 2023-07-15:
I'd guessed jamming by some other signal, colliding synchronized senders are the common case.

You can check the live waterfall using `-F rtl_tcp` and some rtl_tcp enable UI. The rtl_tcp output does not take up any resources when no ckient is connected, it can safely be default enabled.

Note that rtl_433 can receive such a wide range of different signals that it's not possible to have the strongest signal win, rtl_433 will always try to decode the sum of signals.

FosterSM, 2023-07-15:
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2023-07-15 15:01:33
model     : Fineoffset-WHx080                      Msg type  : 0             Station ID: 96
Battery   : 1            Temperature: 20.3 C       Humidity  : 63 %          Wind Direction: 180       Wind avg speed: 3.67      Wind gust : 9.79          Total rainfall: 284.7     Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2023-07-15 15:01:33
model     : Fineoffset-WHx080                      Msg type  : 0             Station ID: 96
Battery   : 1            Temperature: 20.3 C       Humidity  : 63 %          Wind Direction: 180       Wind avg speed: 3.67      Wind gust : 9.79          Total rainfall: 284.7     Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2023-07-15 15:01:33
model     : Fineoffset-WHx080                      Msg type  : 0             Station ID: 39
Battery   : 1            Temperature: 20.0 C       Humidity  : 61 %          Wind Direction: 293       Wind avg speed: 0.00      Wind gust : 0.00          Total rainfall: 217.5     Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2023-07-15 15:01:34
model     : Fineoffset-WHx080                      Msg type  : 0             Station ID: 39
Battery   : 1            Temperature: 20.0 C       Humidity  : 61 %          Wind Direction: 293       Wind avg speed: 0.00      Wind gust : 0.00          Total rainfall: 217.5     Integrity : CRC
bitbuffer_add_bit: Warning: row count limit (50 rows) reached
bitbuffer_add_bit: Warning: row count limit (50 rows) reached
bitbuffer_add_bit: Warning: row count limit (50 rows) reached
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
bitbuffer_add_bit: Warning: row count limit (50 rows) reached
bitbuffer_add_bit: Warning: row count limit (50 rows) reached

Thank you for your comment. I have setup a 3rd reciever to monitor the raw signal (did not want to change the working system at the moment, which is currently ok it seems). However there seems to be something strange  in the output, a: there appears to be an echo and b: a bitbuffer_add_bit warning. See above.

Could this be part of the problem?

FosterSM, 2023-07-15:
Also the times are the same for the two different WH1080 (nearly)???

FosterSM, 2023-07-15:
The echo is not confined to the WH1080 I have two inFactory-TH temp/hum sensors that seems to spit out upto 6 messages with the same time stamp others sensors only send one message.

FosterSM, 2023-07-15:
The bitbuffer_add_bit warning is probably a noise issue stemming from the fact that my 3rd receiver does not have a proper aerial, I have ordered one and it should be here tomorrow.

FosterSM, 2023-07-15:
It is now 530pm and the timestamps for the WH1080's have drifted apart by 1 second, so it looks like one of the WH1080 transmitter timers is very slighly faster than the other and therfore perioically they will clash. I wonder if it is worth tabulating the TX intervals (as accurately I can) of the various sources I am receiving (including the foreign ones I am not interested in) to try and predict potential outage periods?

FosterSM, 2023-07-15:
Having this experience has made me understand a little more why my frequency hopping strategy discussed in https://github.com/merbanan/rtl_433/discussions/2478 makes an improvement in my reception reliability. it however is not the full answer.

zuckschwerdt, 2023-07-15:
The bitbuffer_add_bit warning is just diagnostics, basically noise is overflowing the capacity, very unlikely to be real data.

If signals are far enough apart we could in theory split the received band into (slightly overlapping) sub-bands and decode each on it's own. But we'd need to know which band width to expect for the general case (somewhat wider than any FSK deviation I guess), and we'd need advanced signal processing which is heavy on the cpu. Not worth it for rare jamming and collisions so far.

FosterSM, 2023-07-15:
Yes I agree, the time is now 10pm and the two transmitters are now 2 seconds apart which means that I could calculate when the next blocking overlap will occur. Also the fact that the two transmitters have the same characteristics due to their design probably means they are more likely to block each other than overlapping signals from differnet transmitter designs, given my frequency hopping strategy etc. Once I have finished testing my current system and I dispense with the test WH1080 the overall system stability should return.

I use the JSON log output from the rtl_433 for my project, I assume the time stamps given in the output are based on the machine time the rtl_433 is running on, if the timestamps reports where to be increased to include ms resolution there is a possibily that run time analysis could give warning that signal collision is imminent and some appropriate action taken. I am not asking for a rtl_433 change just floating an idea.

zuckschwerdt, 2023-07-16:
Got you covered, see `rtl_433 -M`
```
...
	Use "time:usec" to add microseconds to date time meta data.
...
```

FosterSM, 2023-07-16:
Excellent, thank you.

gdt, 2023-09-26:
This was a question, and it seems resolved.

closed 2023-09-26 by gdt