merbanan/rtl_433 #2569: Add support: Cotech FT0203/18-3676 Anemometer

GreatAlbatross, 2023-07-17:
I've been around this sensor for a few years, and finally got around to working out what I think the bits are.
However, I'm not quite confident enough to write a proper decoder.

It's a battery+solar powered anemometer. 
I don't think it supports full degree accuracy, just cardinal and intercardinal (N, NW, W, SW etc.)

Example data: 
002938c800001dfffe06 - AVG speed 0.0M/S, GUST speed 0.0M/S, DIR: West 
002938c02428b5ffff98 - AVG speed 1.2M/S, GUST speed 1.4M/S, DIR: East

From what I've worked out from observing the receiving station (and a bit of deduction):

# Example: 002938c02428b5ffff98
#        : AAAAAABCDDEEFFGGGHHH
# A 6 bytes, some sort of code, does not change, I think it's the ID of the unit.
# B 1 byte , battery status
# C 1 byte , maybe direction flag? activates when I is >255 (I think)
# D 2 bytes, Averaged wind speed * 20
# E 2 bytes, Gust wind speed * 20
# F 2 bytes, Wind direction 
# G 3 bytes, Always FFF
# I 3 bytes, maybe last two are direction when combined with C as 255

Flex decoder I wrote to get some info out of the unit:
decoder n=Anemometer,m=OOK_MC_ZEROBIT,s=504 ,l=0 ,r=972,bits=79,get=battery_ok:@24:{4},get=wind_avg_m_s:@32:{8}:%d,get=gust_speed_m_s:@40:{8}:%d,get=direction:@48:{8}:[1:North 91:North-East 181:East 16:South-East 105:South-West 29:West 119:North-West]

[BitBench with a rough idea of mapping](https://triq.net/bitbench#c=002938c0242869ffff98&f=ID%3Ahhhhhh%20BATT%3Abbbb%20DIRFLAG%3Abbbb%20WIND20%3Ad%20GUST20%3Ad%20DIR%3Ahh%20IGN%3Add%20ANG%3Ad&cw=4)

If anyone could help adding support, it would be most appreciated, as I think this is as far as I can go!
If the data is needed more raw, please let me know. 
And as mentioned, I'm pretty sure the device only registers to 1/8 of a circle accuracy, so please don't worry too much about decoding the mystery numbers!

zuckschwerdt, 2023-07-17:
The "bytes" (8 bit) you list are "nibbles" (4 bit), but it's easy enough to understand as-is.

The wind direction mapping does not seem obvious, what is south? Do they combine with `C`? Can you put all 8 directions in the BitBench?
(e.g. lines of ` 002938c800001dfffe06 [AVG 0.0 GUST 0.0 West]`)

Usually we expect a checksum, likely at the end -- could the that last byte(s) be some checksum? Can you report a list of codes (e.g. 20-40 codes)?

GreatAlbatross, 2023-07-18:
Oops, sorry. My CS teacher would be giving me a withering look for that.
An of course, the day I need some more data, the wind is dead still. 
I gave someone a cup of coffee, and had them use it as a fidget spinner for 10 minutes.

The wind directions were not obvious. I had to sit looking at the receiver for a while.
I couldn't see a correlation, hence the mapping out in the flex decoder.

I think I may be completely wrong about the last 3 digits being related to angle.

Data: 8 lines where I've verified the direction, 6 lines with direction and speeds, 22 lines just for checksums.

[BitBench](https://triq.net/bitbench#c=%5BValues%20where%20I%20only%20verified%20the%20direction%5D&c=002938c0000169ffff70%20%5BSouth%5D&c=002938c00007c3fffe3a%20%5BSouth-West%5D&c=002938c800001dfffe06%20%5BWest%5D&c=002938c8000077ffff62%20%5BNorth-West%5D&c=002938c0000001ffff32%20%5BNorth%5D&c=002938c000005bffffba%20%5BNorth-East%5D&c=002938c00206b5fffebc%20%5BEast%5D&c=002938c000010fffff5e%20%5BSouth-East%5D&c=&c=&c=%5BValues%20where%20I%20verified%20all%5D&c=002938c01e4969ffff46%20%5BAVG%203.6%20GUST%201.5%20South%5D&c=002938c0020e01ffffb0%20%5BAVG%200.1%20GUST%200.7%20North%5D&c=002938c0146a01ffffda%20%5BAVG%202.1%20GUST%208.9%20North%5D&c=002938c866921dffff48%20%5BAVG%205.1%20GUST%207.3%20West%5D&c=002938c848da1dfffffa%20%5BAVG%203.6%20GUST%2010.9%20West%5D&c=002938c076ee01fffeba%20%5BAVG%205.9%20GUST%2011.9%20North%5D&c=&c=%5BValues%20where%20I%20just%20gathered%20examples%20for%20checksums%3A%5D&c=002938c0203c01ffff90&c=002938c80e221dfffff6&c=002938c812281dffffb0&c=002938c8263c1dfffe2c&c=002938c824341dffff96&c=002938c8121a1dfffece&c=002938c80a141dffffa8&c=002938c80a141dffffa8&c=002938c80e1a1dffffa2&c=002938c01234b5ffffbe&c=002938c81a3c1dffffb8&c=002938c0000001ffff32&c=002938c00a1a01ffffb8&c=002938c0102801ffff22&c=002938c01a3c01fffe8c&c=002938c0223cb5ffff26&c=002938c00206b5fffebc&c=002938c00000b5fffe22&c=002938c00c22b5fffe74&c=002938c00a14b5ffff8c&c=002938c00006b5ffff1a&c=002938c0020769ffffee&f=ID%3Ahhhhhh%20BATT%3Abbbb%20DIRFLAG%3Abbbb%20AVG%3Ad%20GUST%3Ad%20DIR%3Ahh%20%3F%3F%3F%3Add%20%3F%3F%3F%3Ad&cw=4)

The data tagged DIR correlates with direction.
South: 69, South-West: C3, West: 1D, North-West: 77, North: 01, North-East: 5B, East: 5B, South-East: 0F
WIND and GUST are the decimal conversions of the raw values, which are 20x real value.

Thanks again for taking a look at this.

zuckschwerdt, 2023-07-19:
You need to shift on bit left, e.g. try this: `ID:24h BATT:4b DIRFLAG:3b AVG:d GUST:d DIR:8d ?:16h CHK?:8h x`

GreatAlbatross, 2023-07-19:
:open_mouth: 
Holy cow, you got it!

The windspeed values are now correct.
And that lines everything up with the direction flag being equal to 256
West (14) + 256 = 270, which is correct.
North-West (59), plus 256 = 315, also correct.

The block of FFFF is now always FFFF too.

[Updated bitbench so it's all here](https://triq.net/bitbench#c=%5BValues%20where%20I%20only%20verified%20the%20direction%5D&c=002938c0000169ffff70%20%5BSouth%5D&c=002938c00007c3fffe3a%20%5BSouth-West%5D&c=002938c800001dfffe06%20%5BWest%5D&c=002938c8000077ffff62%20%5BNorth-West%5D&c=002938c0000001ffff32%20%5BNorth%5D&c=002938c000005bffffba%20%5BNorth-East%5D&c=002938c00206b5fffebc%20%5BEast%5D&c=002938c000010fffff5e%20%5BSouth-East%5D&c=&c=&c=%5BValues%20where%20I%20verified%20all%5D&c=002938c01e4969ffff46%20%5BAVG%203.6%20GUST%201.5%20South%5D&c=002938c0020e01ffffb0%20%5BAVG%200.1%20GUST%200.7%20North%5D&c=002938c0146a01ffffda%20%5BAVG%202.1%20GUST%208.9%20North%5D&c=002938c866921dffff48%20%5BAVG%205.1%20GUST%207.3%20West%5D&c=002938c848da1dfffffa%20%5BAVG%203.6%20GUST%2010.9%20West%5D&c=002938c076ee01fffeba%20%5BAVG%205.9%20GUST%2011.9%20North%5D&c=&c=%5BValues%20where%20I%20just%20gathered%20examples%20for%20checksums%3A%5D&c=002938c0203c01ffff90&c=002938c80e221dfffff6&c=002938c812281dffffb0&c=002938c8263c1dfffe2c&c=002938c824341dffff96&c=002938c8121a1dfffece&c=002938c80a141dffffa8&c=002938c80a141dffffa8&c=002938c80e1a1dffffa2&c=002938c01234b5ffffbe&c=002938c81a3c1dffffb8&c=002938c0000001ffff32&c=002938c00a1a01ffffb8&c=002938c0102801ffff22&c=002938c01a3c01fffe8c&c=002938c0223cb5ffff26&c=002938c00206b5fffebc&c=002938c00000b5fffe22&c=002938c00c22b5fffe74&c=002938c00a14b5ffff8c&c=002938c00006b5ffff1a&c=002938c0020769ffffee&f=ID%3Ahhhhhh%20BATT%3Abbbb%20DIRFLAG%3Abbbb%20AVG%3Ad%20GUST%3Ad%20DIR%3Ahh%20%3F%3F%3F%3Add%20%3F%3F%3F%3Ad&cw=4)

zuckschwerdt, 2023-07-19:
The `FFFF` block might be an optional UV sensor? Is there documentation about such a feature?

The checksum is a CRC-8 with poly 0x31, init is either 0xd4 for the full code (with a 0-bit prepended) or 0xc0 for the code with the leading 7 0-bits skipped.

GreatAlbatross, 2023-07-19:
Nothing about a UV sensor I'm afraid. I guess it's possible the board was used for something else? 
  But I have never seen anything other than FFFF.
  There is no documentation, as far as I know.  
  The only thing I can add leafing through the manual, is that the resolution of the anemometer is 45 degrees, maximum speed is 50M/S, and the update interval is 16 seconds.

GreatAlbatross, 2023-09-11:
Hope this is OK to add here: While I haven't made any progress on writing a full decoder for rtl_433, I have now duct-taped together working integration with HomeAssistant.

The direction values are mapped rather than calculated, and HASS has to handle dividing the speed values by 10, but it is working.
https://community.home-assistant.io/t/basic-support-for-cotech-ft0203-18-3676-anemometer/608702

zuckschwerdt, 2023-09-11:
If you don't have the time or if writing in C isn't your expertise then I can add the decoder.
I will need a few samples (https://triq.org/rtl_433/ANALYZE.html) uploaded here as zip. And it would be helpful if you could test the decoder (from a PR) before merge.

klohner, 2024-01-09:
@GreatAlbatross Would you be able to zip up a few .cu8 samples of this signal and attach to this issue?

klohner, 2024-01-09:
Figured out another piece of the code.  The 29th bit is being used as a 9th bit for the direction.  If set, add 256 to what you're getting DIR and you'll get a compass heading in degrees.  For example, a heading to the west is DIR "014" in decimal.  But this 9th bit is set, so add 256 for 270.  270 degrees is West.

[BitBench](https://triq.net/bitbench#c=002938c800001dfffe06%20%5B%20-%20AVG%20speed%200.0M%2FS%2C%20GUST%20speed%200.0M%2FS%2C%20DIR%3A%20West%5D&c=002938c02428b5ffff98%20%5B-%20AVG%20speed%201.2M%2FS%2C%20GUST%20speed%201.4M%2FS%2C%20DIR%3A%20East%5D&c=&c=%5BValues%20where%20I%20only%20verified%20the%20direction%5D&c=002938c0000169ffff70%20%5BSouth%5D&c=002938c00007c3fffe3a%20%5BSouth-West%5D&c=002938c800001dfffe06%20%5BWest%5D&c=002938c8000077ffff62%20%5BNorth-West%5D&c=002938c0000001ffff32%20%5BNorth%5D&c=002938c000005bffffba%20%5BNorth-East%5D&c=002938c00206b5fffebc%20%5BEast%5D&c=002938c000010fffff5e%20%5BSouth-East%5D&c=&c=&c=%5BValues%20where%20I%20verified%20all%5D&c=002938c01e4969ffff46%20%5BAVG%201.5%20GUST%203.6%20South%5D&c=002938c0020e01ffffb0%20%5BAVG%200.7%20GUST%200.1%20North%5D&c=002938c0146a01ffffda%20%5BAVG%208.9%20GUST%202.1%20North%5D&c=002938c866921dffff48%20%5BAVG%207.3%20GUST%205.1%20West%5D&c=002938c848da1dfffffa%20%5BAVG%2010.9%20GUST%203.6%20West%5D&c=002938c076ee01fffeba%20%5BAVG%2011.9%20GUST%205.9%20North%5D&c=&c=%5BValues%20where%20I%20just%20gathered%20examples%20for%20checksums%3A%5D&c=002938c0203c01ffff90&c=002938c80e221dfffff6&c=002938c812281dffffb0&c=002938c8263c1dfffe2c&c=002938c824341dffff96&c=002938c8121a1dfffece&c=002938c80a141dffffa8&c=002938c80a141dffffa8&c=002938c80e1a1dffffa2&c=002938c01234b5ffffbe&c=002938c81a3c1dffffb8&c=002938c0000001ffff32&c=002938c00a1a01ffffb8&c=002938c0102801ffff22&c=002938c01a3c01fffe8c&c=002938c0223cb5ffff26&c=002938c00206b5fffebc&c=002938c00000b5fffe22&c=002938c00c22b5fffe74&c=002938c00a14b5ffff8c&c=002938c00006b5ffff1a&c=002938c0020769ffffee&f=ID%3A24h%20BATT%3A4b%20DIR_BIT_NINE%3A1b%20%3F%3F%3A2b%20AVG%3A8d%20GUST%3A8d%20DIR%3A8d%20FFFF%3A16h%20%3F%3Ab%20CHK%3F%3A8h&cw=4)

GreatAlbatross, 2024-04-05:
> @GreatAlbatross Would you be able to zip up a few .cu8 samples of this signal and attach to this issue?

Sorry for the slow reply.
If this would still be useful, how do I go about recording?

The battery on my device has recently failed, so I reasoned it was a good time to capture values before and after replacing!
(It runs during daylight hours due to the solar panel)

ProfBoc75, 2024-04-05:
> Sorry for the slow reply. If this would still be useful, how do I go about recording?
>
> The battery on my device has recently failed, so I reasoned it was a good time to capture values before and [...]

Hi @GreatAlbatross, try this to get some recording with your flex configuration:

    rtl_433 -X "n=Anemometer,m=OOK_MC_ZEROBIT,s=504,l=0,r=972,bits>=79" -S unknown

Each time a signal is matching your flex decoder, write down the recording file name and its information from station display, zip few cu8 files and drop here in a comment with the information.

You may have to play with the gain and / or remove the antenna to get good signal. drop cu8 here https://triq.org/pdv/ to see if the signal is good quality.

GreatAlbatross, 2024-04-08:
I'm not sure if I managed to get any useful data in the cu8 files, I'm afraid.
Any that I looked at, didn't seem to match the data being picked up.

Battery change is done, and it's gone from:
002938c0141a01ffff72 to 
00293b600c0eb5fffffa

ProfBoc75, 2024-04-08:
The cu8 file is not good, not the anemometer. Try other captures. You can replay the cu8 with the same flex decoder to select good cu8 files. More details [Analyze.md](https://github.com/merbanan/rtl_433/blob/master/docs/ANALYZE.md)

I notice that the wind max speed is 50 m/s , from the user guide, and based on the current bitbench layout one byte is not enough and another 9 bit MSB is required like for wind dir. Not sure about the battery flag also.

My [bitbench](https://triq.net/bitbench#c=%5B%20AVG%200.0M%2FS%2C%20GUST%200.0M%2FS%2C%20West%5D&c=002938c800001dfffe06&c=&c=%5B%20AVG%201.2M%2FS%2C%20GUST%201.4M%2FS%2C%20East%20but%20expected%201.8%20%26%202.0%20%3A%5D&c=002938c02428b5ffff98&c=&c=%5BValues%20where%20I%20only%20verified%20the%20direction%5D&c=002938c0000169ffff70%20%5BSouth%5D&c=002938c00007c3fffe3a%20%5BSouth-West%5D&c=002938c800001dfffe06%20%5BWest%5D&c=002938c8000077ffff62%20%5BNorth-West%5D&c=002938c0000001ffff32%20%5BNorth%5D&c=002938c000005bffffba%20%5BNorth-East%5D&c=002938c00206b5fffebc%20%5BEast%5D&c=002938c000010fffff5e%20%5BSouth-East%5D&c=&c=&c=%5BValues%20where%20I%20verified%20all%5D&c=002938c01e4969ffff46%20%5BAVG%201.5%20GUST%203.6%20South%5D&c=002938c0020e01ffffb0%20%5BAVG%200.1%20GUST%200.7%20North%5D&c=002938c0146a01ffffda%20%5BAVG%202.1%20GUST%208.9%20North%20expected%201.0%20%26%205.3%5D&c=002938c866921dffff48%20%5BAVG%205.1%20GUST%207.3%20West%5D&c=002938c848da1dfffffa%20%5BAVG%203.6%20GUST%2010.9%20West%5D&c=002938c076ee01fffeba%20%5BAVG%205.9%20GUST%2011.9%20North%5D&c=&c=%5BValues%20where%20I%20just%20gathered%20examples%20for%20checksums%3A%5D&c=002938c0203c01ffff90&c=002938c80e221dfffff6&c=002938c812281dffffb0&c=002938c8263c1dfffe2c&c=002938c824341dffff96&c=002938c8121a1dfffece&c=002938c80a141dffffa8&c=002938c80a141dffffa8&c=002938c80e1a1dffffa2&c=002938c01234b5ffffbe&c=002938c81a3c1dffffb8&c=002938c0000001ffff32&c=002938c00a1a01ffffb8&c=002938c0102801ffff22&c=002938c01a3c01fffe8c&c=002938c0223cb5ffff26&c=002938c00206b5fffebc&c=002938c00000b5fffe22&c=002938c00c22b5fffe74&c=002938c00a14b5ffff8c&c=002938c00006b5ffff1a&c=002938c0020769ffffee&c=&c=%5B%20Battery%20change%20is%20done%2C%20and%20it's%20gone%20from%3A%20%5D&c=002938c0141a01ffff72%20%5B%20battery_low%20%5D&c=00293b600c0eb5fffffa%20%5B%20new%20battery%20%5D&f=7x%20ID%3A16h%20BATT_LEVEL%3F%204d%20MSB%3F%201d%20DIR_MSB%201d%20MSB%3F%201d%20MSB%3F%201d%20AVG_LSB%208d%20GUST_LSB%208d%20DIR_LSB%208d%20F%3A16h%20CHK%3F%3A8h&z=1&cw=4) update and possible MSB

ProfBoc75, 2024-04-09:
Well, looks like we already have a very close decoder handle by rtl_433, `cotech_36_7959.c` same CRC-8 (0x31, 0xc0) and data layout very close with more information like lux/uv/rain.temp/hum than here.

Assumption that we have the msb in the same order: [bitbench](https://triq.net/bitbench#c=%5B%20AVG%200.0M%2FS%2C%20GUST%200.0M%2FS%2C%20West%5D&c=&c=002938c800001dfffe06&c=%5B%20AVG%201.2M%2FS%2C%20GUST%201.4M%2FS%2C%20East%20but%20expected%201.8%20%26%202.0%20%3A%5D&c=&c=002938c02428b5ffff98&c=%5BValues%20where%20I%20only%20verified%20the%20direction%5D&c=&c=002938c0000169ffff70%20%5BSouth%5D&c=002938c00007c3fffe3a%20%5BSouth-West%5D&c=002938c800001dfffe06%20%5BWest%5D&c=002938c8000077ffff62%20%5BNorth-West%5D&c=002938c0000001ffff32%20%5BNorth%5D&c=002938c000005bffffba%20%5BNorth-East%5D&c=002938c00206b5fffebc%20%5BEast%5D&c=002938c000010fffff5e%20%5BSouth-East%5D&c=%5BValues%20where%20I%20verified%20all%5D&c=&c=002938c01e4969ffff46%20%5BAVG%201.5%20GUST%203.6%20South%5D&c=002938c0020e01ffffb0%20%5BAVG%200.1%20GUST%200.7%20North%5D&c=002938c0146a01ffffda%20%5BAVG%202.1%20GUST%208.9%20North%20expected%201.0%20%26%205.3%5D&c=002938c866921dffff48%20%5BAVG%205.1%20GUST%207.3%20West%5D&c=002938c848da1dfffffa%20%5BAVG%203.6%20GUST%2010.9%20West%5D&c=002938c076ee01fffeba%20%5BAVG%205.9%20GUST%2011.9%20North%5D&c=%5BValues%20where%20I%20just%20gathered%20examples%20for%20checksums%3A%5D&c=&c=002938c0203c01ffff90&c=002938c80e221dfffff6&c=002938c812281dffffb0&c=002938c8263c1dfffe2c&c=002938c824341dffff96&c=002938c8121a1dfffece&c=002938c80a141dffffa8&c=002938c80a141dffffa8&c=002938c80e1a1dffffa2&c=002938c01234b5ffffbe&c=002938c81a3c1dffffb8&c=002938c0000001ffff32&c=002938c00a1a01ffffb8&c=002938c0102801ffff22&c=002938c01a3c01fffe8c&c=002938c0223cb5ffff26&c=002938c00206b5fffebc&c=002938c00000b5fffe22&c=002938c00c22b5fffe74&c=002938c00a14b5ffff8c&c=002938c00006b5ffff1a&c=002938c0020769ffffee&c=%5B%20Battery%20change%20is%20done%2C%20and%20it's%20gone%20from%3A%20%5D&c=&c=002938c0141a01ffff72%20%5B%20battery_low%20%5D&c=00293b600c0eb5fffffa%20%5B%20new%20battery%20%5D&f=7x%20PP%208h%20ID%2011h%20BATT_LEVEL%201b%20%3F%201b%20DIR_MSB%201b%20GUST_MSB%201b%20AVG_MSB%201b%20AVG_LSB%208d%20GUST_LSB%208d%20DIR_LSB%208d%20F%3A16h%20CHK%3F%3A8h&z=1&cw=4)

Few remarks:
- I excluded the 7 first bits to get exact 9 bytes (72 bits instead of 79 bits)
- The first byte 0x14, doesn't alter the CRC and could be excluded remaining 8 bytes.

Raw:

    {79}00 29 38 c8 00 00 1d ff fe 06

Left shift 7 bits:

    {72}14 9c 64 00 00 0e ff ff 03
        PP[II IM SS GG DD FF FF]CC
             |  |
        /----+  +-----\
         123 4 5 6 7 8  
         III B F MSB_DIR 
                   MSB_GUST 
                     MSB_AVG
Data Layout:
- PP: 8b Preamble 0x14
- II: 11b ID
- Battery: 1b
- Fixed: 1b = 0
- MSB Wind Dir: 1b
- MSB Wind Gust: 1b
- SS: 9b AVG Wind Speed (MSB 1b + LSB 8b)
- GG: 8b LSB Wind Gust
- DD: 8b LSB Wind Dir
- FFFF: 16b Fixed 0xFFFF
- CRC: 8b poly 0x31, init 0xC0

All should be confirmed by new captures.

GreatAlbatross, 2024-04-09:
[Further examples added to bitbench](https://triq.net/bitbench#c=%5B%20AVG%200.0M%2FS%2C%20GUST%200.0M%2FS%2C%20West%5D&c=&c=002938c800001dfffe06&c=%5B%20AVG%201.2M%2FS%2C%20GUST%201.4M%2FS%2C%20East%20but%20expected%201.8%20%26%202.0%20%3A%5D&c=&c=002938c02428b5ffff98&c=%5BValues%20where%20I%20only%20verified%20the%20direction%5D&c=&c=002938c0000169ffff70%20%5BSouth%5D&c=002938c00007c3fffe3a%20%5BSouth-West%5D&c=002938c800001dfffe06%20%5BWest%5D&c=002938c8000077ffff62%20%5BNorth-West%5D&c=002938c0000001ffff32%20%5BNorth%5D&c=002938c000005bffffba%20%5BNorth-East%5D&c=002938c00206b5fffebc%20%5BEast%5D&c=002938c000010fffff5e%20%5BSouth-East%5D&c=%5BValues%20where%20I%20verified%20all%5D&c=&c=002938c01e4969ffff46%20%5BAVG%201.5%20GUST%203.6%20South%5D&c=002938c0020e01ffffb0%20%5BAVG%200.1%20GUST%200.7%20North%5D&c=002938c0146a01ffffda%20%5BAVG%202.1%20GUST%208.9%20North%20expected%201.0%20%26%205.3%5D&c=002938c866921dffff48%20%5BAVG%205.1%20GUST%207.3%20West%5D&c=002938c848da1dfffffa%20%5BAVG%203.6%20GUST%2010.9%20West%5D&c=002938c076ee01fffeba%20%5BAVG%205.9%20GUST%2011.9%20North%5D&c=%5BValues%20where%20I%20just%20gathered%20examples%20for%20checksums%3A%5D&c=&c=002938c0203c01ffff90&c=002938c80e221dfffff6&c=002938c812281dffffb0&c=002938c8263c1dfffe2c&c=002938c824341dffff96&c=002938c8121a1dfffece&c=002938c80a141dffffa8&c=002938c80a141dffffa8&c=002938c80e1a1dffffa2&c=002938c01234b5ffffbe&c=002938c81a3c1dffffb8&c=002938c0000001ffff32&c=002938c00a1a01ffffb8&c=002938c0102801ffff22&c=002938c01a3c01fffe8c&c=002938c0223cb5ffff26&c=002938c00206b5fffebc&c=002938c00000b5fffe22&c=002938c00c22b5fffe74&c=002938c00a14b5ffff8c&c=002938c00006b5ffff1a&c=002938c0020769ffffee&c=%5B%20Battery%20change%20is%20done%2C%20and%20it's%20gone%20from%3A%20%5D&c=&c=002938c0141a01ffff72%20%5B%20battery_low%20%5D&c=00293b600c0eb5fffffa%20%5B%20new%20battery%20%5D&c=&c=%5BFurther%20captures%20following%20battery%20replacement%5D&c=00293b682a5077ffff36&c=00293b602a4201fffe68&c=00293b6014225bfffe42&c=00293b681a2277fffe2c&c=00293b602a49c3ffff72&c=00293b60467601fffe36&c=00293b603650b5ffff1a&c=00293b68323c1dfffe92&c=00293b681e341dffff66&c=00293b60284201ffffce&c=00293b683c501dfffea6&c=00293b6820481dffffee&c=00293b681c281dfffeea&c=00293b603049c3fffe96&f=7x%20PP%208h%20ID%2011h%20BATT_LEVEL%201b%20%3F%201b%20DIR_MSB%201b%20GUST_MSB%201b%20AVG_MSB%201b%20AVG_LSB%208d%20GUST_LSB%208d%20DIR_LSB%208d%20F%3A16h%20CHK%3F%3A8h&z=1&cw=4)

And a gzipped cu8 which I'm pretty sure is correct. I realised that as I had written a very rudimentary encoder previously, it might have been skipping saving when unknown was specified.
[g083_433.92M_250k.cu8.gz](https://github.com/merbanan/rtl_433/files/14917679/g083_433.92M_250k.cu8.gz)

gdt, 2024-06-04:
Status?

gdt, 2024-06-23:
@GreatAlbatross Where are we on this?  Are you working on a PR to add a flex decoder, or writing a real decoder, or ?

GreatAlbatross, 2024-06-24:
I don't think my flex decoder is up to contribution quality, and I don't think I have the skills to write a real decoder unfortunately. (at least, not in a sensible timeframe!)

I'm happy to do some additional captures to help things.

klohner, 2024-06-24:
Some notes I have on this device/decoder FWIW:

-  Caution using the "OOK_MC_ZEROBIT" demodulator -- it provides a misleading result in this case.  The signal starts with several logical "1" demodulated MC bits, but the MC_ZEROBIT demodulator demodulates these as logical "0" bits until the first "true" logical "0" bit.  In this example, it is demodulating data as "{79}00293b602a49c3ffff72" but the actual MC demodulation should be "{79}ffa93b602a49c3ffff72"

[image]

```
$ rtl_433 -r g083_433.92M_250k.cu8 -X "n=Cotech-PCM-raw,m=OOK_PCM,s=488,l=488,r=1500,bits>=158" -X "n=Cotech-PCM,m=OOK_PCM,s=488,l=488,r=1500,bits>=158,symbol_zero={2}4,symbol_one={2}8" -X "n=Cotech-ZB,m=OOK_MC_ZEROBIT,s=488,l=0,r=1500,bits>=79"
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2024-06-24T13:21:29.067959
model     : Cotech-PCM-raw                         count     : 1
num_rows  : 1            rows      :
len       : 160          data      : aaaa99965a9a695559996596a55aaaaaaaaa6a58
codes     : {160}aaaa99965a9a695559996596a55aaaaaaaaa6a58
Modulation: ASK          Freq      : 433.9 MHz
RSSI      : -0.1 dB      SNR       : 14.8 dB       Noise     : -14.9 dB
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2024-06-24T13:21:29.067959
model     : Cotech-PCM   count     : 1             num_rows  : 1
rows      :
len       : 79           data      : ffa93b602a49c3ffff72
codes     : {79}ffa93b602a49c3ffff72
Modulation: ASK          Freq      : 433.9 MHz
RSSI      : -0.1 dB      SNR       : 14.8 dB       Noise     : -14.9 dB
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2024-06-24T13:21:29.067959
model     : Cotech-ZB    count     : 1             num_rows  : 1
rows      :
len       : 79           data      : 00293b602a49c3ffff72
codes     : {79}00293b602a49c3ffff72
Modulation: ASK          Freq      : 433.9 MHz
RSSI      : -0.1 dB      SNR       : 14.8 dB       Noise     : -14.9 dB
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
```

- Using OOK_MC_ZEROBIT, the sync (preamble) byte gets mis-demodulated as "14" when it should really be "d4"

- If the logical byte before the "d4" sync is assumed to be "ff", the CRC-8 at the end of the remaining 72-bit value is a recognized as "[CRC-8/NRSC-5](https://reveng.sourceforge.io/crc-catalogue/1-15.htm#crc.cat.crc-8-nrsc-5)"

- The Cotech-367954 actually has an identical sync/preamble byte (though its decoder is misleading because `cotech_36_7959.c` is using OOK_MC_ZEROBIT for the initial demodulation) and its CRC is the same algorithm.

```
# Cotech FT0203/18-3676
$ reveng -w 8 -s ffd49db01524e1ffffb9 ffd49c6400000effff03 ffd49c6012145affffcc ffd49c600000b4ffffb8 ffd49c600003e1ffff1d
width=8  poly=0x31  init=0xff  refin=false  refout=false  xorout=0x00  check=0xf7  residue=0x00  name="CRC-8/NRSC-5"

# Cotech-367959
$ reveng -w 8 -s ffd4c5400000f10318846d22fffbfbe5 ffd4c54400002b0318846d22fffbfb5b ffd4c54000007e0318846d22fffbfbc5 ffd4c5400107bf0348846d22fffbfb2f
width=8  poly=0x31  init=0xff  refin=false  refout=false  xorout=0x00  check=0xf7  residue=0x00  name="CRC-8/NRSC-5"
```

- I can't find agreement on the "battery" indicator bit.  If `cotech_36_7959.c` is correct, this logical bit is "0" for battery OK, and "1" for low.  In the Cotech FT0203/18-3676 samples, this bit is "0" in all cases - so battery was always OK, even before replacement?  I think ID changes each reboot, so we'd need captures of several reboots to get a better feel for what is ID bits and what is the battery_ok bit.

- It may be the case that there is a "model" or "type" nibble just before the id byte.  This might make it easier/practical to combine decoders for both Cotech devices into one decoder c file.

- Symbol length for both Cotech devices seems to be 488 µs.

- I can craft Cotech FT0203/18-3676 and Cotech-367959 decoders in flex (with no CRC verification and limited support of presenting decoded values) like this:

```
# somewhat based on cotech_36_7959.c
# tested on samples in https://github.com/zuckschwerdt/rtl_433_tests/tree/master/tests/cotech/cotech_36-7959/01
# preamble {16}a665 is MC {8}d4
# prefix result with {16}ffd4 and final byte is "CRC-8/NRSC-5" (poly=0x31  init=0xff)
decoder {
  name        = Cotech-367959-Flex,
  modulation  = OOK_PCM,
  short       = 488,
  long        = 488,
  reset       = 5000,
[... 10 lines cut ...]
  get         = CRC:@104:{8}:%02x,
}

# https://github.com/merbanan/rtl_433/issues/2569
decoder {
  name        = Cotech FT0203/18-3676-Flex,
  modulation  = OOK_PCM,
  short       = 488,
  long        = 488,
  reset       = 5000,
  bits       >= 160,
  bits       <= 176,
  preamble    = {16}a665,
  symbol_zero = {2}4,
  symbol_one  = {2}8,
  get         = model:@0:{4}:%d,
  get         = id:@4:{8}:%d,
  get         = bat:@12:{1}:%d,
  get         = DIR_MSB:@13:{1}:%x,
  get         = AVG m/s x10:@15:{9}:%d,
  get         = CRC:@56:{8}:%02x,
}
```

- Updated [BitBench](https://triq.net/bitbench#c=%7B79%7Dffa93b602a49c3ffff72%20%5Bg083_433.92M_250k.cu8%20MC%20%20%20%20%20%20%20%20%20%20%20%20actual%20prefix%20is%20y11101010%5D&c=%7B79%7D00293b602a49c3ffff72%20%5Bg083_433.92M_250k.cu8%20MCZB%20mis-demodulated%20prefix%20is%20y00001010%5D&c=&c=%5B%20AVG%200.0M%2FS%2C%20GUST%200.0M%2FS%2C%20West%5D&c=&c=002938c800001dfffe06&c=%5B%20AVG%201.2M%2FS%2C%20GUST%201.4M%2FS%2C%20East%20but%20expected%201.8%20%26%202 [... cut ...] =00293b682a5077ffff36&c=00293b602a4201fffe68&c=00293b6014225bfffe42&c=00293b681a2277fffe2c&c=00293b602a49c3ffff72&c=00293b60467601fffe36&c=00293b603650b5ffff1a&c=00293b68323c1dfffe92&c=00293b681e341dffff66&c=00293b60284201ffffce&c=00293b683c501dfffea6&c=00293b6820481dffffee&c=00293b681c281dfffeea&c=00293b603049c3fffe96&f=PP%3AxDh%20MODEL%3A4h%20ID%3A8h%20BAT%3Ab%20DIR_MSB%3A1b%20GUST_MSB%3A1b%20AVG%3A9d%20GUST_LSB%3A8d%20DIR_LSB%3A8d%20F%3A16h%20CRC%3A8h%20xxxx%0A%0A%0A&z=1&a=Sync&m=y10100&cw=4)

GreatAlbatross, 2024-07-11:
Thankyou for that, I'm afraid I missed this update!
I will try to get some more captures and reboots in the near future.

larand54, 2024-12-22:
Hello, I just wonder how this project is going? Done any progress yet.  This is because I is about buying such system "Cotech FT0203" but want to be sure I can include it in my rtl_433 based system first.

//LG

zuckschwerdt, 2024-12-22:
I think the protocol is mostly figured out and there is a working flex decoder. We should upgrade this to a full decoder some day.

MatAhman, 2025-02-14:
Thanks for a good work with the flex decoder. my plan was to use this with my lillygo openmqtt rtl_433 but once i started to dig into it that it doesnt support flex decoders :-( i started rtl_433 on my linux machine with the flex decoder. This is the output i get in rtl_433. i get data on windspeed but not on direction. DIR always reports 0.
  
time      : 2025-02-14 17:02:44
model     : Cotech FT0203/18-3676-Flex             count     : 1
num_rows  : 1            rows      : 
len       : 64           data      : 9580020d87ffff26
model     : 9            id        : 88
bat       : 0            DIR_MSB   : 0             AVG m/s x10: 2
CRC       : 26
codes     : {64}9580020d87ffff26

ProfBoc75, 2025-02-15:
Hi All, I will try during my vacations to work on a decoder ... 

@MatAhman , yes, the flex decoder is not completed, missing metrics, as you can see few data are 9 bit, so one bit from a byte and the other 8 bits from another byte, here DIR_MSB = 0 , and DIR_LSB is missing from flex.

Only AVG Wind speed is ok because the 9 bits are contiguous in the data layout. So definitely a decoder is needed to get the figures, flex is not good enough for this sensor, and not able also to check the CRC.

GreatAlbatross, 2025-03-14:
Let me know if there are any other bits I can assist with.
Happy to test things.