merbanan/rtl_433 #2570: Quit with Ctrl-C

lhofinger, 2023-07-18:
Hello,

I've just discovered rtl_433, I'm using rtl_433-rtlsdr.exe (nightly build) on Windows with RTL SDR V3 dongle. I am surprised by all the devices in the neighborhood that it manages to decode!

I would like to execute it every hour for one or two minutes. What is the right way to exit rtl_433-rtlsdr.exe?

When I press Ctrl-C, it shows "Signal caught, exiting!":
- Sometimes, all is OK, the dongle is cooling down.
- But sometimes, the dongle keeps warm until next run.
- Sometimes, there is an error "[rtlsdr_read_loop] LIBUSB_ERROR_NOT_FOUND: Entity not found! Check your RTL-SDR dongle, USB cables, and power supply. [SDR] async read failed (-5)."

Any idea?

lhofinger, 2023-07-18:
Hello, thank you for your answer and apologize my bad english...

I've just tried several times with -T 5, with an ammeter on the RTL-SDR.

Startups and runnings are always OK.

4 times, rtl_433 ends with: "Time expired, exiting!
[rtlsdr_read_loop] LIBUSB_ERROR_NOT_FOUND: Entity not found! Check your RTL-SDR dongle, USB cables, and power supply.
[SDR] async read failed (-5)." When it ends with this message, the ammeter doesn't drop.

2 times, rtl_433 ends with: "Time expired, exiting!". When it ends without the error, the ammeter is dropping on exit.

Perhaps that when it goes wrong it begins to stop the aquiring, the error "async read failed" arises and the "final" releasing of the RTL-SDR doesn't work when rtl_433 ends.

zuckschwerdt, 2023-07-18:
I think I know the ampmeter reading you see: Try to just disconnect and insert the RTL-SDR. The ampmeter should show the same high value. Only when rtl_433 is started at least once with clean exit the value goes very low.

It's just the "uninitialized" standby current of the RTL-SDR, as best as I know.

Not sure about the exit error. We might find that soon when I add better debugging for crashes/exits. I'll update here when that version is ready.

lhofinger, 2023-07-18:
Hello,
You are right!
- On insertion, before the first run, I see 0.166 A.
- When rtl_433-rtlsdr.exe is running: 0.255 A.
- When rtl_433-rtlsdr.exe exits without error: 0.084 A.
- When rtl_433-rtlsdr.exe exits with the error message: remains at 0.255 A.
Would be great to avoid the error and to drop to 0.084 A after each end of rtl_433-rtlsdr.exe   ;-)

zuckschwerdt, 2023-07-20:
The `PLL not locked` should always appear, it's not significant.

Thanks for the feedback on the librtlsdr. Which one was the broken one before? The one we bundle or the one from Pothos (which version)?

lhofinger, 2023-07-20:
It was broken with de bundle one in https://github.com/merbanan/rtl_433/releases/download/nightly/rtl_433-win-msvc-x64-nightly.zip

gdt, 2023-09-26:
If we believe there is a bug in rtl_433, it probably should be amended to clearly point to this bug; right now it's really hard to tell.

Overall, my read is that the problem does not manifest with up-to-date code, and therefore the issue should be closed.

zuckschwerdt, 2023-10-03:
We are using the latest (0.6.0 2018-08-26,) librtlsdr version from osmocom.

We might want to switch to a librtldsr fork. Not sure if https://github.com/rtlsdrblog/rtl-sdr-blog or https://github.com/librtlsdr/librtlsdr or something else.

We should ask for a retest when #2426 is resolved and there is likely nothing else to do.

gdt, 2023-10-03:
It would be good to detangle the librtlsdr issues and be able to recommend one.   I could consider moving pkgsrc, but I don't know the larger community consensus.  2018 is a clue that things are far from ok.

gdt, 2023-10-14:
Should this be retitled (or moved to fresh issue) to be about whether rtl_sdr should recommend a different rtlsdr fork?  If the bug is not in rtl_433 then strictly it does not belong in the tracker, and I don't see what the path forward is.

gdt, 2024-06-04:
No activity after 7.5 months after feedback tag, and seems concluded.

closed 2024-06-04 by gdt