merbanan/rtl_433 #2572: incorrect wind values for NEXUS 35.1075

michapr, 2023-07-18:
Hi,

I have seen that the wind values for my old NEXUS 35.1075 weather station are incorrect.

provided values:
{"time":"2023-07-18 13:22:03",
"protocol":42,
"model":"Hideki-Wind",
"id":6,"channel":4,"battery_ok":1,"temperature_C":29.6,
**"wind_avg_km_h":3.37962,
"wind_max_km_h":0.80467,**
"wind_approach":1,"wind_dir_deg":45.0,"mic":"CRC","mod":"ASK","freq":433.96502,"rssi":-1.1953,"snr":7.11614,"noise":-8.31143}

wind avg_km_h seem to be correct - it is the same value as provided by RFXCom adapter and the display.
wind_max_km_h is not correct, cannot see any logic here - only "0" is "0", but all other values are not...

```
rfxcom/display  <-> rtl_433
2.2  -  0
1.9  -  0.64374
0.6  -  0.64374
1.2  -  1.28748
2.2  -  4.18429
0.7  -  0.96561
1.1  -  0.96561
1.0   - 0.64374
0,7   - 1.12654
1.5   - 0.80467
1.3  -  0
1.1  -  0.96561
0   -  0
```
Please note, that rfxcom/display here is in m/s - so you should take the values *3.6

Could you recheck this - or is there anything I can do to help? Like debug something?

THANKS

zuckschwerdt, 2023-07-18:
We don't apply any scaling or conversion, avg and max (gust) are plain BCD values, e.g. see `8`, `9`,  and `A` fields here:

https://github.com/merbanan/rtl_433/blob/406918d35f4c764213adb093a400daa1776294b3/src/devices/hideki.c#L32-L35

Does RFXCom maybe smooth or derive the values somehow?

michapr, 2023-07-19:
> We don't apply any scaling or conversion, avg and max (gust) are plain BCD values, e.g. see 8, 9, and A fields here:

Yes, I already have seen it. 
That's why I thought, there maybe a mistake in calculation of the max value.

Look, this is what I receive for the sensors:
[image]

The RFXCom values are the same as on original display - and corresponds to values from a Bresser WS near my weatherstation.
I want to replace the old RFXtrx device I was using last 7+ years...

zuckschwerdt, 2023-07-19:
With `rtl_433 -R 42:v` you should get the raw codes. Can you see any mapping from that raw data to get to valid max values?

michapr, 2023-07-19:
```
# rtl_433 -R 42:v
rtl_433 version unknown inputs file rtl_tcp RTL-SDR SoapySDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/root/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Protocol [42] "HIDEKI TS04 Temperature, Humidity, Wind and Rain Sensor" does not take arguments "v"!
Registered 1 out of 175 device decoding protocols [ 42 ]
```
Protocol [42] "HIDEKI TS04 Temperature, Humidity, Wind and Rain Sensor" does not take arguments "v"!

zuckschwerdt, 2023-07-19:
The rtl_433 version you are using is too old for the raw code output then. It was added 2 years ago in https://github.com/merbanan/rtl_433/commit/f5390cc36815b453c3624ba06443d6c1e7567beb#diff-16800d810fff1b7f521bb9fe034d0e66d645cf421caac2d7bf541d44336a629c

michapr, 2023-07-19:
Hmm... have installed via apt-get install rtl-433 on debian bullseye ...
"rtl-433 is already the newest version (20.11-1)."

Looks like this is the latest version for bullseye?
https://repology.org/project/rtl-433/versions

zuckschwerdt, 2023-07-19:
Not a problem for your sensors. But to easily debug you'd need to compile form source (no need to install just run from the build dir) or pull the package from Bookworm.

michapr, 2023-07-19:
Have compiled. Should wait for some wind now... ;)

have got as sample for the weather station
codes     : {126}f9349ae668861a2186e08000056fa158

what is {126} ? - And following bytes are the raw data as in code described?

zuckschwerdt, 2023-07-19:
The 126 there is the length in bits. The codes are exactly what goes into the decoder function. You will likely need to invert the bits and unstuff the parity (every 9'th bit). BitBench can do that with a format of e.g. `8hx` (ignore) or `8h b` (show).

michapr, 2023-07-19:
Looks like the new version give better (maybe right) values... 
This looks now much better (wind_max > wind_avg now):
[image]

Will tell you next days and close the issue if all ok.

zuckschwerdt, 2023-07-19:
Oh sorry, I was looking at changes in the last 2 years. But the Bullseye version is a bit older, and there was indeed a fix shortly after that version was released: https://github.com/merbanan/rtl_433/commit/dc25708a689a6155d33ad35b80e1473f11613e6d

Mystery solved, I guess?

michapr, 2023-07-22:
looks good.... Thanks for the hint about the "old" version ;) -> solved!

closed 2023-07-22 by michapr