merbanan/rtl_433 #2573: MQTT: MQTT connect error: Unknown error -1

jjkoehorst, 2023-07-22:
Any possible way to get more information regarding this message?

> rtl_433 -f 868M -s 1024k -F mqtt://homeassistant.local:1883,user=mosquitto,pass=somepassword,retain=0,events=rtl_433[/model][/id] -F log -F json -M utc -vvv -R 142

```
rtl_433 version nightly-12-g406918d3 branch master at 202307181439 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/jasper/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...

New defaults active, use "-Y classic -s 250k" if you need the old defaults

MQTT: Publishing MQTT data to homeassistant.local port 1883
MQTT: Publishing events info to MQTT topic "rtl_433[/model][/id]".
Registering protocol [142] "Fine Offset Electronics/ECOWITT WH51, SwitchDoc Labs SM23 Soil Moisture Sensor"
Protocols: Registered 1 out of 245 device decoding protocols
Input: The internals of input handling changed, read about and report problems on PR #1978
SDR: Found 1 device(s)
SDR: trying device 0: Realtek, RTL2838UHIDIR, SN: 00000001
Detached kernel driver
Found Rafael Micro R820T tuner
SDR: Using device 0: Realtek, RTL2838UHIDIR, SN: 00000001, "Generic RTL2832U OEM"
[R82XX] PLL not locked!
SDR: Sample rate set to 1024000 S/s.
Input: Bit detection level set to 0.0 (Auto).
SDR: Tuner gain set to Auto.
Input: Reading samples in async mode...
SDR: rtlsdr_set_center_freq 868000000 = 0
SDR: Tuned to 868.000MHz.
acquire_thread: acquire_thread enter...
Allocating 15 zero-copy buffers
Baseband: low pass filter for 1024000 Hz at cutoff 204800 Hz, 4.9 us
MQTT: MQTT connect error: Unknown error -1
```

zuckschwerdt, 2023-07-22:
Did you check the quoting? If that's run through a shell then comma and brackets might have special meaning, and your pass might be mangled.

jjkoehorst, 2023-07-22:
-F 'mqtt://homeassistant.local:1883,user=mosquitto,pass=mosquitto,retain=0,events=rtl_433[/model][/id]'

but even if I type a non-existing webserver or wrong port or wrong password I get the same issues.
I am running it on a minimal 22.04 ubuntu on a raspberry pi 4 maybe a dependency is missing?

jjkoehorst, 2023-07-22:
I think that might be the issue, not sure if this library makes use of mosquito_pub or similar but that is also not yet working here...

```
mosquitto_pub -h '192.168.1.114' -u mosquitto -p mosquitto -t "test/light1" -m "hello world" -d
Error: No such file or directory
```

zuckschwerdt, 2023-07-22:
Strange that mosquitto_pub won't work. Must be some underlying cause. rtl_433 does not use libs for MQTT though.

jjkoehorst, 2023-07-22:
Might be an authentication issue or something else entirely as without any credentials at least a 

```
mosquitto_pub -h '192.168.1.114' -t "test/light1" -m "hello world" -d
Client (null) sending CONNECT
Client (null) received CONNACK (5)
Connection error: Connection Refused: not authorised.
Error: The connection was refused.
``` 
pops up! The quest continues!

gdt, 2023-07-22:
You should absolutely work through getting everything working with mosquitto_pub before trying rtl_433.   That may mean learning and understanding a fair bit.

Also, run tcpdump and watch the packets.

gdt, 2023-09-24:
Given that mosquitto_pub does not work, this issue doesn't point to a bug in rtl_433.

closed 2023-09-24 by gdt

bjanders, 2024-12-27:
I also had this issue. The problem was that when using a hostname for the MQTT server, rtl_433 will use IPv6 and the MQTT server I'm using does not listen to IPv6. It works when I use an IPv4 address in rtl_433. It would be nice if you could force rtl_433 to use IPv4, or if it would fallback to IPv4 in case IPv6 does work.

The best solution is to fix the MQTT server to listen to IPv6, but there may cases where you don't have control over this.

zuckschwerdt, 2024-12-27:
Indeed, we can't influence name resolution and it's not clear if IPv6 or IPv4 addresses will be tried.

Another problem we found is that the name resolution is strictly DNS and not mDNS, if `homeassistant.local` is the actual name then it can't work (`.local` is the mDNS TLD).

bjanders, 2024-12-29:
Thanks for the reply! Good to know that mDNS does not work. Maybe my problem actually is that I use mDNS. I did a tcpdump when troubleshooting this, but the IPv6 I saw could have been from another MQTT client and not rtl_433.

Is there a feature request for adding mDNS support?

zuckschwerdt, 2024-12-29:
We rely on Mongoose for name resolution, hacking mDNS in there is not likely to happen.

bjanders, 2024-12-29:
mDNS is convenient in non-professional networks and testing/lab environments.

I won't put in a feature request if it's unlikely to get pulled in.

gdt, 2024-12-29:
I think it would be good if mdns were tried, following how the host is configured, and then all resolved addresses were tried until one works.  That's more or less implied by #2695.  But that's a lot of work and it would need someone to design it, get consensus on the approach, build it and test it.  So far there's been no reason to think that will happen.

zuckschwerdt, 2025-01-02:
To enable mDNS name resolution for the whole host (not just rtl_433) look into /etc/nsswitch.conf.
You likely have a `hosts: files dns` line, change that to e.g. `hosts: files mdns4_minimal [NOTFOUND=return] dns` to also resolve mDNS in the `.local` zone.
This needs Avahi installed as resolver for mDNS.

bjanders, 2025-01-02:
I haven't investigated it yet, but other applications on the same host (macOS) are able to resolve, for example ping and ssh.

zuckschwerdt, 2025-01-02:
E.g. ping on macOS would be using `getaddrinfo()` https://github.com/openbsd/src/blob/master/sbin/ping/ping.c#L461
and it seems to just work.

This can be enabled in rtl_433 by using `-DMG_ENABLE_SYNC_RESOLVER=1 -DMG_ENABLE_GETADDRINFO=1` (easiest by changing [mongoose.h](https://github.com/merbanan/rtl_433/blob/master/include/mongoose.h))
But there is a bug in https://github.com/merbanan/rtl_433/blob/master/src/mongoose.c#L2842, it should be `if (mg_ncasecmp(host, "localhost", 9) == 0) {` (note **==**).
Also https://github.com/merbanan/rtl_433/blob/master/src/mongoose.c#L2710 likely should be `if ((rv = getaddrinfo(host, NULL, &hints, &servinfo)) != 0) {` (note **&hints**).

The mDNS result is found and should work. See this by e.g. adding `fprintf(stderr, "parsing result %s\n", inet_ntoa(sa->sin.sin_addr));` to line https://github.com/merbanan/rtl_433/blob/master/src/mongoose.c#L2866