merbanan/rtl_433 #2574: Support for Engelmann water meters?

okainov, 2023-07-23:
It seems like Engelmann utility water meters have some radio transmission capability on 868MHz. It would be great if rtl_433 could handle those signals out-of-the-box.

Screenshot from the manual:
[image]

Manual https://www.engelmann.de/wp-content/uploads/2023/03/1081600001_2022-12-22_BA_WaterStar_M_MID_de_en.pdf
Some docs on official website https://www.engelmann.de/en/applications/water-meter-in-practice/

I have the receiver and the water meter with the radio function activated, but I don't know how can I figure out the signals/decode them.

merbanan, 2023-07-23:
This is wmbus and should be supported out of rhe box. But I suggest you look at the wmbusmeters project instead. The have support for more sensorn.

gdt, 2023-09-26:
This is a question that was answered, so I don't see any value in keeping the issue open.

closed 2023-09-26 by gdt