merbanan/rtl_433 #2576: Add Bresser Water Leakage Sensor PN: 7009975

matthias-bs, 2023-07-24:
Hi,

I've started to decode the Bresser Water Leakage Sensor, see attached file and the following [Bitbench](https://triq.net/bitbench#c=%5BBresser%20Water%20Leak%20Sensor%2C%20PN%207009975%5D&c=C7%2070%2035%2097%2004%2008%2057%2070%2000%2000%2000%2000%2000%2000%2000%2000%2003%20FF%20FF%20FF%20FF%20FF%20FF%20FF%20FF%20FF%20%5BCH7%5D&c=DF%207D%2036%2049%2027%2009%2056%2070%2000%2000%2000%2000%2000%2000%2000%2000%2003%20FF%20FF%20FF%20FF%20FF%20FF%20FF%20FF%20FF%20%5BCH6%5D&c=9E%2030%2079%2084%2033%2006%2055%2070%2000%2000%2000%2000%2000%2000%2000%2000%2003%20FF%20FD%20DF%20FF%20BF%20FF%20DF%20FF%20FF%20%5BCH5%5D&c=37%20D8%2057%2019%2073%2002%2051%2070%2000%2000%2000%2000%2000%2000%2000%2000%2003%20FF%20FF%20FF%20FF%20FF%20BF%20FF%20EF%20FB%20%5Bset%20CH4%2C%20received%20CH1%20-%3E%20switch%20not%20positioned%20correctly%5D&c=E2%20C8%2068%2027%2091%2024%2054%2070%2000%2000%2000%2000%2000%2000%2000%2000%2003%20FF%20FF%20FF%20FF%20FF%20FF%20FF%20FF%20FF%20%5BCH4%5D&c=B3%20DA%2055%2057%2017%2040%2053%2070%2000%2000%2000%2000%2000%2000%2000%2000%2003%20FF%20FF%20FF%20FF%20FF%20FF%20FF%20FF%20FB%20%5BCH3%5D&c=37%20FA%2084%2073%2003%2002%2052%2070%2000%2000%2000%2000%2000%2000%2000%2000%2003%20FF%20FF%20FF%20DF%20FF%20FF%20FF%20FF%20FF%20%5BCH2%5D&c=27%20F3%2080%2002%2052%2088%2051%2070%2000%2000%2000%2000%2000%2000%2000%2000%2003%20FF%20FF%20FF%20FF%20FF%20DF%20FF%20FF%20FF%20%5BCH1%5D&c=A6%20FB%2080%2002%2052%2088%2059%2070%2000%2000%2000%2000%2000%2000%2000%2000%2003%20FD%20F7%20FF%20FF%20BF%20FF%20FF%20FF%20FF%20%5BCH1%2BNSTARTUP%5D&c=A6%20FB%2080%2002%2052%2088%2059%20B0%2000%2000%2000%2000%2000%2000%2000%2000%2003%20FF%20FF%20FF%20FD%20FF%20F7%20FF%20FF%20FF%20%5BCH1%2BNSTARTUP%2BALARM%5D&c=A6%20FB%2080%2002%2052%2088%2059%2070%2000%2000%2000%2000%2000%2000%2000%2000%2003%20FF%20FF%20BF%20F7%20F7%20FD%207F%20FF%20FF%20%5BCH1%2BNSTARTUP%5D&c=A6%20FB%2080%2002%2052%2088%2059%2070%2000%2000%2000%2000%2000%2000%2000%2000%2003%20FF%20FF%20FF%20FF%20FD%20EF%20FF%20FF%20FF&c=A6%20FB%2080%2002%2052%2088%2059%2070%2000%2000%2000%2000%2000%2000%2000%2000%2003%20FF%20FF%20FF%20FF%20FF%20FF%20FF%20FF%20EF&c=A6%20FB%2080%2002%2052%2088%2059%2070%2000%2000%2000%2000%2000%2000%2000%2000%2003%20FF%20FF%20FF%20FF%20FF%20FF%20FF%20FF%20FF&c=A6%20FB%2080%2002%2052%2088%2059%2070%2000%2000%2000%2000%2000%2000%2000%2000%2002%20FF%205F%20BF%20FF%20FF%20FF%20FF%20FB%2075&c=A6%20FB%2080%2002%2052%2088%2059%2070%2000%2000%2000%2000%2000%2000%2000%2000%2003%20FF%20FF%20F7%20FF%20FF%20FB%20FF%20FF%20FF&c=A6%20FB%2080%2002%2052%2088%2059%2070%2000%2000%2000%2000%2000%2000%2000%2000%2001%20FF%20F7%20FE%20FB%20FF%20FF%20BF%20FE%20DF&c=A6%20FB%2080%2002%2052%2088%2059%2070%2000%2000%2000%2000%2000%2000%2000%2000%2003%20FF%20FF%20FF%20FF%20FF%20FF%20FF%20FF%20FF&c=%5BReset%5D&c=C0%2010%2036%2079%2037%2009%2051%2070%2000%2000%2000%2000%2000%2000%2000%2000%2001%201E%20FD%20FD%20FF%20FF%20FF%20DF%20FF%20FF%20%5BCH1%5D&c=C0%2010%2036%2079%2037%2009%2051%20B0%2000%2000%2000%2000%2000%2000%2000%2000%2003%20FE%20FD%20FF%20AF%20FF%20FF%20FF%20FF%20FD%20%5BCH1%2BALARM%5D&c=%5BReset%5D&c=71%209C%2054%2081%2072%2009%2051%2040%2000%2000%2000%2000%2000%2000%2000%2000%200F%20FF%20FF%20FF%20FF%20FF%20FF%20DF%20FF%20FE%20%5BCH1%2BBATT_LO%5D&c=71%209C%2054%2081%2072%2009%2051%2040%2000%2000%2000%2000%2000%2000%2000%2000%200F%20FE%20FF%20FF%20FF%20FF%20FB%20FF%20FF%20FF&c=71%209C%2054%2081%2072%2009%2051%2040%2000%2000%2000%2000%2000%2000%2000%2000%2007%20FD%20F7%20FF%20DF%20FF%20FF%20DF%20FF%20FF&c=71%209C%2054%2081%2072%2009%2051%2080%2000%2000%2000%2000%2000%2000%2000%2000%201F%20FF%20FF%20F7%20FF%20FF%20FF%20FF%20FF%20FF%20%5BCH1%2BBATT_LO%2BALARM%5D&c=F0%2094%2054%2081%2072%2009%2059%2040%2000%2000%2000%2000%2000%2000%2000%2000%200F%20FF%20DF%20FF%20FF%20FF%20FF%20BF%20FD%20F7%20%5BCH1%2BBATT_LO%2BNSTARTUP%5D&c=F0%2094%2054%2081%2072%2009%2059%2080%2000%2000%2000%2000%2000%2000%2000%2000%2003%20FF%20B7%20FF%20ED%20FF%20FF%20FF%20DF%20FF%20%5BCH1%2BBATT_LO%2BNSTARTUP%2BALARM%5D&f=hhhh%20ID%3Ahhhhhhhh%20TYPE%3A4d%20NSTARTUP%3Ab%20CH%3A3d%20ALARM%3Ab%20NALARM%3Ab%20BATT%3Abb%20FLAGS%3Abbbb%20hhhhhhhhhhhhhhhhhhhhhhhhhhhhhhhhhhhh&cw=4).

[bresser_water_leakage_pn7009975.zip](https://github.com/merbanan/rtl_433/files/12144884/bresser_water_leakage_pn7009975.zip)

I've started from [bresser_6in1.c](https://github.com/merbanan/rtl_433/blob/master/src/devices/bresser_6in1.c). The ID seems to change on every reset/battery change. I'm quite confident about the data fields found, but I don't know how to proceed with the checksum or digest. I've tried revsum and revdgst16, but I did not manage to find anything useful. I think reducing message length appropriately could make a difference here, but again no success on my trial-and-error.

Maybe someone with more experience could give me a hint?

Thanks and regards
Matthias

zuckschwerdt, 2023-07-24:
The trailing portion doesn't look like data. Do is this FSK or OOK, what timings? Do you have a .cu8 to share?
The codes just show a very long ID and then some flags. Did you spot a randomly changing filed for the checksum somewhere?

matthias-bs, 2023-07-24:
The RF parameters, timing and message preamble part are the same as for the [6-in-1](https://github.com/merbanan/rtl_433/blob/master/src/devices/bresser_6in1.c) decoder. I thought that the first 16 bits are the checksum and the following 32 bits are the ID (consistent with the 6-in-1 protocol, but with the difference that the ID changes on reset). I presume the real data ends with the trailing zeroes.
Sorry, I do not have a .cu8, because I use my [BresserWeatherSensorReceiver](https://github.com/matthias-bs/BresserWeatherSensorReceiver/tree/main) setup instead of any SDR receiver.

matthias-bs, 2023-07-24:
Well, the first 16 bits do not seem to be a checksum (at least not an efficient one). The patterns remain the same even if the data fields change.

zuckschwerdt, 2023-07-24:
Ok, so FSK with same timings, and you already stripped the preamble (aaaa...) and sync-word (2dd4). I would also expect roughly the same format but all the bits seem to much fixed to be a checksum.

matthias-bs, 2023-08-07:
[image]
[image]
[image]
[image]
[image]

MacH-21, 2023-10-06:
Hi 
It looks like thr crc is calculated over the five bytes following the two crc bytes

C770 3597040857 
DF7D 3649270956 
9E30 7984330655 
37D8 5719730251 
E2C8 6827912454 
B3DA 5557174053 
37FA 8473030252 
27F3 8002528851 
A6FB 8002528859
C010 3679370951
719C 5481720951
F094 5481720959

3597040857 3649270956 7984330655 5719730251 6827912454 5557174053 8473030252 8002528851 8002528859 3679370951 5481720951 5481720959

C:\Users\user\Desktop>reveng -m crc-16/xmodem -c 3597040857 3649270956 7984330655 5719730251 6827912454 5557174053 8473030252 8002528851 8002528859 3679370951 5481720951 5481720959
c770
df7d
9e30
37d8
e2c8
b3da
37fa
27f3
a6fb
c010
719c
f094

@matthias-bs does U2 on the PCB have any markings, It looks like it could be a RFM119 (six pin sot-23) with a 26MHz xtal on rear of the PCB
MaCH

matthias-bs, 2023-10-07:
Hi,

@MacH-21 That's it! Thanks a lot for finding this!

I made many attempts with CRC16 algorithms, but maybe I tried it with the wrong chunks of data, the wrong polynominals or made some technical mistakes.

I could not discover an SMD marking, but it might be covered by the PCB's coating.

I will add the change to  https://github.com/merbanan/rtl_433/pull/2590 soon.

Cheers
Matthias

matthias-bs, 2023-10-08:
Updated https://github.com/merbanan/rtl_433/pull/2590

closed 2023-10-30 by matthias-bs