merbanan/rtl_433 #2577: bug: malformed json output

atlanticn, 2023-07-27:
Looking into my logs today, I found this:

`{"time" : "2023-07-26T12:20:01", "model" : "raw", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 0, "data" : ""}], "codes" : [{0}], "mod" : "FSK", "freq1" : 433.554, "freq2" : 433.663, "rssi" : -12.144, "snr" : 8.041, "noise" : -20.185}`

Notice the malformed json dictionary: "{0}". This crashes python when trying to read the json.

I commented this line to fix the problem: `// Print embedded JSON object verbatim`
https://github.com/merbanan/rtl_433/blob/d616b692a8c80d99f1931c5bfff0b7ec4cba62e1/src/output_file.c#L70

Is this the correct fix?

zuckschwerdt, 2023-07-27:
How did you get reports with zero rows?

atlanticn, 2023-07-27:
I'm getting it from the general purpose decoder, listening to packets of the air.

`-X 'n=raw,m=FSK_MC_ZEROBIT,s=8,l=16,r=32,invert,preamble={24}00002c'`

The zero rows json error occurs rarely, maybe once a week.

zuckschwerdt, 2023-07-27:
This would be the correct fix, can you try and report if that works?
```diff
--- a/src/decoder_util.c
+++ b/src/decoder_util.c
@@ -55,6 +55,10 @@ static char *bitrow_asprint_code(uint8_t const *bitrow, unsigned bit_len)
     }
     // remove last nibble if needed
     row_bytes[2 * (bit_len + 3) / 8] = '\0';
+    // print at least one '0'
+    if (bit_len == 0) {
+        sprintf(row_bytes, "0");
+    }

     // a simple bitrow representation
     row_code = malloc(8 + bit_len / 4 + 1); // "{nnnn}..\0"
```

atlanticn, 2023-07-27:
I will test it and report back in 1-2 week.

So now I should get ["{1}0"] in that case?

zuckschwerdt, 2023-07-27:
Yes, `{0}0` should be the proper representation for an empty row.

The explanation for an empty row here would be: There is a proper row but there is no valid MC contained and thus decodes to an empty row.

Is your protocol really that fast (8 µs = 125 kBit/s), we usually see (50 µs = 20kBit/s)? Also short=long for MC.

atlanticn, 2023-07-31:
Nope, it's still happening [{0}] after your proposed change.

I found the correct location, apparently flex decoder has its own printing routine...

This is the fix, tested:

```diff
--- a/src/devices/flex.c
+++ b/src/devices/flex.c
@@ -306,6 +306,10 @@ static int flex_callback(r_device *decoder, bitbuffer_t *bitbuffer)
 
     for (i = 0; i < bitbuffer->num_rows; i++) {
         print_row_bytes(row_bytes, bitbuffer->bb[i], bitbuffer->bits_per_row[i]);
+        // print at least one '0'
+        if (row_bytes[0] == '\0') {
+            sprintf(row_bytes, "0");
+        }
 
         /* clang-format off */
         row_data[i] = data_make(
```

atlanticn, 2023-08-02:
Should I send a pull request with that modification?

zuckschwerdt, 2023-08-02:
If you like, sure. But change both instances and put this second one just above line 320, the `// a simpler representation for csv output` comment.

PR #2588 (2023-08-05): fix malformed json for empty row {0} to "{0}0"
When using 
`rtl_433 -F json -X 'n=raw,m=FSK_MC_ZEROBIT'`
sometimes rtl_433 outputs json containing an invalid dictionary:
`"codes" : [{0}]`

Fix the source to output "{0}0" in case of an empty row.

Closes #2577
closed 2023-08-05 by zuckschwerdt