merbanan/rtl_433 #2580: Feature idea - SigMF output

adulau, 2023-07-30:
I was wondering if there were already some discussions to have a [SigMF output](https://www.iqengine.org/sigmf) for `rtl_433`.

zuckschwerdt, 2023-07-30:
Yes, I have already seen this. The format isn't hard to produce/consume (generally the same as SigRok).

I wondering about the ecosystem though. Are there interesting SigMF-compatible apps? So far I've seen more interest in /support for [H5](https://en.wikipedia.org/wiki/Hierarchical_Data_Format).

adulau, 2023-07-30:
We are working on a [MISP module for SigMF](https://github.com/MISP/misp-modules/pull/628) for sharing analysis and sample via [MISP](https://www.misp-project.org/).  As we are manually crafting the file for interesting sample from rtl_433, having a direct output for SigMF would save time and make it easier.  I bet it would be useful for all the others tools such as [gr-sigmf](https://github.com/skysafe/gr-sigmf). So for signal analysts, it would be really useful.

zuckschwerdt, 2023-09-22:
There is some early and very basic support in #2632 now.
Closing in favor of discussion in the PR itself.

closed 2023-09-22 by zuckschwerdt