merbanan/rtl_433 #2587: rtl_433 seems to stop working within first 24 hours, restarting it seems to 'fix' temporarily

dimatx, 2023-08-03:
Hi there, I'm pretty confident the issue is my setup. Would appreciate some suggestions on how to troubleshoot. Please let me know what additional info would be helpful provide.

It runs reliably for several hours, and then seems to stop working. Once it stops working, nothing is being written to the logs. Stopping and starting it again seems to resolve it temporarily.

I'm running the rtl_433 (nightly-12-g406918d3 branch master at 202307181439) on BeagleBoneBlack with Nooelec NESDR SMArt v5 with the following arguments:

`rtl_433 -Fmqtt:mqtt.*********.com:1883,user=mqtt,pass=*************,retain=1,devices=rtl_433[/model][/id],events=rtl_433[/model][/id] -H 100 -f 433.92M -f 915M -C si -Y autolevel -R 41 -R 142 -R 192 -R 231 -R 228 -Fkv -v -M level -M noise`

I am running it as a daemon (/etc/systemd/system/rtl433.service) like this:

```
[Unit]
Description=rtl_433 SDR Receiver Daemon
After=network-online.target

[Service]
ExecStart=/usr/local/bin/rtl_433 -Fmqtt:mqtt.*********.com:1883,user=mqtt,pass=*************,retain=1,devices=rtl_433[/model][/id],events=rtl_433[/model][/id] -H 100 -f 433.92M -f 915M -C si -Y autolevel -R 41 -R 142 -R 192 -R 231 -R 228 -Fkv -v -M level -M noise
Restart=always

[Install]
WantedBy=multi-user.target
```

zuckschwerdt, 2023-08-03:
Nothing strange there. Better place `-F kv` last and `-v` first.
I guess you don't really want `kv` as it will just pollute the systemd logs. Using `-F log` would be recommended.

About the suddenly missing reception: Do you still get the "level" and hoppig notices? If so maybe the sender drifted out of band? Did you check that the reception is not too close to the band edges?

dimatx, 2023-08-03:
Thanks for the suggestions. When it stopped working there was nothing in the logs since the day prior.

I'll try it now with:
`rtl_433 -v -Fmqtt:mqtt.******.com:1883,user=mqtt,pass=**********,retain=1,devices=rtl_433[/model][/id],events=rtl_433[/model][/id] -H 100 -f 433.92M -f 915M -C si -Y autolevel -R 41 -R 142 -R 192 -R 231 -R 228 -M level -M noise -F log`

dimatx, 2023-08-04:
It stopped working after 1hr (It's 21:00 here at the time of grabbing that status). Here's what the daemon status looks like:

```
debian@BeagleBone:~$ sudo systemctl status rtl433
[sudo] password for debian: 
● rtl433.service - rtl_433 SDR Receiver Daemon
     Loaded: loaded (/etc/systemd/system/rtl433.service; enabled; vendor preset: enabled)
     Active: active (running) since Thu 2023-08-03 18:06:48 EDT; 2h 53min ago
   Main PID: 7246 (rtl_433)
      Tasks: 3 (limit: 1027)
     Memory: 4.6M
        CPU: 16min 27.435s
     CGroup: /system.slice/rtl433.service
             └─7246 /usr/local/bin/rtl_433 -v -Fmqtt:mqtt.*********.com 1883 user mqtt pass ************ retain 1 devices rtl_433[/model][/id] events rtl_433[/model][>

Aug 03 19:00:20 BeagleBone rtl_433[7246]: Auto Level: Current noise level -15.7 dB, estimated noise -15.8 dB
Aug 03 19:00:30 BeagleBone rtl_433[7246]: Auto Level: Current noise level -15.7 dB, estimated noise -15.7 dB
Aug 03 19:00:40 BeagleBone rtl_433[7246]: Auto Level: Current noise level -15.8 dB, estimated noise -15.7 dB
Aug 03 19:00:50 BeagleBone rtl_433[7246]: Auto Level: Current noise level -15.8 dB, estimated noise -15.8 dB
Aug 03 19:01:00 BeagleBone rtl_433[7246]: Auto Level: Current noise level -15.7 dB, estimated noise -15.7 dB
Aug 03 19:01:06 BeagleBone rtl_433[7246]: Auto Level: Estimated noise level is -15.1 dB, adjusting minimum detection level to -12.1 dB
Aug 03 19:01:10 BeagleBone rtl_433[7246]: Auto Level: Current noise level -15.7 dB, estimated noise -15.7 dB
Aug 03 19:01:20 BeagleBone rtl_433[7246]: Auto Level: Current noise level -15.7 dB, estimated noise -15.7 dB
Aug 03 19:01:30 BeagleBone rtl_433[7246]: Auto Level: Current noise level -15.7 dB, estimated noise -15.8 dB
Aug 03 19:01:40 BeagleBone rtl_433[7246]: Auto Level: Current noise level -15.8 dB, estimated noise -15.7 dB
```

dimatx, 2023-08-04:
Seems to still have issues running as a daemon, but just running from command line does not seem to experience the issue I had (been running for 12 hours now) 🤔. Are there any more resilient/reliable ways run things as a service than my method?

zuckschwerdt, 2023-08-04:
Is it maybe logging too much? Then try without `F log`
Or you could use a workaround like `-T 1h` to quit every hour.

dimatx, 2023-08-04:
Thank you, will try that. And the `-T 1h`, if used in the systemd daemon, it will restart the service if it quits?

zuckschwerdt, 2023-08-04:
With `Restart=always` systemd should do that.

dimatx, 2023-08-05:
Thanks. Without any logging and no -v it seems to be working, so must be just too much logging for the device.

closed 2023-08-05 by dimatx