merbanan/rtl_433 #2591: rtl_433_mqtt_hass: MQTT entity name starts with device name (Home Assistant 2023.8.0)

noahwilliamsson, 2023-08-09:
As of Home Assistant 2023.8.0, the `examples/rtl_433_mqtt_hass.py` script causes Home Assistant to log warnings such as:

```
WARNING (MainThread) [homeassistant.components.mqtt.mixins] 
  MQTT entity name starts with the device name in your config
  {
    'device_class': <SensorDeviceClass.HUMIDITY: 'humidity'>,
    'name': 'Bresser-3CH-1-176-H',
    'unit_of_measurement': '%',
    'value_template': Template<template=({{ value|float }}) renders=0>,
    'state_class': <SensorStateClass.MEASUREMENT: 'measurement'>,
    'state_topic': 'rtl_433/devices/Bresser-3CH/1/176/humidity',
    'unique_id': 'Bresser-3CH-1-176-H',
    'device': {
      'identifiers': ['Bresser-3CH-1-176'],
      'name': 'Bresser-3CH-1-176',
      'model': 'Bresser-3CH',
      'manufacturer': 'rtl_433',
      'connections': []
    },
    'force_update': False,
    'qos': 0,
    'enabled_by_default': True,
    'payload_not_available': 'offline',
    'encoding': 'utf-8',
    'availability_mode': 'latest',
    'payload_available': 'online'
  }
  , this is not expected. Please correct your configuration.
  The device name prefix will be stripped off the entity name and becomes '-UTC'
```

There's a bug report at https://github.com/home-assistant/core/issues/97656 for a different Home Assistant integration.  Via links in that report, I found:

- [PSA: MQTT Name changes in 2023.8](https://community.home-assistant.io/t/psa-mqtt-name-changes-in-2023-8/598099)

There's an example of how they want things to be named in:

- https://community.home-assistant.io/t/psa-mqtt-name-changes-in-2023-8/598099/14

The thread also refers to the MQTT breaking changes re: naming in the Home Assistant 2023.8.0 release notes.

- https://www.home-assistant.io/blog/2023/08/02/release-20238/#breaking-changes (see the MQTT section)

zuckschwerdt, 2023-08-10:
If I read that right the first `'name': 'Bresser-3CH-1-176-H',` should now be just `'name': 'H',`?

Does this need to be a toggle? This feels like it won't be compatible with pre HA 2023.8.0?

@noahwilliamsson PR if you like and it's clear what to do. Also maybe @unverbraucht can help?

gdt, 2023-08-10:
I'm unclear on the details and will try to have a look -- but my take is that rtl_433 master branch need only be compatible with the most recent release of HA.  rtl_433 doesn't even have releases on a regular basis, so it seems beyond reasonable to try to maintain compat with old HA releases.   People who want to run old HA can run old scripts from the repo.   But I think the right thing to do is update HA regularly, for all the usual "run the latest stable release" reasons.

zuckschwerdt, 2023-08-10:
Sounds reasonable. (sidenote: something in HA broke for me when upgrading from 2023.4.6 and I rolled back and didn't update since)

unverbraucht, 2023-08-11:
Yes, seems that we need to drop the device name from the entity_name, in the release notes they say:

This means any MQTT entity which is part of a device will [automatically have its friendly_name attribute prefixed with the device name](https://developers.home-assistant.io/docs/core/entity/#has_entity_name-true-mandatory-for-new-integrations).

Today I'll look into the other issue #2554  , maybe I can take a stab at both.

unverbraucht, 2023-08-11:
Actualli it looks like HA already goes ahead and removes the device name with the current release. This is latest HA with no changes from my side:

[image]

BTW would it make sense to rename entities to something more verbose like "temperature" instead of "T"? The original topics that RTL433 create are already named "temperature" and so on, so we could reuse these.

I wonder if

PR #2594 (2023-08-11): fix(#2591): Remove device name from Home Assistant entities
Fixes #2591 . 
I also propose to give the entities a longer name than we do currently. Since HA will strip the device name, our suffix -T becomes a device called "-T". Instead I propose that we use the human readable sensor name.

This would then look like this with current HA:
[image]
[image]
commit 9e0fb41e (2023-08-11): Fix HA script remove device name from entities (#2594)

* Fixes #2591: Remove device name from entity name, use more readable entity names
* Remove changes that slipped in from  https://github.com/merbanan/rtl_433/pull/2593
closed 2023-08-11 by zuckschwerdt