merbanan/rtl_433 #2599: Reolink doorbell and La Crosse TX141TH-BCH collision

segdy, 2023-08-15:
Previously I used rtl433 with DSC Security Sensors, Acurite temperature sensors and my Reolink doorbell. Its config is:

```
# Reolink doorbell
#
# This decoder reads button presses from a EV1527 based Reolink doorbell.
#
# Analyzed by @sdalu in #2277

decoder {
        name        = Reolink-Doorbell,
        modulation  = OOK_PWM,
        short       = 352,
        long        = 956,
        tolerance   = 242,
        gap         = 872,
        reset       = 9360,
        bits       >= 24,
        rows       >= 30,
        unique,
        get         = @0:{24}:id,
}
```

Now I got a Lacrosse TX141TH-BCH and now I always receive the Lacrosse and Reolink at the same time:

```
time      : 1692081571.923782
model     : LaCrosse-TX141THBv2                    Sensor ID : 2f
Channel   : 0            Battery   : 1             Temperature: 26.80 C      Humidity  : 60 %          Test?     : No            Integrity : CRC
Modulation: ASK          Freq      : 434.0 MHz
RSSI      : -0.1 dB      SNR       : 22.8 dB       Noise     : -22.9 dB
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 1692081571.923782
model     : Reolink-Doorbell                       count     : 12            num_rows  : 49            len       : 41            data      : ffffffffff0
id        : 16777215
Modulation: ASK          Freq      : 434.0 MHz
RSSI      : -0.1 dB      SNR       : 22.8 dB       Noise     : -22.9 dB
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 1692081572.906844
```

Command line:

```
/usr/local/bin/rtl_433 -F mqtt://10.227.4.10,user=rtl433,pass=xxx,retain=0 -F kv -M time:unix:usec:utc -M level -M noise -M stats -C si -R 23 -R 11 -R 73 -c /usr/local/etc/rtl_433/Reolink-Doorbell.conf
```

Is there a way to make the Doorbell not trigger by the TX141TH-BCH ?

zuckschwerdt, 2023-08-15:
Yes, observe the code for your doorbell and add it to the config, e.g.
```
        match         = {25}abcdef8,
```

closed 2023-08-15 by zuckschwerdt

segdy, 2023-08-18:
Thank you, perfect!