merbanan/rtl_433 #2601: Produce raw output that can be used by rtl_wmbus

testuser7, 2023-08-15:
Is it possible to produce output that can be read by `rtl_wmbus`?
I was trying:
`rtl_433 -R 0 -X n=izar,m=FSK_PCM,s=10,l=10,r=500 -Y squelch -f 868.95M -w - | rtl_wmbus -p s -a`

This one is working on the same dongle:
`rtl_sdr -f 868.95M -s 1.6e6 - 2>/dev/null | rtl_wmbus -p s -a`

Binary output from `rtl_sdr` is much different than from `rtl_433` with `-w` option.

zuckschwerdt, 2023-08-15:
Default output for `kv` (screen output) is stdout. You are mixing both text and sample data together.
Also you'd need that `-s 1.6e6` for rtl_433 too (`-s 1600k`). So either
`rtl_433 -R 0 -f 868.95M -s 1600k -w - | rtl_wmbus -p s -a`
or
`rtl_433 -R 0 -X n=izar,m=FSK_PCM,s=10,l=10,r=500 -Y squelch -f 868.95M -s 1600k -F kv:output.txt -w - | rtl_wmbus -p s -a`

closed 2023-08-15 by zuckschwerdt

testuser7, 2023-08-15:
Thanks for the clarification.
If I want to use a general purpose decoder and don't want to save any file additionally this command doesn't work:
`rtl_433 -R 0 -X n=izar,m=FSK_PCM,s=10,l=10,r=500 -Y squelch -f 868.95M -s 1600k -w - | rtl_wmbus -p s -a`
I have to add the `-F null` option for it to work:
`rtl_433 -R 0 -X n=izar,m=FSK_PCM,s=10,l=10,r=500 -Y squelch -f 868.95M -s 1600k -F null -w - | rtl_wmbus -p s -a`

However, the `rtl_433 -R 0 -f 868.95M -s 1600k -w - | rtl_wmbus -p s -a` command works fine.

zuckschwerdt, 2023-08-16:
If you want to see the KV output from the decoder on screen you could also redirect to stderr (as only stdout is needed for the raw output): `-F kv:/dev/stderr`