merbanan/rtl_433 #2604: Segmentation fault on Raspberry Pi Zero W

dimatx, 2023-08-18:
Trying to migrate a working setup to a Raspberry Pi Zero W. Using [this](https://www.loverpi.com/products/loverpi-microusb-to-4-port-usb-otg-hub) OTG USB Hub as the Pi Zero only has a microUSB port. Getting a segmentation fault and hoping to see if anyone has troubleshooting suggestions. rtl_433 and syslog below. Thanks!
```
pi@rpizw0:/var/log $ rtl_433 -vvvv
rtl_433 version nightly-2-g4f152360 branch master at 202308160920 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/pi/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registering protocol [1] "Silvercrest Remote Control"
Registering protocol [2] "Rubicson, TFA 30.3197 or InFactory PT-310 Temperature Sensor"
Registering protocol [3] "Prologue, FreeTec NC-7104, NC-7159-675 temperature sensor"
Registering protocol [4] "Waveman Switch Transmitter"
Registering protocol [8] "LaCrosse TX Temperature / Humidity Sensor"
Registering protocol [10] "Acurite 896 Rain Gauge"
Registering protocol [11] "Acurite 609TXC Temperature and Humidity Sensor"
Registering protocol [12] "Oregon Scientific Weather Sensor"
Registering protocol [15] "KlikAanKlikUit Wireless Switch"
Registering protocol [16] "AlectoV1 Weather Sensor (Alecto WS3500 WS4500 Ventus W155/W044 Oregon)"
Registering protocol [17] "Cardin S466-TX2"
Registering protocol [18] "Fine Offset Electronics, WH2, WH5, Telldus Temperature/Humidity/Rain Sensor"
Registering protocol [19] "Nexus, FreeTec NC-7345, NX-3980, Solight TE82S, TFA 30.3209 temperature/humidity sensor"
Registering protocol [20] "Ambient Weather F007TH, TFA 30.3208.02, SwitchDocLabs F016TH temperature sensor"
Registering protocol [21] "Calibeur RF-104 Sensor"
Registering protocol [22] "X10 RF"
Registering protocol [23] "DSC Security Contact"
Registering protocol [25] "Globaltronics GT-WT-02 Sensor"
Registering protocol [26] "Danfoss CFR Thermostat"
Registering protocol [29] "Chuango Security Technology"
Registering protocol [30] "Generic Remote SC226x EV1527"
Registering protocol [31] "TFA-Twin-Plus-30.3049, Conrad KW9010, Ea2 BL999"
Registering protocol [32] "Fine Offset Electronics WH1080/WH3080 Weather Station"
Registering protocol [33] "WT450, WT260H, WT405H"
Registering protocol [34] "LaCrosse WS-2310 / WS-3600 Weather Station"
Registering protocol [35] "Esperanza EWS"
Registering protocol [36] "Efergy e2 classic"
Registering protocol [38] "Generic temperature sensor 1"
Registering protocol [39] "WG-PB12V1 Temperature Sensor"
Registering protocol [40] "Acurite 592TXR Temp/Humidity, 592TX Temp, 5n1 Weather Station, 6045 Lightning, 899 Rain, 3N1, Atlas"
Registering protocol [41] "Acurite 986 Refrigerator / Freezer Thermometer"
Registering protocol [42] "HIDEKI TS04 Temperature, Humidity, Wind and Rain Sensor"
Registering protocol [43] "Watchman Sonic / Apollo Ultrasonic / Beckett Rocket oil tank monitor"
Registering protocol [44] "CurrentCost Current Sensor"
Registering protocol [45] "emonTx OpenEnergyMonitor"
Registering protocol [46] "HT680 Remote control"
Registering protocol [47] "Conrad S3318P, FreeTec NC-5849-913 temperature humidity sensor"
Registering protocol [48] "Akhan 100F14 remote keyless entry"
Registering protocol [49] "Quhwa"
Registering protocol [50] "OSv1 Temperature Sensor"
Registering protocol [51] "Proove / Nexa / KlikAanKlikUit Wireless Switch"
Registering protocol [52] "Bresser Thermo-/Hygro-Sensor 3CH"
Registering protocol [53] "Springfield Temperature and Soil Moisture"
Registering protocol [54] "Oregon Scientific SL109H Remote Thermal Hygro Sensor"
Registering protocol [55] "Acurite 606TX Temperature Sensor"
Registering protocol [56] "TFA pool temperature sensor"
Registering protocol [57] "Kedsum Temperature & Humidity Sensor, Pearl NC-7415"
Registering protocol [58] "Blyss DC5-UK-WH"
Registering protocol [59] "Steelmate TPMS"
Registering protocol [60] "Schrader TPMS"
Registering protocol [63] "Efergy Optical"
Registering protocol [67] "Radiohead ASK"
Registering protocol [68] "Kerui PIR / Contact Sensor"
Registering protocol [69] "Fine Offset WH1050 Weather Station"
Registering protocol [70] "Honeywell Door/Window Sensor, 2Gig DW10/DW11, RE208 repeater"
Registering protocol [71] "Maverick ET-732/733 BBQ Sensor"
Registering protocol [73] "LaCrosse TX141-Bv2, TX141TH-Bv2, TX141-Bv3, TX141W, TX145wsdth, (TFA, ORIA) sensor"
Registering protocol [74] "Acurite 00275rm,00276rm Temp/Humidity with optional probe"
Registering protocol [75] "LaCrosse TX35DTH-IT, TFA Dostmann 30.3155 Temperature/Humidity sensor"
Registering protocol [76] "LaCrosse TX29IT, TFA Dostmann 30.3159.IT Temperature sensor"
Registering protocol [77] "Vaillant calorMatic VRT340f Central Heating Control"
[... 197 lines cut ...]
Aug 18 18:47:14 rpizw0 kernel: [  640.018229] usb 1-1.4: dvb_usb_v2: will use the device's hardware PID filter (table count: 32)
Aug 18 18:47:14 rpizw0 kernel: [  640.018322] dvbdev: DVB: registering new adapter (Realtek RTL2832U reference design)
Aug 18 18:47:14 rpizw0 kernel: [  640.018358] usb 1-1.4: media controller created
Aug 18 18:47:14 rpizw0 kernel: [  640.018776] dvbdev: dvb_create_media_entity: media entity 'dvb-demux' registered.
Aug 18 18:47:14 rpizw0 kernel: [  640.065658] i2c i2c-11: Added multiplexed i2c bus 12
Aug 18 18:47:14 rpizw0 kernel: [  640.065714] rtl2832 11-0010: Realtek RTL2832 successfully attached
Aug 18 18:47:14 rpizw0 kernel: [  640.065870] usb 1-1.4: DVB: registering adapter 0 frontend 0 (Realtek RTL2832 (DVB-T))...
Aug 18 18:47:14 rpizw0 kernel: [  640.065934] dvbdev: dvb_create_media_entity: media entity 'Realtek RTL2832 (DVB-T)' registered.
Aug 18 18:47:14 rpizw0 kernel: [  640.068538] r820t 12-001a: creating new instance
Aug 18 18:47:14 rpizw0 kernel: [  640.084126] r820t 12-001a: Rafael Micro r820t successfully identified, chip type: R820T
Aug 18 18:47:14 rpizw0 kernel: [  640.109924] rtl2832_sdr rtl2832_sdr.0.auto: Registered as swradio0
Aug 18 18:47:14 rpizw0 kernel: [  640.109962] rtl2832_sdr rtl2832_sdr.0.auto: Realtek RTL2832 SDR attached
Aug 18 18:47:14 rpizw0 kernel: [  640.109974] rtl2832_sdr rtl2832_sdr.0.auto: SDR API is still slightly experimental and functionality changes may follow
Aug 18 18:47:14 rpizw0 kernel: [  640.118634] Registered IR keymap rc-empty
Aug 18 18:47:14 rpizw0 kernel: [  640.118863] rc rc0: Realtek RTL2832U reference design as /devices/platform/soc/20980000.usb/usb1/1-1/1-1.4/rc/rc0
Aug 18 18:47:14 rpizw0 kernel: [  640.129441] rc rc0: lirc_dev: driver dvb_usb_rtl28xxu registered at minor = 0, raw IR receiver, no transmitter
Aug 18 18:47:14 rpizw0 kernel: [  640.129726] input: Realtek RTL2832U reference design as /devices/platform/soc/20980000.usb/usb1/1-1/1-1.4/rc/rc0/input6
Aug 18 18:47:14 rpizw0 kernel: [  640.132349] usb 1-1.4: dvb_usb_v2: schedule remote query interval to 200 msecs
Aug 18 18:47:14 rpizw0 kernel: [  640.141748] usb 1-1.4: dvb_usb_v2: 'Realtek RTL2832U reference design' successfully initialized and connected
Aug 18 18:47:14 rpizw0 mtp-probe: checking bus 1, device 13: "/sys/devices/platform/soc/20980000.usb/usb1/1-1/1-1.4"
Aug 18 18:47:14 rpizw0 mtp-probe: bus: 1, device: 13 was not an MTP device
Aug 18 18:47:14 rpizw0 systemd-udevd[842]: event0: Process '/usr/sbin/th-cmd --socket /var/run/thd.socket --passfd --udev' failed with exit code 1.
Aug 18 18:47:14 rpizw0 mtp-probe: checking bus 1, device 13: "/sys/devices/platform/soc/20980000.usb/usb1/1-1/1-1.4"
Aug 18 18:47:14 rpizw0 mtp-probe: bus: 1, device: 13 was not an MTP device
```

dimatx, 2023-08-21:
Not happening without a USB hub, so must be some issue with it.

closed 2023-08-21 by dimatx

zuckschwerdt, 2023-08-21:
The crash is likely an unrelated problem on shutdown -- I still need to find the time to locate that problem.