merbanan/rtl_433 #2606: Issues with

pyrodex, 2023-08-20:
Referencing this posting in the group from last year:

https://groups.google.com/g/rtl_433/c/Is2bSWStKlc

I purchased a few of these from AliExpress and am not having the same luck as previously reported.

Here is my configuration:

`
# Ansible Managed file, please do not edit locally!
output kv
hop_interval  60
frequency     433.92M
#frequency     315.00M
frequency     345M
#frequency     868.00M
frequency     915M
convert     si
report_meta time:iso:usec:tz
report_meta level
report_meta noise
report_meta stats
report_meta protocol

# To disable comment out the protocol line
# List was generated on 2023-08-20 from current version
protocol 01 #   Silvercrest Remote Control
protocol 02 #   Rubicson, TFA 30.3197 or InFactory PT-310 Temperature Sensor
protocol 03 #   Prologue, FreeTec NC-7104, NC-7159-675 temperature sensor
protocol 04 #   Waveman Switch Transmitter
#protocol 06 # * ELV EM 1000
#protocol 07 # * ELV WS 2000
protocol 08 #   LaCrosse TX Temperature / Humidity Sensor
protocol 10 # * Acurite 896 Rain Gauge
protocol 11 #   Acurite 609TXC Temperature and Humidity Sensor
protocol 12 #   Oregon Scientific Weather Sensor
#protocol 13 # * Mebus 433
#protocol 14 # * Intertechno 433
protocol 15 #   KlikAanKlikUit Wireless Switch
protocol 16 #   AlectoV1 Weather Sensor (Alecto WS3500 WS4500 Ventus W155/W044 Oregon)
protocol 17 #   Cardin S466-TX2
protocol 18 #   Fine Offset Electronics, WH2, WH5, Telldus Temperature/Humidity/Rain Sensor
protocol 19 #   Nexus, FreeTec NC-7345, NX-3980, Solight TE82S, TFA 30.3209 temperature/humidity sensor
protocol 20 #   Ambient Weather F007TH, TFA 30.3208.02, SwitchDocLabs F016TH temperature sensor
protocol 21 #   Calibeur RF-104 Sensor
protocol 22 #   X10 RF
protocol 23 #   DSC Security Contact
#protocol 24 # * Brennenstuhl RCS 2044
protocol 25 #   Globaltronics GT-WT-02 Sensor
protocol 26 #   Danfoss CFR Thermostat
protocol 29 #   Chuango Security Technology
protocol 30 #   Generic Remote SC226x EV1527
protocol 31 #   TFA-Twin-Plus-30.3049, Conrad KW9010, Ea2 BL999
protocol 32 #   Fine Offset Electronics WH1080/WH3080 Weather Station
protocol 33 #   WT450, WT260H, WT405H
protocol 34 #   LaCrosse WS-2310 / WS-3600 Weather Station
protocol 35 #   Esperanza EWS
protocol 36 #   Efergy e2 classic
protocol 37 # * Inovalley kw9015b, TFA Dostmann 30.3161 (Rain and temperature sensor)
protocol 38 #   Generic temperature sensor 1
protocol 39 #   WG-PB12V1 Temperature Sensor
protocol 40 #   Acurite 592TXR Temp/Humidity, 5n1 Weather Station, 6045 Lightning, 899 Rain, 3N1, Atlas
protocol 41 #   Acurite 986 Refrigerator / Freezer Thermometer
protocol 42 #   HIDEKI TS04 Temperature, Humidity, Wind and Rain Sensor
protocol 43 #   Watchman Sonic / Apollo Ultrasonic / Beckett Rocket oil tank monitor
protocol 44 #   CurrentCost Current Sensor
protocol 45 #   emonTx OpenEnergyMonitor
protocol 46 #   HT680 Remote control
protocol 47 #   Conrad S3318P, FreeTec NC-5849-913 temperature humidity sensor
protocol 48 #   Akhan 100F14 remote keyless entry
[... 178 lines cut ...]
    long=968,
    reset=916,
    gap=0,
    tolerance=261,
    bits=33,
    unique,
    get=Location:@0:{16}:[62080:HotWaterHeater 5853:KevinSink 7133:KitchenSink],
    get=Message:@16:{8}:[250:Alarm],
    get=Battery:@24:{4}:[14:Ok 12:Low]
}
`

I am testing with a unit that via rtl_433 reports the following:

`
time      : 2023-08-20T12:04:19.304477-0400        Protocol  : 30
model     : Generic-Remote                         House Code: 62080
Command   : 205          Tri-State : 110XX000101Z
Modulation: ASK          Freq      : 433.9 MHz
RSSI      : -0.1 dB      SNR       : 13.3 dB       Noise     : -13.5 dB
`

Would appreciate the assistance on getting this resolved. 

Thanks!

zuckschwerdt, 2023-08-20:
If you can: grab a sample file, check there is a good signal and upload here as zip. See https://triq.org/rtl_433/ANALYZE.html

pyrodex, 2023-08-20:
These help?

https://www.icloud.com/iclouddrive/008kV4vl1UfQMNEUCJP2dNeYA#g003%5F433.92M%5F250k

https://www.icloud.com/iclouddrive/06euv-z3y0neBntG4gCcGuiew#g002%5F433.92M%5F250k

pyrodex, 2023-08-25:
> If you can: grab a sample file, check there is a good signal and upload here as zip. See https://triq.org/rtl_433/ANALYZE.html

Here you go.

[Archive.zip](https://github.com/merbanan/rtl_433/files/12439275/Archive.zip)

gdt, 2023-10-15:
@pyrodex What's the status of this issue?  Are you still trying to work on a decoder?

pyrodex, 2023-10-15:
I’ve been trying but for side tracked by another project of course…

klohner, 2024-01-09:
I took a look at this signal and got the same result @zuckschwerdt saw.  This does not look like the same kind of signal as seen in the "DY-SQ100B Water Leakage Sensor" referred to in [this message thread].(https://groups.google.com/g/rtl_433/c/Is2bSWStKlc)

You can use a flex decoder like this on your signal:

`rtl_433 -R0 -X "n=WaterLeak,m=OOK_PCM,s=512,l=512,g=5000,r=15000,bits=106,symbol_zero={4}e,symbol_one={4}8,unique,repeats>=5,get=CODE:@1:{24}:%06x" -r g002_433.92M_250k.cu8`

```
model     : WaterLeak    count     : 7             num_rows  : 10            len       : 25            data      : 8780328       DATA      : 0f0065
```

You might need to tweak the gap (`g=`) value, and maybe tweak or remove the `,unique,repeats>=5` if this isn't sensitive enough in your environment.

The CODE "get" value assumes an initial sync bit instead of a trailing bit in case that seems more useful for you.

Here's a version formatted for rtl_433.conf:

```
decoder {
    name=MyWaterLeak,
    modulation=OOK_PCM,
    short=512,
    long=512,
    gap=5000,
    reset=15000,
    bits=106,
    symbol_zero={4}e,
    symbol_one={4}8,
    unique,
    repeats>=5,
    get=CODE:@1:{24}:%06x
}

```

gdt, 2024-06-04:
Is someone going to make a PR to put that flex decoder in the sources?  What do we need to do to close this issue?