merbanan/rtl_433 #2608: libtool: link: require no space between `-L' and `-L' When using make and python3.11 configuration

charlesdsmith, 2023-08-20:
When I try to run `make` after configuring the Makefile for python3.11 I get the following error: 

`/usr/lib64/apr-1/build/libtool --silent --mode=link gcc -Wl,-z,relro,-z,now   -o src/server/mod_wsgi.la  -rpath /usr/lib64/httpd/modules -module -avoid-version    src/server/wsgi_validate.lo src/server/wsgi_thread.lo src/server/wsgi_stream.lo src/server/wsgi_server.lo src/server/wsgi_restrict.lo src/server/wsgi_metrics.lo src/server/wsgi_memory.lo src/server/wsgi_logger.lo src/server/wsgi_interp.lo src/server/wsgi_daemon.lo src/server/wsgi_convert.lo src/server/wsgi_buckets.lo src/server/wsgi_apache.lo src/server/mod_wsgi.lo -L -L -lpython
libtool: link: require no space between `-L' and `-L'
apxs:Error: Command failed with rc=65536`

Any help with this? Thank you

charlesdsmith, 2023-08-20:
I went into the MakeFile and removed the space between `-L` and `-L` but now I'm getting `cannot find -lpython`

charlesdsmith, 2023-08-21:
Now i've edited the MakeFile so it could find python, these are the parts I edited:
`APXS = /usr/bin/apxs
PYTHON = /usr/local/lib/python3.11

DESTDIR =
LIBEXECDIR = /usr/lib64/httpd/modules

CPPFLAGS =
CFLAGS =  -Wc,-g -Wc,-O2
LDFLAGS =  -L/usr/lib -L/usr/local/lib/python3.11 -L/usr/local/bin/python3.11-config
LDLIBS =  -lpython3

SRCFILES = src/server/mod_wsgi.c src/server/wsgi_*.c

all : src/server/mod_wsgi.la`

 but when I try to load the module into Apache I get the error `httpd: Syntax error on line 57 of /etc/httpd/conf/httpd.conf: Cannot load modules/mod_wsgi.so into server: /etc/httpd/modules/mod_wsgi.so: undefined symbol: PyCMethod_New`

Does mod_wsgi definitely work with python3.11? I downloaded it here: https://www.python.org/ftp/python/3.11.4/Python-3.11.4.tgz 

Is that correct?

gdt, 2023-08-21:
This content in my view doesn't belong in an issue, as it seems overwhelmingly likely that you have a misconfiguration or a local system problem rather than a bug.  I suggest using the mailinglist or opening a Discussion.

When you do, you'll need to explain far more about your environment.   Operating system, version, cpu type, rtl_433 version (should be git master), and how you are building.

Also there is no wsgi in the rtl-433 sources!

closed 2023-08-21 by zuckschwerdt