merbanan/rtl_433 #2609: Cannot get sensor working with Home Assistant

msw90, 2023-08-21:
Hello all,

I have invested at least 20 hours trying to get this "random" rain sensor working which I ordered online. It is sending on 433 MHz and I can see data in rtl_433 in my Home Assistant, but I have no idea how to make it work as a "sensor" in Home Assistant. [https://de.aliexpress.com/item/4000761757290.html?spm=a2g0o.productlist.main.113.40a11a4ai4ApmT&algo_pvid=7268c84f-3ff7-4f75-b2ec-817f087b593c&algo_exp_id=7268c84f-3ff7-4f75-b2ec-817f087b593c-56&pdp_npi=4%40dis%21EUR%219.10%217.74%21%21%219.67%21%21%4021038edc16926521674732478eb75c%2110000007566136247%21sea%21DE%212572487528%21&curPageLogUid=4xN4NsvgHRqx](url)

This is what I get when touching the sensor to trigger it: Detected OOK package	2023-08-21 23:10:49
Analyzing pulses...
Total count:  454,  width: 607.34 ms		(151834 S)
Pulse width distribution:
 [ 0] count:   13,  width: 5448 us [5144;6996]	(1362 S)
 [ 1] count:   61,  width: 1512 us [1432;1828]	( 378 S)
 [ 2] count:  323,  width:  644 us [568;784]	( 161 S)
 [ 3] count:   18,  width: 3436 us [3092;4068]	( 859 S)
 [ 4] count:   37,  width: 2360 us [2252;2708]	( 590 S)
 [ 5] count:    1,  width:  980 us [980;980]	( 245 S)
 [ 6] count:    1,  width:    0 us [0;0]	(   0 S)
Gap width distribution:
 [ 0] count:  289,  width:  156 us [128;192]	(  39 S)
 [ 1] count:   54,  width:  112 us [92;128]	(  28 S)
 [ 2] count:   72,  width:  208 us [196;268]	(  52 S)
 [ 3] count:   11,  width:   60 us [56;68]	(  15 S)
 [ 4] count:    4,  width:   44 us [44;48]	(  11 S)
 [ 5] count:   10,  width: 1572 us [1428;1748]	( 393 S)
 [ 6] count:    2,  width:  276 us [260;292]	(  69 S)
 [ 7] count:   10,  width:   80 us [76;84]	(  20 S)
 [ 8] count:    1,  width:  376 us [376;376]	(  94 S)
Pulse period distribution:
 [ 0] count:   14,  width: 5692 us [5516;7160]	(1423 S)
 [ 1] count:   60,  width: 1644 us [1632;1684]	( 411 S)
 [ 2] count:  321,  width:  800 us [792;844]	( 200 S)
 [ 3] count:   23,  width: 3660 us [3276;4164]	( 915 S)
 [ 4] count:   35,  width: 2476 us [2444;2524]	( 619 S)
Pulse timing distribution:
 [ 0] count:   13,  width: 5448 us [5144;6996]	(1362 S)
 [ 1] count:   71,  width: 1520 us [1428;1828]	( 380 S)
 [ 2] count:  323,  width:  644 us [568;784]	( 161 S)
 [ 3] count:   18,  width: 3436 us [3092;4068]	( 859 S)
 [ 4] count:   37,  width: 2360 us [2252;2708]	( 590 S)
 [ 5] count:    1,  width:  980 us [980;980]	( 245 S)
 [ 6] count:    1,  width:    0 us [0;0]	(   0 S)
 [ 7] count:  289,  width:  156 us [128;192]	(  39 S)
 [ 8] count:   54,  width:  112 us [92;128]	(  28 S)
 [ 9] count:   72,  width:  208 us [196;268]	(  52 S)
 [10] count:   11,  width:   60 us [56;68]	(  15 S)
 [11] count:    4,  width:   44 us [44;48]	(  11 S)
 [12] count:    2,  width:  276 us [260;292]	(  69 S)
 [13] count:   10,  width:   80 us [76;84]	(  20 S)
 [14] count:    1,  width:  376 us [376;376]	(  94 S)
Level estimates [high, low]:  14925,   1636
RSSI: -0.4 dB SNR: 9.6 dB Noise: -10.0 dB
Frequency offsets [F1, F2]:     626,      0	(+2.4 kHz, +0.0 kHz)
Guessing modulation: No clue...
Detected OOK package	2023-08-21 23:10:50
Analyzing pulses...
Total count:   85,  width: 112.41 ms		(28102 S)
Pulse width distribution:
 [ 0] count:   59,  width:  632 us [588;700]	( 158 S)
 [ 1] count:   13,  width: 1468 us [1428;1532]	( 367 S)
 [ 2] count:    6,  width: 2332 us [2304;2388]	( 583 S)
 [ 3] count:    4,  width: 3164 us [3128;3204]	( 791 S)
 [ 4] count:    3,  width: 4628 us [4316;5188]	(1157 S)
Gap width distribution:
 [ 0] count:   71,  width:  172 us [140;216]	(  43 S)
 [ 1] count:   10,  width:  116 us [100;132]	(  29 S)
 [ 2] count:    1,  width: 1424 us [1424;1424]	( 356 S)
 [ 3] count:    1,  width:  336 us [336;336]	(  84 S)
 [ 4] count:    1,  width:   88 us [88;88]	(  22 S)
Pulse period distribution:
 [ 0] count:   59,  width:  800 us [796;804]	( 200 S)
 [ 1] count:   13,  width: 1640 us [1636;1648]	( 410 S)
 [ 2] count:    6,  width: 2476 us [2476;2484]	( 619 S)
 [ 3] count:    4,  width: 3320 us [3316;3332]	( 830 S)
 [ 4] count:    2,  width: 5668 us [5524;5812]	(1417 S)
Pulse timing distribution:
 [ 0] count:   59,  width:  632 us [588;700]	( 158 S)
 [ 1] count:   14,  width: 1464 us [1424;1532]	( 366 S)
 [ 2] count:    6,  width: 2332 us [2304;2388]	( 583 S)
 [ 3] count:    4,  width: 3164 us [3128;3204]	( 791 S)
 [ 4] count:    3,  width: 4628 us [4316;5188]	(1157 S)
 [ 5] count:   71,  width:  172 us [140;216]	(  43 S)
 [ 6] count:   10,  width:  116 us [100;132]	(  29 S)
 [ 7] count:    1,  width:  336 us [336;336]	(  84 S)
 [ 8] count:    1,  width:   88 us [88;88]	(  22 S)
 [ 9] count:    1,  width: 51884 us [51884;51884]	(12971 S)
Level estimates [high, low]:  15043,   1660
RSSI: -0.4 dB SNR: 9.6 dB Noise: -9.9 dB
Frequency offsets [F1, F2]:     338,      0	(+1.3 kHz, +0.0 kHz)
Guessing modulation: No clue...

It sucks that I have absolutely no clue or any other idea what to do, but maybe someone is able to help me.

Thanks!

AutomGuy, 2023-08-22:
it seems to be, a decoder is not available yet. 
with which options did you start rtl_433?

Best Regards

AutomGuy, 2023-08-22:
maybe christian @zuckschwerdt  can tell you which flex decoder parameters you need for rtl 433.
and afterwards, we assume that you received encoded datas you can go forward to analyze the data.
BR

msw90, 2023-08-22:
Thanks for your response! I tried with "generic motion sensor" which is decoder 87, but I also did not know if I need to create a custom flex decoder... but it seems that the analyzer does not find enough information about the modulation for that.

AutomGuy, 2023-08-22:
but motion sensor decoder 87 means IMHO a movement of a person or animal in front of the sensor. not in the sense of movement of a drive.

zuckschwerdt, 2023-08-22:
Aside from the dominat ~650 µs pulses there are many other lengths, so it could be PCM and not just simple PWM.
We need samples to look at. Grab some and upload as zip here. https://triq.org/rtl_433/ANALYZE.html

msw90, 2023-08-22:
Again, Thank you for taking time to look into this, I really appreciate it!

Some additional background: I bought this sensor standalone but it is usually paired with a motor to open/close windows. It even has a pairing button at the bottom, but I have not use for that. I was hoping that the signal the sensor sends on contact with water can be used by my Home Assistant somehow.

During my numerous tests in the past days the sensor was logged as "Motion Sensor" sometimes, which made me think I should use protocol 87 for this. However, I do not know how to replicate this currently.

I also use rtl_433 to receive data from my weather station on 868M.

msw90, 2023-08-22:
> Aside from the dominat ~650 µs pulses there are many other lengths, so it could be PCM and not just simple PWM. We need samples to look at. Grab some and upload as zip here. https://triq.org/rtl_433/ANALYZE.html

I will try to capture the samples and will provide them.

msw90, 2023-08-22:
rtl_433 tells me that many files were written, but I cannot find any file on my Home Assistant system (proxmox). I will keep looking

AutomGuy, 2023-08-23:
i get the feeling you are not familar with rtl_433. i assume you use a linux box. open a terminal -> create a new folder named "rain_sensor" for example. go into that direcetory and start rtl_433 as described in document [ https://triq.org/rtl_433/ANALYZE.html]( https://triq.org/rtl_433/ANALYZE.html) under "grab a sample". after a while quit rtl_433 with ctrl-c. there should be some .. .cu8 files in this directory -> put the files into a zip and upload the zip file here.

gdt, 2023-09-24:
I retitled this.  Currently this issue doesn't really have information that could be used to solve what's requested.

gdt, 2023-10-14:
Is there a path forward?  So far there isn't decoding information that needs to be captured.

klohner, 2024-01-09:
@msw90 - Any luck creating .cu8 files from this sensor?  Would you be able to zip and attach them to this issue?

gdt, 2024-06-04:
This isn't making any progress and there isn't really useful information in the PR for the next person.  Feel free to ask for reopening if actually active.

closed 2024-06-04 by gdt

msw90, 2024-11-25:
Sorry for being dormant for a while, just got back to this issue today and have some news. 
I start rtl433 only with mqtt and kv output defined, nothing else. Now I get the following log when touching the rain sensor:
`_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
[rtl_433] time      : 2024-11-25 22:22:17+0100
[rtl_433] model     : Generic-Motion                         code      : 7ffe7
[rtl_433] _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ ` 
I would like to make some use of this data in HASS by making this a sensor which triggers, but have no idea how to do it. Any ideas?

msw90, 2024-11-25:
Adding some more information, because I need additional support. I want to receive weather data on 868M, which works if I enable the required protocol (173). However, I also want to receive data from the rain sensor on 433M. I have added hopping betweeen frequencies, however I need to add the rain sensor as a decoder, correct?
This is what I get, but cannot figure out how to make a decode out of it. Any support available?

[rtl_433] Detected OOK package	2024-11-25 23:52:22+0100
[rtl_433] {"time" : "2024-11-25 23:52:22+0100", "model" : "Generic-Motion", "code" : "fffff"}
[rtl_433] Analyzing pulses...
[rtl_433] Total count:  628,  width: 719.28 ms		(179821 S)
[rtl_433] Pulse width distribution:
[rtl_433]  [ 0] count:   12,  width: 5068 us [4996;5124]	(1267 S)
[rtl_433]  [ 1] count:   70,  width: 1212 us [1096;1340]	( 303 S)
[rtl_433]  [ 2] count:  223,  width:  264 us [252;328]	(  66 S)
[rtl_433]  [ 3] count:   22,  width: 2180 us [1936;2552]	( 545 S)
[rtl_433]  [ 4] count:    6,  width: 3140 us [2956;3240]	( 785 S)
[rtl_433]  [ 5] count:  189,  width:  428 us [336;508]	( 107 S)
[rtl_433]  [ 6] count:   32,  width: 1564 us [1520;1628]	( 391 S)
[rtl_433]  [ 7] count:   74,  width:  720 us [680;796]	( 180 S)
[rtl_433] Gap width distribution:
[rtl_433]  [ 0] count:  294,  width:  516 us [412;548]	( 129 S)
[rtl_433]  [ 1] count:   63,  width:  140 us [116;160]	(  35 S)
[rtl_433]  [ 2] count:  218,  width:  360 us [288;416]	(  90 S)
[rtl_433]  [ 3] count:   11,  width: 1716 us [1584;1772]	( 429 S)
[rtl_433]  [ 4] count:   19,  width:   96 us [80;112]	(  24 S)
[rtl_433]  [ 5] count:   12,  width:   68 us [56;76]	(  17 S)
[rtl_433]  [ 6] count:   10,  width:   44 us [44;52]	(  11 S)
[rtl_433] Pulse period distribution:
[rtl_433]  [ 0] count:   13,  width: 5476 us [4976;5524]	(1369 S)
[rtl_433]  [ 1] count:  101,  width: 1652 us [1636;1684]	( 413 S)
[rtl_433]  [ 2] count:  479,  width:  804 us [796;840]	( 201 S)
[rtl_433]  [ 3] count:   26,  width: 2476 us [2452;2520]	( 619 S)
[rtl_433]  [ 4] count:    8,  width: 3640 us [3320;4140]	( 910 S)
[rtl_433] Pulse timing distribution:
[rtl_433]  [ 0] count:   12,  width: 5068 us [4996;5124]	(1267 S)
[rtl_433]  [ 1] count:   70,  width: 1212 us [1096;1340]	( 303 S)
[rtl_433]  [ 2] count:  257,  width:  272 us [252;336]	(  68 S)
[rtl_433]  [ 3] count:   31,  width: 2052 us [1672;2552]	( 513 S)
[rtl_433]  [ 4] count:    6,  width: 3140 us [2956;3240]	( 785 S)
[rtl_433]  [ 5] count:  568,  width:  468 us [336;548]	( 117 S)
[rtl_433]  [ 6] count:   34,  width: 1568 us [1520;1628]	( 392 S)
[rtl_433]  [ 7] count:   74,  width:  720 us [680;796]	( 180 S)
[rtl_433]  [ 8] count:   63,  width:  140 us [116;160]	(  35 S)
[rtl_433]  [ 9] count:   19,  width:   96 us [80;112]	(  24 S)
[rtl_433]  [10] count:   12,  width:   68 us [56;76]	(  17 S)
[rtl_433]  [11] count:   99,  width:  352 us [332;376]	(  88 S)
[rtl_433]  [12] count:   10,  width:   44 us [44;52]	(  11 S)
[rtl_433]  [13] count:    1,  width: 51244 us [51244;51244]	(12811 S)
[rtl_433] Level estimates [high, low]:  13929,    673
[rtl_433] RSSI: -0.7 dB SNR: 13.2 dB Noise: -13.9 dB
[rtl_433] Frequency offsets [F1, F2]:    1447,      0	(+5.5 kHz, +0.0 kHz)
[rtl_433] Guessing modulation: No clue...

zuckschwerdt, 2024-11-26:
That still looks strange and like PCM. You need to capture a .cu8 file and upload as zip here.

msw90, 2024-11-27:
Sure, I just tried that by adding this to my config file: write_file Regensensor.cu8
However, I cannot find the file anywhere in my Home Assistant virtual machine - where is the file saved?

zuckschwerdt, 2024-11-27:
That is not recommended, you are capturing the whole SDR data stream, a very big file, by default 500 kB /s.

Read https://triq.org/rtl_433/ANALYZE.html then use `-S unknown` The files are saved in the startup folder, likely the user home directory.