merbanan/rtl_433 #2611: Documentation : list devices compatible with drivers

psa-jforestier, 2023-08-25:


zuckschwerdt, 2023-08-25:
Yes we still need such a driver-support "database". Marking this in code is currently the only choice.

I have a half-finished support/experience (price, quality, …) page where users can contribute such infos -- I hope to finish it soon-ish.

psa-jforestier, 2023-08-25:
Good news ! it should be a huge work to do such a database. Let us know when you want user contribution to enlarge this DB.

ProfBoc75, 2023-08-25:
Salut @psa-jforestier : yes it's some kind of archeology , I did it with Emax w6 (device 215) . Optex is a french rebrand of chineese products , but all around the world you have lot of other subbrands doing the same. So it's very difficult to get compatibility except to wait and track feedback from users. This takes times, and sometime it's not fully compatible, same protocol but missing a sensor, like weather station with or without wind gust or the uv/lux... This adding another level of complexity. But good idea to get such compatibility matrix. The approach could be like rfxcom user guide where compatible devices are listed and link to a protocol, here is decoder.

gdt, 2024-06-04:
This issue now includes which-frequency-is-used-by-device in addition to which-device-is-decoded-by-driver, enabling us to close #917

closed 2024-06-04 by gdt