merbanan/rtl_433 #2614: http web interface: changing center frequency with the webui do not work in some case

psa-jforestier, 2023-08-25:
Changing the center frquency with the web interface is not working if we use the drop down menu.

To reproduce :  
- start `rtl_433_32bit_static.exe -F http -F log -v` (I use the windows nightly build 202308141346)
- start a private browsing session (on my side : Firefox 102, language for content set to [fr])
- open http://localhost:8433
- wait a few second for the page to load the rxui content, wait for "connected" to be green.
- note the center frequency : `433 920 000 Hz`
- If i click on the the down arrow to popup the drop-down menu of frequencies, I see number are formatted like this :
```
433.920.000 Hz
868.000.000 Hz
315.000.000 Hz
...
```
- When selecting the second entry in the menu (`868.000.000 Hz`), it replace the value by "868" and on the rtl_433 console I see 
`SDR: Tuned to 868.000000.`
- **And nothing is decoded, probably because the decoder is centered in 868Hz**.
- From the web UI, I click on the Center Frequency field and this time I enter by hand the frequency `868000000[enter]' without space and without using the autompletion, and it works.
- Each time I select a frequency from the dropdown menu, it tune to a wrong value, after a while rtl_433 is not responding anymore (need to ^C)

The same happends for the Sample Rate. Changing the default browser language to [en] do not fix the problem.

Maybe the rxui is not suppose to be at production level, but I find it really usfull for monitorning. And I did not find the Git repo of the web rxui.

zuckschwerdt, 2023-08-25:
This must have broken with one of the last major framework (VueJS) upgrades. I'll take a look now.

zuckschwerdt, 2023-08-25:
Looks like some other problem with the HTTP/WS api. The number decoding works and the tuning is ok.
I still need to find time to debug a likely deadlock in the new threading code. The old release version (22.11) should work.

gdt, 2024-06-04:
@psa-jforestier Can you retest with updated code?

psa-jforestier, 2024-06-11:
Hello.
Seems not fixed.
To reproduce : 
```
C:> rtl_433_64bit_static.exe  -F http -F log -v
rtl_433 version -128-NOTFOUND branch  at 202404231952 inputs file rtl_tcp RTL-SDR
HTTP server: Starting HTTP server at 0.0.0.0 port 8433
HTTP server: Serving HTTP-API on address 0.0.0.0:8433, serving .
(...)
SDR: Tuned to 433.920MHz.
Baseband: low pass filter for 250000 Hz at cutoff 25000 Hz, 40.0 us
```
Then I start FireFox 126 in a Private Browser session. I set the Language setting of FF to English(US) (confirmed by looking at the http headers : Accept-Language=en-US,fr;q=0.5 ).

I go to http://localhost:8433/ . It loads https://triq.org/rxui/js/app.375e3b31.js and chunk-vendors.6385070e.js

The RxUi indicate center frequency is "433,920,000  Hz". The drop-down list have value like "868.000.000 Hz" :
[image]

I select the first value on the drop down list ("433.920.000 Hz"), then rtl_433 tune to 433 Hz :
[image]

It is possible to enter the frequency in RxUi but only by typing the full frequency with all digits. Selecting from the drop down list do not work.

It seems there is confusion somewhere in the JS code when handling digits, but the code is minified I'm not able to deep debug on it.

gdt, 2024-06-11:
Thanks for testing; feedback label dropped :)

ProfBoc75, 2024-06-11:
I guess it could be regional settings issue between thousand and decimal separator.
With english language settings, the decimal is the dot . and the thousand is the coma, for french, this is the reverse.
As mentioned by @psa-jforestier , firefox is set to FR language and could explain the behavior.

zuckschwerdt, 2024-06-11:
Parser code is this
```js
class NumberParser {
  constructor(locale) {
    locale = locale || (new Intl.NumberFormat()).resolvedOptions().locale
    const numberFormat = new Intl.NumberFormat(locale)
    if (numberFormat.formatToParts) {
      const parts = new Intl.NumberFormat(locale).formatToParts(12345.6)
      this._group = new RegExp(`[${parts.find(d => d.type === 'group').value}]`, 'g')
      this._decimal = new RegExp(`[${parts.find(d => d.type === 'decimal').value}]`)
    } else { // Fallback
      const number = 123456.789
      const numberStr = numberFormat.format(number)
      const found = numberStr.match(/123(.*)456(.*)789/)
      this._group = new RegExp(`[${found[1]}]`, 'g')
      this._decimal = found[2]
    }
  }
  parse(string) {
    string = string.toString().trim().toUpperCase()
    let scale = 1
    if (string.includes('K')) {
        scale = 1e3
    } else if (string.includes('M')) {
        scale = 1e6
    } else if (string.includes('G')) {
        scale = 1e9
    }
    string = string
      .replace(this._group, '')
      .replace(this._decimal, '.')
      .replace('K', '.')
      .replace('M', '.')
      .replace('G', '.')
    return parseFloat(string) * scale
  }
}

const defaultNumberParser = new NumberParser()

export { NumberParser, defaultNumberParser }
```

psa-jforestier, 2024-06-11:
I noticed this error is not related the browser language setting, because I have the same behavior wherever the browser is set in French or English. As you noticed, it is more related to the operating system setting used by the browser (my OS is in French).
The `utils/NumberParser.parse()` function failed because the `locale` is not properly detected (it is valued to "fr-FR" even if I set the browser language to "fr"). This little lib seems bugged at some point.
RxUi is not working also with Edge (browser in French), same problem.

zuckschwerdt, 2024-06-11:
I've pushed an update of rxui, selection should work a little better now. The preset numbers now use spaces instead of dot or comma.

psa-jforestier, 2024-06-11:
Yes it is fixed now, thank you very much !

gdt, 2024-06-11:
I guess rxui is in another repo so this bug never belonged here ;-)  But it seems fixed.

closed 2024-06-11 by gdt