merbanan/rtl_433 #2617: Fineoffset-WH2 - No protocol

ots3, 2023-09-04:
Hi! 

Thanks for making and maintaining this fabulous software. I have been using it for several years.

I have some sensors (Teldus FT007TH) that RTL_433 recognize as "Fineoffset-WH2". But it does not print protocol or channel (1-8) information. 

This has been like this for a while. Compiled and verified with latest version today.

Regards, OT

{"time":"2023-09-04 13:41:09","model":"Fineoffset-WH2","id":151,"temperature_C":25.2,"humidity":34,"mic":"CRC","mod":"ASK","freq":433.9369,"rssi":-1.41325,"snr":16.12197,"noise":-17.5352}

Also "Ford TPMS" protocol 89 gets printed as readable not as a single line.
 ["protocol"]=>
  int(89)
  ["model"]=>
  string(4) "Ford"
  ["type"]=>
  string(4) "TPMS"

zuckschwerdt, 2023-09-04:
The WH2 protocol has no channel data. Run with `rtl_433 -R 18:vv` and paste the codes here to see if your Teldus has more data.

You must be using some other software on top of rtl_433, as we don't have any output with `=>` symbols.

ots3, 2023-09-04:
Thanks! I use rtl_433 to post json to a syslog-server that inserts values into mariadb. I made various applications to facilitate different tasks on top of this. 

"-R 18:vv -M level" output:
{"time":"2023-09-04 15:00:56","model":"Fineoffset-WH2","id":247,"temperature_C":24.5,"humidity":34,"mic":"CRC","mod":"ASK","freq":433.99482,"rssi":-0.102188,"snr":16.51533,"noise":-16.6175}

merbanan, 2023-09-04:
Follow the instructions/info from this ticket.

https://github.com/merbanan/rtl_433/issues/1561

zuckschwerdt, 2023-09-04:
You would need a somewhat current rtl_433 version compiled from git sources. And the output then should be e.g.
```
codes     : {48}ff4d10f62138
```

Not needed anymore with the mentioned ticket ;)

Ok, I get what you meant by "Ford TMPS" vs "Ford"+"TPMS" -- that's actually correct. You would need to concat `type` + " " + `model` to get "Ford TMPS".

ots3, 2023-09-04:
OK! Sorry for that! My mistake. Here is the output generated with "-R 18:vv" from two different sensors.

time      : 2023-09-04 16:00:56
model     : Fineoffset-WH2                         ID        : 247
Temperature: 24.4 C      Humidity  : 34 %          Integrity : CRC
[pulse_slicer_pwm] Fine Offset Electronics, WH2, WH5, Telldus Temperature/Humidity/Rain Sensor
codes     : {48}ff4f70f4223e
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2023-09-04 16:01:09
model     : Fineoffset-WH2                         ID        : 151
Temperature: 25.2 C      Humidity  : 34 %          Integrity : CRC
[pulse_slicer_pwm] Fine Offset Electronics, WH2, WH5, Telldus Temperature/Humidity/Rain Sensor
codes     : {48}ff4970fc2200

merbanan, 2023-09-04:
Use the same sensor and change the channel setting. Somehow it should change the output pattern of the data.

ots3, 2023-09-05:
So from what I can see, the channel is reflected in the ID's bit 5,6,7. The ID is not random as I see on other sensors.

I can conclude that I should not use these sensors, as multiple sensors on same channel will represent a problem.

ch 1:  135 / h:  87 / b: 10000111
ch 2:  151 / h:  97 / b: 10010111
ch 3:  167 / h:  A7 / b: 10100111
ch 4:  183 / h:  B7 / b: 10110111
ch 5:  199 / h:  C7 / b: 11000111
ch 6:  215 / h:  D7 / b: 11010111
ch 7:  231 / h:  E7 / b: 11100111
ch 8:  247 / h:  F7 / b: 11110111

Channel 1:
--------------------------------------------------------------------------------------------------
time      : 2023-09-05 08:53:52
model     : Fineoffset-WH2                         ID        : 135
Temperature: 27.1 C      Humidity  : 39 %          Integrity : CRC
[pulse_slicer_pwm] Fine Offset Electronics, WH2, WH5, Telldus Temperature/Humidity/Rain Sensor
codes     : {48}ff48710f271d

Channel 2:
--------------------------------------------------------------------------------------------------
time      : 2023-09-05 09:05:14
model     : Fineoffset-WH2                         ID        : 151
Temperature: 28.5 C      Humidity  : 39 %          Integrity : CRC
[pulse_slicer_pwm] Fine Offset Electronics, WH2, WH5, Telldus Temperature/Humidity/Rain Sensor
codes     : {48}ff49711d2731

Channel 3:
--------------------------------------------------------------------------------------------------
time      : 2023-09-05 09:04:09
model     : Fineoffset-WH2                         ID        : 167
Temperature: 28.3 C      Humidity  : 36 %          Integrity : CRC
[pulse_slicer_pwm] Fine Offset Electronics, WH2, WH5, Telldus Temperature/Humidity/Rain Sensor
codes     : {48}ff4a711b24a4

Channel 4:
--------------------------------------------------------------------------------------------------
time      : 2023-09-05 09:03:14
model     : Fineoffset-WH2                         ID        : 183
Temperature: 28.2 C      Humidity  : 36 %          Integrity : CRC
[pulse_slicer_pwm] Fine Offset Electronics, WH2, WH5, Telldus Temperature/Humidity/Rain Sensor
codes     : {48}ff4b711a24cb

Channel 5:
--------------------------------------------------------------------------------------------------
time      : 2023-09-05 09:01:09
model     : Fineoffset-WH2                         ID        : 199
Temperature: 27.8 C      Humidity  : 36 %          Integrity : CRC
[pulse_slicer_pwm] Fine Offset Electronics, WH2, WH5, Telldus Temperature/Humidity/Rain Sensor
codes     : {48}ff4c711624ed

Channel 6:
--------------------------------------------------------------------------------------------------
time      : 2023-09-05 09:05:58
model     : Fineoffset-WH2                         ID        : 215
Temperature: 28.7 C      Humidity  : 39 %          Integrity : CRC
[pulse_slicer_pwm] Fine Offset Electronics, WH2, WH5, Telldus Temperature/Humidity/Rain Sensor
codes     : {48}ff4d711f27e6

Channel 7:
--------------------------------------------------------------------------------------------------
time      : 2023-09-05 09:06:33
model     : Fineoffset-WH2                         ID        : 231
Temperature: 28.8 C      Humidity  : 40 %          Integrity : CRC
[pulse_slicer_pwm] Fine Offset Electronics, WH2, WH5, Telldus Temperature/Humidity/Rain Sensor
codes     : {48}ff4e7120287f

Channel 8:
--------------------------------------------------------------------------------------------------
time      : 2023-09-05 08:46:32
model     : Fineoffset-WH2                         ID        : 247
Temperature: 25.2 C      Humidity  : 36 %          Integrity : CRC
[pulse_slicer_pwm] Fine Offset Electronics, WH2, WH5, Telldus Temperature/Humidity/Rain Sensor
codes     : {48}ff4f70fc24af

merbanan, 2023-09-05:
That was a pretty clear investigation. Can you trigger low battery somehow also ? I am guessing that is bit 8. And with bits 1-4 having a fixed pattern we should be able to detect the Teldus FT007TH model and map the id against the channel.

Can you submit a patch for this ?

zuckschwerdt, 2023-09-05:
Do we know if the timing/pattern is distinct? Should it be detected as "Fineoffset-TelldusProove"?

ots3, 2023-09-05:
Not able to detect any change in the ID-bit 8 with low battery. 

{48}ff 4b 71 1a 24 cb Normal Voltage
{48}ff 4b 70 fd 22 f3 Low Voltage

ots3, 2023-09-05:
time      : 2023-09-05 11:14:38
model     : Fineoffset-WH2                         ID        : 183
Temperature: 25.3 C      Humidity  : 34 %          Integrity : CRC
[pulse_slicer_pwm] Fine Offset Electronics, WH2, WH5, Telldus Temperature/Humidity/Rain Sensor
codes     : {48}ff4b70fd22f3
[pulse_slicer_pwm] Fine Offset Electronics, WH2, WH5, Telldus Temperature/Humidity/Rain Sensor
codes     : {0}0, {17}fe968
[pulse_slicer_pwm] Fine Offset Electronics, WH2, WH5, Telldus Temperature/Humidity/Rain Sensor
codes     : {0}0, {29}87e91798

gdt, 2023-09-24:
What is this issue about now?  It seems like the WH2 driver could be improved, but the issue title doesn't say that.

metayan, 2024-05-13:
When running `rtl_433` `23.11-103-g9064ccd0` with
```sh
rtl_433 -M utc -M protocol -M level -M stats -M hires -g 100 -F json:$(date +%Y%m%d-%H%M%S).json
```
the protocol number is not output.

### Decoded Frames
```json
{
  "time": "2023-04-29 04:35:27.862308",
  "model": "Fineoffset-WH2",
  "id": 231,
  "temperature_C": 21.7,
  "mic": "CRC",
  "mod": "ASK",
  "freq": 433.893,
  "rssi": -11.224,
  "snr": 26.149,
  "noise": -37.373
}
{
  "time": "2023-04-29 04:35:33.135195",
  "model": "Fineoffset-WH2",
  "id": 135,
  "temperature_C": 16,
  "humidity": 63,
  "mic": "CRC",
  "mod": "ASK",
  "freq": 433.893,
  "rssi": -3.056,
  "snr": 36.078,
  "noise": -39.134
}
```

### Stats
```json
{
  "time": "2023-04-29 04:37:52.216120",
  "enabled": 211,
  "since": "2023-04-29T04:27:52",
  "frames": {
    "count": 42,
    "fsk": 0,
    "events": 41
  },
  "stats": [
    {
      "device": 0,
      "name": "Fine Offset Electronics, WH2, WH5, Telldus Temperature/Humidity/Rain Sensor",
      "events": 272,
      "ok": 2,
      "messages": 2,
      "abort_length": 270
    },
    {
      "device": 50,
      "name": "OSv1 Temperature Sensor",
      "events": 39,
      "ok": 39,
      "messages": 39
    }
  ]
}
```

For comparison (one of the frames that appears in the stats above as device 50):
```json
{
  "time": "2023-04-29 04:35:51.873855",
  "protocol": 50,
  "model": "Oregon-v1",
  "id": 0,
  "channel": 1,
  "battery_ok": 1,
  "temperature_C": 3.8,
  "mic": "CHECKSUM",
  "mod": "ASK",
  "freq": 433.823,
  "rssi": -0.195,
  "snr": 31.949,
  "noise": -32.144
}
```

All devices that appear in my logs reveal the protocol, obeying `-M protocol`. Except the "Fineoffset-WH2".

gdt, 2024-06-04:
This is a dup of #2543.

closed 2024-06-04 by gdt