merbanan/rtl_433 #2619: Support for Govee Water/Leak sensor

ifuchs, 2023-09-12:
I have several Govee H5054 water leak sensors that I would like to bring into Home Assistant. rtl_433 Autodiscovery is seeing the device but only shows the battery status. My guess is that the key of interest is "event" which shows if there is a leak or if the button on the device is pressed. Home Assistant is adding the device but only with the battery_ok status. Could "event" be added as a valid key and would this solve the problem I am seeing?

ifuchs, 2023-09-12:
@zuckschwerdt OK, well I already have the Govee water sensors but that last comment does not give much hope of making it work in HA. It would appear that all I can do is return them. Sigh.

gdt, 2023-09-12:
@ifuchs I think you can probably make them work,but you'll need to code up "if wet event then wet, and if no wet event for 30 minutes then dry" somehow.   I don't have any and am understanding only from reading list/github traffic.   So take my comments in that light.

ifuchs, 2023-09-12:
Can someone explain to me why when I run the rtl_433 (rtl_433 version nightly-33-gf775d1ac branch master at 202304221552 inputs file rtl_tcp RTL-SDR) on my raspberry pi with mqtt going to HA, the Govee device is seen (although I am still trying to get the payload recognized) but if I simply run rtl_433 from the CLI on my Mac w v22.11, the Govee does not show up. Did this work in the older RPi version and break in the most recent version?

ifuchs, 2023-09-13:
Further simplified failure: rtl_433 on the RPi0W shows the Govee when button pressed. Same command on Mac does not. Mac is running v22.11. FWIW, GQRX and URH both see the Govee transmissions.

gdt, 2023-09-13:
22.11 is ancient.   compile git master.
(also this belongs as discussion or mailing list, not as an issue, IMHO)

ifuchs, 2023-09-13:
Sorry, but if 22.11 is ancient why is that what shows up on the Github page as the latest release?

[image]

zuckschwerdt, 2023-09-13:
It's a year old, we still need to debug an rare exit failure before we release. Check the nightly changelog for interesting changes. E.g.

> Add support for new revision of Govee H5054 water leak detector (https://github.com/merbanan/rtl_433/pull/2273)

ifuchs, 2023-09-13:
OK, I upgraded to the master branch and my Mac with rtl_sdr now sees the Govee (as does the Rpi0W.) Now if I could just get it to work with HA I could go on to something else. I will refrain form using issues for this sort of thing in the future. Thanks.

deviantintegral, 2023-09-15:
@ifuchs if you switch to the “next” addon that ships rtl_433 master. It’s the same idea with the auto discovery addon.

https://github.com/pbkhrv/rtl_433-hass-addons/tree/main/rtl_433-next

ifuchs, 2023-09-17:
I replaced the rtl_433_mqtt_hass.py file in the rtl_433_mqtt_autodiscovery addin (by using portainer and going into the addin container. I am not clear what else needs to be replaced.I don't run the rtl_433 addin, I run rtl_433 on a Raspberry Pi to send MQTT messages to HA. If I casue the Govee sensor to send a water leak message, it shows up in the HA MQTT broker (I see that with MQTT explorer on my Mac) but I do not see any entity in HA for this event. What am I missing? Do I need to replace rtl_433 on my Raspberry Pi? (HA does show Govee device, but no entity for the event.) Any help would be very much appreciated.

ifuchs, 2023-09-17:
FWIW, I updated rtl_433 on the RPI0w to 202308211241 with no noticeable change.

gdt, 2023-10-15:
At least something has been merged and I can't immediately understand what is left to do.  Should this issue be closed?  If not, please add a comment with a very brief summary of the remaining work.

theshaww, 2024-01-18:
Has anyone gotten these leak detectors to work?  I just want it to notify me when there's a leak so I can flip off my water pump breaker with HA.

closed 2024-01-18 by zuckschwerdt