merbanan/rtl_433 #2621: [Input] Input device start failed, exiting!

endmarsfr, 2023-09-16:
Hi,
I compiled latest source ([70d84d0](https://github.com/merbanan/rtl_433/commit/70d84d01e1be87b459f7a10825966f3262b7dd34))
Compilation worked fine, but when i tried to run rtl_433, I had an error : [Input] Input device start failed, exiting!

`dietpi@radio:~/share/rtl_433/build/src$ ./rtl_433 -d 1
rtl_433 version -128-NOTFOUND branch master at 202308211241 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/dietpi/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
[Protocols] Registered 212 out of 246 device decoding protocols [ 1-4 8 10-12 15-17 19-23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119-121 124-128 130-149 151-161 163-168 170-175 177-197 199 201-215 217-228 230-232 234-241 243-244 246 ]
Detached kernel driver
Found Rafael Micro R820T tuner
[SDR] Using device 1: Realtek, RTL2838UHIDIR, SN: 00000001, "Generic RTL2832U OEM"
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Allocating 15 zero-copy buffers
[Input] Input device start failed, exiting!
Reattached kernel driver`

But if i tried to compile and run latest release (branch 22.11) everything works fine.
How I can make work latest version ?

`rtl_433 version 22.11 branch  at 202211191645 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/dietpi/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 191 out of 223 device decoding protocols [ 1-4 8 11-12 15-17 19-23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 177-197 199 201-215 217-223 ]
Detached kernel driver
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
r82xx_write: i2c wr failed=-1 reg=10 len=1
r82xx_set_freq: failed=-1
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 433.920MHz.
Allocating 15 zero-copy buffers
baseband_demod_FM: low pass filter for 250000 Hz at cutoff 25000 Hz, 40.0 us
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2023-09-16 13:10:57
model     : LaCrosse-TX141W                        Sensor ID : 59456
Channel   : 0            Battery level: 1          Temperature: 29.00 C      Humidity  : 51 %          Test?     : 0             Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _`

zuckschwerdt, 2023-09-16:
Your input didn't start within 0.5 seconds. This is unusual. Try `-D restart`

endmarsfr, 2023-09-16:
Hi ,
Thanks for your reply :)
Unfortunately it didn't help.
When I try -D restart it fails and claims device busy as you can see in log below 
It is so strange that it worked fine with the 22.11 release and not with the latest version.
I am using it on an Odroid C2 with the latest fresh DietPi v8.21.1 bookworm.

`[Protocols] Registered 211 out of 244 device decoding protocols [ 1-4 8 10-12 15-17 19-23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119-121 124-128 130-149 151-161 163-168 170-175 177-197 199 201-215 217-228 230-232 234-241 243-244 ]
Detached kernel driver
Found Rafael Micro R820T tuner
[SDR] Using device 1: Realtek, RTL2838UHIDIR, SN: 00000001, "Generic RTL2832U OEM"
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Allocating 15 zero-copy buffers
[Input] Input device start failed, restarting!
usb_claim_interface error -6
[sdr_open_rtl] Failed to open rtlsdr device #1.
[sdr_open_rtl] Unable to open a device`

endmarsfr, 2023-09-16:
I have compiled rtl_433 version 22.11-91-g17fe8254 branch from 202302101353 and it works fine:
I am looking for the commit that caused the fail

`dietpi@radio:~$ ./rtl_433 -d 1
rtl_433 version 22.11-91-g17fe8254 branch  at 202302101353 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/dietpi/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
[Protocols] Registered 204 out of 237 device decoding protocols [ 1-4 8 11-12 15-17 19-23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119-121 124-128 130-149 151-161 163-168 170-175 177-197 199 201-215 217-228 230-232 234-237 ]
[SDR] Found 2 device(s)
[SDR] trying device  1:  Realtek, RTL2838UHIDIR, SN: 00000001
Detached kernel driver
Found Rafael Micro R820T tuner
[SDR] Using device 1: Generic RTL2832U OEM
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Allocating 15 zero-copy buffers
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2023-09-16 18:05:59
model     : Nexus-TH     House Code: 3
Channel   : 1            Battery   : 1             Temperature: 26.60 C      Humidity  : 5 %
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _`

endmarsfr, 2023-09-16:
I found the commit causing the fail.
it is Change to async SDR acquire thread (#1978)

**Failed
commit 46de49f35825825288b2165d6e102c0d51396682
Author: Christian W. Zuckschwerdt <zany@triq.net>
Date:   Tue Mar 7 12:36:35 2023 +0100
    Change to async SDR acquire thread (#1978)**

**Work fine**
commit fa13eb69ce52cd47fff61dad0ddaf642362199d5 
Author: Christian W. Zuckschwerdt <christian@zuckschwerdt.org>
Date:   Mon Mar 6 17:50:30 2023 +0100
    minor: Fix Celsia-CZC1 typo

zuckschwerdt, 2023-09-16:
You input somehow seems slow to start. Insert a `return;` before line 1525 in `rtl_433.c` https://github.com/merbanan/rtl_433/blob/master/src/rtl_433.c#L1525
Report how many times you see the warning then. Each warning print is 250 ms delay.

endmarsfr, 2023-09-17:
Hi,

Unfortunately, the fix also failed.
Is it possible to increase the delay from 250ms to something else for testing?
`dietpi@radio:~$ ./rtl_433 -d 0
rtl_433 version nightly-3-g70d84d01 branch master at 202308211241 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/dietpi/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
[Protocols] Registered 212 out of 246 device decoding protocols [ 1-4 8 10-12 15-17 19-23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119-121 124-128 130-149 151-161 163-168 170-175 177-197 199 201-215 217-228 230-232 234-241 243-244 246 ]
Found Rafael Micro R820T tuner
[SDR] Using device 0: Realtek, RTL2838UHIDIR, SN: 00000002, "Generic RTL2832U OEM"
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Allocating 15 zero-copy buffers
[Input] Input device start failed, exiting!
[Input] Input device start failed, exiting!
[Input] Input device start failed, exiting!
[Input] Input device start failed, exiting!
[Input] Input device start failed, exiting!
[Input] Input device start failed, exiting!
[Input] Input device start failed, exiting!
[Input] Input device start failed, exiting!
[Input] Input device start failed, exiting!
[Input] Input device start failed, exiting!
[Input] Input device start failed, exiting!
[Input] Input device start failed, exiting!
[Input] Input device start failed, exiting!
[Input] Input device start failed, exiting!`

zuckschwerdt, 2023-09-17:
I misread the source, the first timer is 2.5 secs, each next one 1.5 secs.
The log shows that you didn't receive any data from the dongle for nearly `2.5+1.5*14 = 23` seconds?
Something else must be broken. The whole point of that check is to catch these problems.

But yes, you can just disable the timer by removing line https://github.com/merbanan/rtl_433/blob/master/src/rtl_433.c#L1992

zuckschwerdt, 2023-09-17:
It's not clear what is happening with your RTL-SDR. You can change sdr.c and add some `fprintf(stderr, "DEBUG line X\n");` statements.
Add them the async start https://github.com/merbanan/rtl_433/blob/master/src/sdr.c#L551
And activate them in the callback https://github.com/merbanan/rtl_433/blob/master/src/sdr.c#L487

Is the `rtlsdr_read_async()` called and blocking? Is the callback called and processing the frames? Does it get stuck? I.e. does `dev->rtlsdr_cb()` return and you see "rtlsdr_read_cb cb done."?

endmarsfr, 2023-09-17:
Hi there,
Thanks for your help.
But first, when I commented line 1992 in rtl_433.c
Nothing happened, no frame received.

Then I uncommented line 1992 in rtl_433.c and added debug prints in sdr.c.
1. In callback : fprintf(stderr, "DEBUG line 487 in callback \n");
2. before and after async start
Here is the resulting log:

[Protocols] Registered 212 out of 246 device decoding protocols
[Input] The internals of input handling changed, read about and report problems on PR #1978
[SDR] Found 2 device(s)
[SDR] trying device 1: Realtek, RTL2838UHIDIR, SN: 00000001
Detached kernel driver
Found Rafael Micro R820T tuner
[SDR] Using device 1: Realtek, RTL2838UHIDIR, SN: 00000001, "Generic RTL2832U OEM"
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
[SDR] Sample rate set to 250000 S/s.
[Input] Bit detection level set to 0.0 (Auto).
[SDR] Tuner gain set to Auto.
[Input] Reading samples in async mode...
[SDR] rtlsdr_set_center_freq 433920000 = 0
[SDR] Tuned to 433.920MHz.
[acquire_thread] acquire_thread enter...
**DEBUG line 551 Before rtlsdr_read_async**
Allocating 15 zero-copy buffers
**DEBUG line 487 in callback
DEBUG line 487 in callback
DEBUG line 487 in callback
DEBUG line 487 in callback
DEBUG line 487 in callback
DEBUG line 487 in callback
DEBUG line 487 in callback**
[Input] Input device start failed, exiting!
[sdr_stop] EXITING...
[sdr_stop] JOINING...
**DEBUG line 551 After rtlsdr_read_async**
[rtlsdr_read_loop] rtlsdr_read_async done
[acquire_thread] acquire_thread async stop...
[acquire_thread] acquire_thread done...
[sdr_stop] EXITED.
[rtl_433] stopping...
[sdr_stop] EXITING...
[sdr_stop] Already exiting.
[sdr_stop] EXITING...
[sdr_stop] Already exiting.
Reattached kernel driver

It is so strange that before this commit "Change to async SDR acquire thread (#1978)" my device was working fine and still working fine
I can run more tests if you want.

Regards

zuckschwerdt, 2023-09-17:
That looks normal and expected, we are receiving data. I wonder where the data is lost?

Can you uncomment lines 523 and 526 (`fprintf(stderr, "rtlsdr_read_cb cb...\n");` and `fprintf(stderr, "rtlsdr_read_cb cb done.\n");`)

If that works the next place data goes is `rtl_433.c`, uncomment lines 1460, 1467, 1469.

And then uncomment and check if line 1402 is called.

endmarsfr, 2023-09-17:
here is the resulting log :

[Protocols] Registered 212 out of 246 device decoding protocols
[Input] The internals of input handling changed, read about and report problems on PR #1978
[SDR] Found 2 device(s)
[SDR] trying device 1: Realtek, RTL2838UHIDIR, SN: 00000001
Detached kernel driver
Found Rafael Micro R820T tuner
[SDR] Using device 1: Realtek, RTL2838UHIDIR, SN: 00000001, "Generic RTL2832U OEM"
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
[SDR] Sample rate set to 250000 S/s.
[Input] Bit detection level set to 0.0 (Auto).
[SDR] Tuner gain set to Auto.
[Input] Reading samples in async mode...
[SDR] rtlsdr_set_center_freq 433920000 = 0
[SDR] Tuned to 433.920MHz.
[acquire_thread] acquire_thread enter...
DEBUG line 551 Before rtlsdr_read_async
Allocating 15 zero-copy buffers
timer_handler: -1, 0, 0xaaaae9c80a90, 0xffffecbeaca0
DEBUG line 487 in callback
rtlsdr_read_cb cb...
1694958771.281070 acquire_callback...
acquire_callback bc send...
acquire_callback bc done...
rtlsdr_read_cb cb done.
timer_handler: -1, 0, 0xaaaae9c80a90, 0xffffecbeaca0
DEBUG line 487 in callback
rtlsdr_read_cb cb...
1694958771.815264 acquire_callback...
acquire_callback bc send...
acquire_callback bc done...
rtlsdr_read_cb cb done.
timer_handler: -1, 0, 0xaaaae9c80a90, 0xffffecbeaca0
DEBUG line 487 in callback
rtlsdr_read_cb cb...
1694958772.325637 acquire_callback...
acquire_callback bc send...
acquire_callback bc done...
rtlsdr_read_cb cb done.
timer_handler: -1, 0, 0xaaaae9c80a90, 0xffffecbeaca0
DEBUG line 487 in callback
rtlsdr_read_cb cb...
1694958772.849468 acquire_callback...
acquire_callback bc send...
acquire_callback bc done...
rtlsdr_read_cb cb done.
timer_handler: -1, 6, 0xaaaae9c80a90, 0xffffecbeaca0
timer event, current time: 1694958773.15, next timer: 1694958774.65
timer_handler: -1, 0, 0xaaaae9c80a90, 0xffffecbeaca0
DEBUG line 487 in callback
rtlsdr_read_cb cb...
1694958773.374219 acquire_callback...
acquire_callback bc send...
acquire_callback bc done...
rtlsdr_read_cb cb done.
timer_handler: -1, 0, 0xaaaae9c80a90, 0xffffecbeaca0
DEBUG line 487 in callback
rtlsdr_read_cb cb...
1694958773.898067 acquire_callback...
acquire_callback bc send...
acquire_callback bc done...
rtlsdr_read_cb cb done.
timer_handler: -1, 0, 0xaaaae9c80a90, 0xffffecbeaca0
DEBUG line 487 in callback
rtlsdr_read_cb cb...
1694958774.422302 acquire_callback...
acquire_callback bc send...
acquire_callback bc done...
rtlsdr_read_cb cb done.
timer_handler: -1, 6, 0xaaaae9c80a90, 0xffffecbeaca0
timer event, current time: 1694958774.65, next timer: 1694958776.16
[Input] Input device start failed, exiting!
[sdr_stop] EXITING...
[sdr_stop] JOINING...
DEBUG line 551 After rtlsdr_read_async
[rtlsdr_read_loop] rtlsdr_read_async done
[acquire_thread] acquire_thread async stop...
[acquire_thread] acquire_thread done...
[sdr_stop] EXITED.
timer_handler: -1, 0, 0xaaaae9c80a90, 0xffffecbeaca0
[rtl_433] stopping...
[sdr_stop] EXITING...
[sdr_stop] Already exiting.
[sdr_stop] EXITING...
[sdr_stop] Already exiting.
Reattached kernel driver
timer_handler: -1, 0, 0xaaaae9c80a90, 0xffffecbeac80
timer_handler: -1, 5, 0xaaaae9c80a90, (nil)

zuckschwerdt, 2023-09-17:
Seems like the events get stuck/lost in the thread IPC, i.e. when going from
`acquire_callback()`'s `mg_broadcast(mgr, sdr_handler, (void *)ev, sizeof(*ev));` to the actual `sdr_handler()`.

You did uncomment line 1402, right? But we never see the output of
`fprintf(stderr, "%s: %d, %d, %p, %p\n", __func__, nc->sock, ev_type, nc->user_data, ev_data);`
e.g. `sdr_handler: -1, 0, ...`

The event loop is running as we see the timer fire, but somehow that IPC pipe isn't working?
Is there anything special about your setup? About the hardware or distro or settings, security, anything?

endmarsfr, 2023-09-17:
Yes, I uncommented all "fprintf" in sdr.c and rtl_433.c, and this one too :

`static void timer_handler(struct mg_connection *nc, int ev, void *ev_data)
{
    fprintf(stderr, "%s: %d, %d, %p, %p\n", __func__, nc->sock, ev, nc->user_data, ev_data);
    r_cfg_t *cfg = (r_cfg_t *)nc->user_data;

Nothing special in my config (I have 2 RTL_SDR devices) , here is my dmesg log : 

> [ 9690.417798] input: Realtek RTL2832U reference design as /devices/platform/soc/c9100000.usb/usb1/1-1/1-1.3/rc/rc1/input10
> [ 9690.418225] usb 1-1.3: dvb_usb_v2: schedule remote query interval to 200 msecs
> [ 9690.427021] usb 1-1.3: dvb_usb_v2: 'Realtek RTL2832U reference design' successfully initialized and connected
> [ 9978.894824] r820t 3-001a: destroying instance
> [ 9978.896227] dvb_usb_v2: 'Realtek RTL2832U reference design:1-1.3' successfully deinitialized and disconnected
> [ 9983.778710] usb 1-1.3: dvb_usb_v2: found a 'Realtek RTL2832U reference design' in warm state
> [ 9983.843488] usb 1-1.3: dvb_usb_v2: will pass the complete MPEG2 transport stream to the software demuxer
> [ 9983.843554] dvbdev: DVB: registering new adapter (Realtek RTL2832U reference design)
> [ 9983.843578] usb 1-1.3: media controller created
> [ 9983.846237] dvbdev: dvb_create_media_entity: media entity 'dvb-demux' registered.
> [ 9983.862279] i2c i2c-2: Added multiplexed i2c bus 3
> [ 9983.862302] rtl2832 2-0010: Realtek RTL2832 successfully attached
> [ 9983.862351] usb 1-1.3: DVB: registering adapter 0 frontend 0 (Realtek RTL2832 (DVB-T))...
> [ 9983.862380] dvbdev: dvb_create_media_entity: media entity 'Realtek RTL2832 (DVB-T)' registered.
> [ 9983.862523] r820t 3-001a: creating new instance
> [ 9983.869770] r820t 3-001a: Rafael Micro r820t successfully identified, chip type: R820T
> [ 9983.885363] rtl2832_sdr rtl2832_sdr.3.auto: Registered as swradio0
> [ 9983.885385] rtl2832_sdr rtl2832_sdr.3.auto: Realtek RTL2832 SDR attached
> [ 9983.885391] rtl2832_sdr rtl2832_sdr.3.auto: SDR API is still slightly experimental and functionality changes may follow
> [ 9983.894998] Registered IR keymap rc-empty
> [ 9983.895257] rc rc1: Realtek RTL2832U reference design as /devices/platform/soc/c9100000.usb/usb1/1-1/1-1.3/rc/rc1
> [ 9983.895461] rc rc1: lirc_dev: driver dvb_usb_rtl28xxu registered at minor = 1, raw IR receiver, no transmitter
> [ 9983.895685] input: Realtek RTL2832U reference design as /devices/platform/soc/c9100000.usb/usb1/1-1/1-1.3/rc/rc1/input11
> [ 9983.896808] usb 1-1.3: dvb_usb_v2: schedule remote query interval to 200 msecs
> [ 9983.905771] usb 1-1.3: dvb_usb_v2: 'Realtek RTL2832U reference design' successfully initialized and connected
> [12933.694657] r820t 3-001a: destroying instance
> [12933.695679] dvb_usb_v2: 'Realtek RTL2832U reference design:1-1.3' successfully deinitialized and disconnected
> [12938.539890] usb 1-1.3: dvb_usb_v2: found a 'Realtek RTL2832U reference design' in warm state
> [12938.604741] usb 1-1.3: dvb_usb_v2: will pass the complete MPEG2 transport stream to the software demuxer
> [12938.604804] dvbdev: DVB: registering new adapter (Realtek RTL2832U reference design)
> [12938.604826] usb 1-1.3: media controller created
> [12938.607652] dvbdev: dvb_create_media_entity: media entity 'dvb-demux' registered.
> [12938.625855] i2c i2c-2: Added multiplexed i2c bus 3
> [12938.625878] rtl2832 2-0010: Realtek RTL2832 successfully attached
> [12938.625931] usb 1-1.3: DVB: registering adapter 0 frontend 0 (Realtek RTL2832 (DVB-T))...
> [12938.625967] dvbdev: dvb_create_media_entity: media entity 'Realtek RTL2832 (DVB-T)' registered.
> [12938.626112] r820t 3-001a: creating new instance
> [12938.633275] r820t 3-001a: Rafael Micro r820t successfully identified, chip type: R820T
> [12938.649369] rtl2832_sdr rtl2832_sdr.3.auto: Registered as swradio0
> [12938.649394] rtl2832_sdr rtl2832_sdr.3.auto: Realtek RTL2832 SDR attached
> [12938.649400] rtl2832_sdr rtl2832_sdr.3.auto: SDR API is still slightly experimental and functionality changes may follow
> [12938.662876] Registered IR keymap rc-empty
> [12938.666865] rc rc1: Realtek RTL2832U reference design as /devices/platform/soc/c9100000.usb/usb1/1-1/1-1.3/rc/rc1
> [12938.668657] rc rc1: lirc_dev: driver dvb_usb_rtl28xxu registered at minor = 1, raw IR receiver, no transmitter
> [12938.670164] input: Realtek RTL2832U reference design as /devices/platform/soc/c9100000.usb/usb1/1-1/1-1.3/rc/rc1/input12
> [12938.673773] usb 1-1.3: dvb_usb_v2: schedule remote query interval to 200 msecs
> [12938.682498] usb 1-1.3: dvb_usb_v2: 'Realtek RTL2832U reference design' successfully initialized and connected`
> 
>

zuckschwerdt, 2023-09-17:
I'll see if we can replace the pipe IPC with pthreads cond mechanism -- there was another blocking problem getting that pipe stuck and it seems worthwhile tcheck. I'll let you know when code is ready for testing.

zuckschwerdt, 2023-09-18:
Just to be sure, can you undo all changes (fresh clone or undo with e.g. `git checkout src/rtl_433.c`), and then only uncomment three lines:
https://github.com/merbanan/rtl_433/blob/master/src/rtl_433.c#L1357
https://github.com/merbanan/rtl_433/blob/master/src/rtl_433.c#L1421
https://github.com/merbanan/rtl_433/blob/master/src/rtl_433.c#L1423

You do see the acquire_callback send and done but never the sdr_handler?

zuckschwerdt, 2023-09-18:
Also you can set https://github.com/merbanan/rtl_433/blob/master/src/mongoose.c#L465 to `1` and you should see
`mongoose.c:4120         read 48 from ctl socket` -- but that is the operation that is failing, either it's not reading or not acting upon the read.

endmarsfr, 2023-09-18:
Hi,

here is the resulting log for the test :

> [Protocols] Registered 212 out of 246 device decoding protocols [ 1-4 8 10-12 15-17 19-23 25-26 29-36 38-60 63 67-71 73-
> 100 102-105 108-116 119-121 124-128 130-149 151-161 163-168 170-175 177-197 199 201-215 217-228 230-232 234-241 243-244 246 ]
> Detached kernel driver
> Found Rafael Micro R820T tuner
> [SDR] Using device 1: Realtek, RTL2838UHIDIR, SN: 00000001, "Generic RTL2832U OEM"
> Exact sample rate is: 250000.000414 Hz
> [R82XX] PLL not locked!
> mongoose.c:4142         0xaaaac9f14dd0 using select()
> mongoose.c:2618         ==================================
> mongoose.c:2619         init mgr=0xaaaac9f14dd0
> mongoose.c:2419         0xaaaac9f14dd0 0xaaaac9f14500
> mongoose.c:3532         0xaaaac9f14500 (nil) 0 -> 1695053366
> Allocating 15 zero-copy buffers
> acquire_callback bc send...
> acquire_callback bc done...
> acquire_callback bc send...
> acquire_callback bc done...
> acquire_callback bc send...
> acquire_callback bc done...
> acquire_callback bc send...
> acquire_callback bc done...
> mongoose.c:2449         0xaaaac9f14500 user ev=6 ev_data=0xffffc2a96dd0 flags=0x0 rmbl=0 smbl=0
> mongoose.c:3532         0xaaaac9f14500 (nil) 0 -> 1695053368
> mongoose.c:2471         0xaaaac9f14500 after user flags=0x0 rmbl=0 smbl=0
> acquire_callback bc send...
> acquire_callback bc done...
> acquire_callback bc send...
> acquire_callback bc done...
> acquire_callback bc send...
> acquire_callback bc done...
> mongoose.c:2449         0xaaaac9f14500 user ev=6 ev_data=0xffffc2a96dd0 flags=0x0 rmbl=0 smbl=0
> mongoose.c:3532         0xaaaac9f14500 (nil) 0 -> 1695053369
> [Input] Input device start failed, exiting!
> mongoose.c:2471         0xaaaac9f14500 after user flags=0x0 rmbl=0 smbl=0
> Reattached kernel driver
> mongoose.c:2634         0xaaaac9f14dd0
> mongoose.c:2449         0xaaaac9f14500 user ev=5 ev_data=(nil) flags=0x800 rmbl=0 smbl=0
> mongoose.c:2471         0xaaaac9f14500 after user flags=0x800 rmbl=0 smbl=0
> dietpi@radio:~$

zuckschwerdt, 2023-09-18:
Thanks, that's good evidence that the mg_broadcast() is sent and then the message disappears. It's like the used socket pair is broken. Very strange.

Is it Armbian Jammy on both devices? I only have a RPi3b free right now, I guess I'll try to put ARM64 Armbian on it to reproduce this behaviour.

endmarsfr, 2023-09-18:
Hi,
Yes, you can install it on a rapsberry pi too (my device is an Odroid C2).
You can download the iso here: https://dietpi.com/#downloadinfo
Regards

zuckschwerdt, 2023-09-18:
- Armbian_23.8.1_Rpi4b_jammy_current_6.1.50.img works without problems.
- DietPi_RPi-ARMv8-Bookworm.img works without problems.

E.g. after a fresh install use
```
$ echo -e "\nblacklist dvb_usb_rtl28xxu\nblacklist rtl2832\nblacklist rtl2830" |sudo tee -a /etc/modprobe.d/blacklist.conf
$ sudo apt install build-essential git librtlsdr-dev ninja-build cmake
$ git clone https://github.com/merbanan/rtl_433.git
$ cd rtl_433 ; mkdir build ; cd build
$ cmake -DFORCE_COLORED_BUILD:BOOL=ON -GNinja .. && cmake --build . -j 4
$ ./src/rtl_433 -Y autolevel -Y magest -M level -M noise -M time:usec
```

Something has to be special about your installs? I really don't know what would be stopping the `mg_broadcast()` messages.

endmarsfr, 2023-09-19:
hi
I installed an access point on the device using https://raspap.com/.
But if mg_broadcast() deals with pipe, there is no reason why RaspAp was the cause.
Regards

zuckschwerdt, 2023-09-19:
Since I can't reproduce we'll need you to debug this step by step to find out what's broken.
Before the `mg_broadcast()` call in https://github.com/merbanan/rtl_433/blob/master/src/rtl_433.c#L1422
insert
`    fprintf(stderr, "mg broadcast pipe is %d %d\n", mgr->ctl[0], mgr->ctl[0]);`
It should show e.g. `mg broadcast pipe is 6 6`
Which means the socket ID 6 is connected on both ends.

endmarsfr, 2023-09-20:
Thank you very much.
I will try this and get back to you.

endmarsfr, 2023-09-20:
Hi, 
here the resulting log for the test  : mg broadcast pipe is 9 9
> [Protocols] Registered 212 out of 246 device decoding protocols [ 1-4 8 10-12 15-17 19-23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119-121 124-128 130-149 151-161 163-168 170-175 177-197 199 201-215 217-228 230-232 234-241 243-244 246 ]
> Detached kernel driver
> Found Rafael Micro R820T tuner
> r82xx_write: i2c wr failed=-1 reg=0c len=7
> [R82XX] PLL not locked!
> [SDR] Using device 1: Realtek, RTL2838UHIDIR, SN: 00000001, "Generic RTL2832U OEM"
> Exact sample rate is: 250000.000414 Hz
> [R82XX] PLL not locked!
> Allocating 15 zero-copy buffers
> mg broadcast pipe is 9 9
> mg broadcast pipe is 9 9
> mg broadcast pipe is 9 9
> mg broadcast pipe is 9 9
> mg broadcast pipe is 9 9
> mg broadcast pipe is 9 9
> mg broadcast pipe is 9 9
> [Input] Input device start failed, exiting!
> Reattached kernel driver

endmarsfr, 2023-09-20:
Here is the resulting log :

> [Protocols] Registered 212 out of 246 device decoding protocols [ 1-4 8 10-12 15-17 19-23 25-26 29-36 38-60 63 67-71 73-
> 100 102-105 108-116 119-121 124-128 130-149 151-161 163-168 170-175 177-197 199 201-215 217-228 230-232 234-241 243-244 246 ]
> Detached kernel driver
> Found Rafael Micro R820T tuner
> [SDR] Using device 1: Realtek, RTL2838UHIDIR, SN: 00000001, "Generic RTL2832U OEM"
> Exact sample rate is: 250000.000414 Hz
> [R82XX] PLL not locked!
> Allocating 15 zero-copy buffers
> mg_socket_if_poll 0 8 0
> mg broadcast pipe is 9 9
> mg_broadcast send to 9: 8 + 40
> mg_broadcast sent 48 recv...
> mg_broadcast recv from 9: 18446744073709551615 40
> mg_socket_if_poll 0 8 0
> mg broadcast pipe is 9 9
> mg_broadcast send to 9: 8 + 40
> mg_broadcast sent 48 recv...
> mg_broadcast recv from 9: 18446744073709551615 40
> mg_socket_if_poll 0 8 0
> mg broadcast pipe is 9 9
> mg_broadcast send to 9: 8 + 40
> mg_broadcast sent 48 recv...
> mg_broadcast recv from 9: 18446744073709551615 40
> mg_socket_if_poll 0 8 0
> mg broadcast pipe is 9 9
> mg_broadcast send to 9: 8 + 40
> mg_broadcast sent 48 recv...
> mg_broadcast recv from 9: 18446744073709551615 40
> mg_socket_if_poll 0 8 0
> mg broadcast pipe is 9 9
> mg_broadcast send to 9: 8 + 40
> mg_broadcast sent 48 recv...
> mg_broadcast recv from 9: 18446744073709551615 40
> mg_socket_if_poll 0 8 0
> mg broadcast pipe is 9 9
> mg_broadcast send to 9: 8 + 40
> mg_broadcast sent 48 recv...
> mg_broadcast recv from 9: 18446744073709551615 40
> mg_socket_if_poll 0 8 0
> mg broadcast pipe is 9 9
> mg_broadcast send to 9: 8 + 40
> mg_broadcast sent 48 recv...
> mg_broadcast recv from 9: 18446744073709551615 40
> mg_socket_if_poll 0 8 0
> [Input] Input device start failed, exiting!
> Reattached kernel driver
> mg_socket_if_poll 0 8 0

zuckschwerdt, 2023-09-20:
Ohh I see, the `recv()` has an unhandeld error! 18446744073709551615 = 0xFFFFFFFFFFFFFFFF = -1
https://man7.org/linux/man-pages/man2/recv.2.html#RETURN_VALUE
Change the section https://github.com/merbanan/rtl_433/blob/master/src/mongoose.c#L3466
to output `perror()`
```C
    fprintf(stderr, "%s send to %d: %lu + %lu\n", __func__, mgr->ctl[0], offsetof(struct ctl_msg, message), len);
    dummy = MG_SEND_FUNC(mgr->ctl[0], (char *) &ctl_msg,
                         offsetof(struct ctl_msg, message) + len, 0);
    fprintf(stderr, "%s sent %ld recv...\n", __func__, dummy);
    dummy = MG_RECV_FUNC(mgr->ctl[0], (char *) &len, 1, 0);
    fprintf(stderr, "%s recv from %d: %ld %lu\n", __func__, mgr->ctl[0], dummy, len);
    if (dummy != 1) perror("mg_broadcast recv failed");
```
Now we can really know what failed.

endmarsfr, 2023-09-20:
Here is the result : "Connection refused"
What could be the cause of connection denied ? 
Could RaspAP (https://raspap.com/) be the cause?

> Use "-F log" if you want any messages, warnings, and errors in the console.
> Detached kernel driver
> Found Rafael Micro R820T tuner
> Exact sample rate is: 250000.000414 Hz
> [R82XX] PLL not locked!
> Allocating 15 zero-copy buffers
> mg_socket_if_poll 0 8 0
> mg broadcast pipe is 9 9
> mg_broadcast send to 9: 8 + 40
> mg_broadcast sent 48 recv...
> mg_broadcast recv from 9: -1 40
> mg_broadcast recv failed: Connection refused
> mg_socket_if_poll 0 8 0
> mg broadcast pipe is 9 9
> mg_broadcast send to 9: 8 + 40
> mg_broadcast sent 48 recv...
> mg_broadcast recv from 9: -1 40
> mg_broadcast recv failed: Connection refused
> mg_socket_if_poll 0 8 0
> mg broadcast pipe is 9 9
> mg_broadcast send to 9: 8 + 40
> mg_broadcast sent 48 recv...
> mg_broadcast recv from 9: -1 40
> mg_broadcast recv failed: Connection refused
> mg_socket_if_poll 0 8 0
> mg broadcast pipe is 9 9
> mg_broadcast send to 9: 8 + 40
> mg_broadcast sent 48 recv...
> mg_broadcast recv from 9: -1 40
> mg_broadcast recv failed: Connection refused
> mg_socket_if_poll 0 8 0
> mg broadcast pipe is 9 9
> mg_broadcast send to 9: 8 + 40
> mg_broadcast sent 48 recv...
> mg_broadcast recv from 9: -1 40
> mg_broadcast recv failed: Connection refused
> mg_socket_if_poll 0 8 0
> mg broadcast pipe is 9 9
> mg_broadcast send to 9: 8 + 40
> mg_broadcast sent 48 recv...
> mg_broadcast recv from 9: -1 40
> mg_broadcast recv failed: Connection refused
> mg_socket_if_poll 0 8 0
> mg broadcast pipe is 9 9
> mg_broadcast send to 9: 8 + 40
> mg_broadcast sent 48 recv...
> mg_broadcast recv from 9: -1 40
> mg_broadcast recv failed: Connection refused
> mg_socket_if_poll 0 8 0
> Reattached kernel driver
> mg_socket_if_poll 0 8 0

zuckschwerdt, 2023-09-20:
Yes, I remembered a unix fifo pipe but it's actually a UDP socket pair on 127.0.0.1.
Last thing to check here is if the setup worked https://github.com/merbanan/rtl_433/blob/master/src/mongoose.c#L4144
```C
#if MG_ENABLE_BROADCAST
  int r = mg_socketpair(iface->mgr->ctl, SOCK_DGRAM);
  fprintf(stderr, "%s: %d\n", __func__, r);
#endif
```
Result there should be `1`. Otherwise something in the network stack must be blocking reads, from your config I don't see what though.
But the rule `-A POSTROUTING -j MASQUERADE`, i.e. send every packet through MASQ doesn't look right.

endmarsfr, 2023-09-20:
The rules are generated by RaspAP, I shouldn't change them.

The network stack seems to be OK:
> [Protocols] Registered 212 out of 246 device decoding protocols [ 1-4 8 10-12 15-17 19-23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119-121 124-128 130-149 151-161 163-168 170-175 177-197 199 201-215 217-228 230-232 234-241 243-244 246 ]
> Found Rafael Micro R820T tuner
> [SDR] Using device 1: Realtek, RTL2838UHIDIR, SN: 00000001, "Generic RTL2832U OEM"
> Exact sample rate is: 250000.000414 Hz
> [R82XX] PLL not locked!
> **mg_socket_if_init: 1**
> Allocating 15 zero-copy buffers
> mg_socket_if_poll 0 8 0
> mg broadcast pipe is 9 9
> mg_broadcast send to 9: 8 + 40
> mg_broadcast sent 48 recv...
> mg_broadcast recv from 9: -1 40
> mg_broadcast recv failed: Connection refused
> mg_socket_if_poll 0 8 0
> mg broadcast pipe is 9 9
> mg_broadcast send to 9: 8 + 40
> mg_broadcast sent 48 recv...
> mg_broadcast recv from 9: -1 40
> mg_broadcast recv failed: Connection refused
> mg_socket_if_poll 0 8 0
> mg broadcast pipe is 9 9
> mg_broadcast send to 9: 8 + 40
> mg_broadcast sent 48 recv...
> mg_broadcast recv from 9: -1 40
> mg_broadcast recv failed: Connection refused
> mg_socket_if_poll 0 8 0
> mg broadcast pipe is 9 9
> mg_broadcast send to 9: 8 + 40
> mg_broadcast sent 48 recv...
> mg_broadcast recv from 9: -1 40
> mg_broadcast recv failed: Connection refused
> mg_socket_if_poll 0 8 0
> mg broadcast pipe is 9 9
> mg_broadcast send to 9: 8 + 40
> mg_broadcast sent 48 recv...
> mg_broadcast recv from 9: -1 40
> mg_broadcast recv failed: Connection refused
> mg_socket_if_poll 0 8 0
> mg broadcast pipe is 9 9
> mg_broadcast send to 9: 8 + 40
> mg_broadcast sent 48 recv...
> mg_broadcast recv from 9: -1 40
> mg_broadcast recv failed: Connection refused
> mg_socket_if_poll 0 8 0
> mg broadcast pipe is 9 9
> mg_broadcast send to 9: 8 + 40
> mg_broadcast sent 48 recv...
> mg_broadcast recv from 9: -1 40
> mg_broadcast recv failed: Connection refused
> mg_socket_if_poll 0 8 0
> [Input] Input device start failed, exiting!
> mg_socket_if_poll 0 8 0

zuckschwerdt, 2023-09-20:
> The rules are generated by RaspAP, I shouldn't change them.

Just as a test stop iptables, retry. After restarting iptables or a reboot you didn't change anything.

endmarsfr, 2023-09-20:
I found the cause 
Uninstalling raspAP allows the latest rtl_433 to work fine.
But I need RaspAP to capture traffic from my other device.
I think rtl_433 may be connected to the wrong network.
I am recovering my device with raspAP

mg_broadcast recv from 9: 1 1
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2023-09-20 16:18:21.863622
model     : Prologue-TH  subtype   : 9             id        : 28
Channel   : 1            Battery   : 0             Temperature: 21.90 C      Humidity  : 0 %           Button    : 0
Modulation: ASK          Freq      : 433.9 MHz
RSSI      : -23.9 dB     SNR       : 9.8 dB        Noise     : -33.7 dB
mg broadcast pipe is 9 9
mg_broadcast send to 9: 8 + 40
mg_broadcast sent 48 recv...

closed 2023-09-20 by zuckschwerdt

endmarsfr, 2023-09-20:
> No need to uninstall. Just this one rule is broken: ` -A POSTROUTING -j MASQUERADE`

I tried to comment the rule, you're right, but I broke raspAP and that means I could not use the new implementation of rtl_433 and raspAP together :(.

zuckschwerdt, 2023-09-20:
The fault is with RaspAP. Masquerading all traffic is not a reasonable rule. You might get lucky adding `! -i lo` but why would you masq the primary address on eth0 or wlan0 at all?
The other rule might be wrong too as you don't seem to be using `192.168.50.0/24`?

endmarsfr, 2023-09-20:
I found the way to solve my problem 
I have to use these rules

# Allow broadcast traffic without MASQUERADE
sudo iptables -t nat -A POSTROUTING -o eth0 -d 255.255.255.255 -j ACCEPT

# Apply MASQUERADE to non-broadcast traffic
sudo iptables -t nat -A POSTROUTING -o eth0 -j MASQUERADE

But since "/etc/iptables/rules.v4" is generated by raspAP, I am afraid that this file could be changed by any raspAP update.

For now, I don't know how to change the contents of the file to make the rules work:

```
1 # Generated by iptables-save v1.8.9 (nf_tables) on Thu Sep 14 13:55:28 2023
 2 *nat
 3 :PREROUTING ACCEPT [7:540]
 4 :INPUT ACCEPT [4:408]
 5 :OUTPUT ACCEPT [0:0]
 6 :POSTROUTING ACCEPT [0:0]
 7 -A POSTROUTING -j MASQUERADE
 8 -A POSTROUTING -s 192.168.50.0/24 ! -d 192.168.50.0/24 -j MASQUERADE
 9 COMMIT
10 # Completed on Thu Sep 14 13:55:28 2023
```

You can now add this rule to the documentation as a requirement to allow mongoose mg_broadcast.
`sudo iptables -t nat -A POSTROUTING -o eth0 -d 255.255.255.255 -j ACCEPT`

zuckschwerdt, 2023-09-20:
The `eth0` interface should not be concerned. `127.0.0.1` is on `lo`. The whole masq isn't great, I prefer explicit SNAT and DNAT.

I will add more error checking and warnings, but we can't reasonably document workarounds for broken systems.

Thanks for working through this till the fix, but I guess raspAP might fix this problem some day and hopefully nobody else suffers through this :)

commit 596c3d6f (2023-10-02): Add warning if firewall blocks loopback (closes #2621)
commit 08e23910 (2023-10-02): Add warning if firewall blocks loopback (closes #2621)