merbanan/rtl_433 #2622: rtl_433_mqtt_hass.py creates devices with "manufacturer": "rtl_433"

fallenpegasus, 2023-09-17:
For example, for one of the entities it creates for me

===
device:
  identifiers:
    - Acurite-515-50
  name: Acurite-515-50
  model: Acurite-515
  manufacturer: rtl_43
===

should be

===
device:
  identifiers:
    - Acurite-515-50
  name: Acurite-515-50
  model: 515
  manufacturer: Acurite
===

gdt, 2023-09-17:
Is this with git master?   Can you provide the json input that causes this?

zuckschwerdt, 2023-09-17:
Best I can see that was added in #1390 "Manufacturer in the 'device' is required to create a group of sensors in Home Assistant."

gdt, 2023-09-17:
In that case, I think this is not a bug, except that the man page, err comments at beginning of file, should explain this.    I have on my personal todo list to understand the mqtt autoconfig stuff a lot better, and will have a look (not this week).

gdt, 2023-09-24:
I'm going to close this as not a bug.    If there is a "json output from acurite should have manufacturer acurite" bug, then filing an issue about that seems fair.   But we need to wind back the processing change to where the problem is in order to have something precise to talk about.

closed 2023-09-24 by gdt