merbanan/rtl_433 #2627: Nice Flor-s Rolling Code

Jev1337, 2023-09-19:
Hello,

This may be unrelated (sorry if it is), however I am trying to clone my remote which happens to be a Nice Flor-s Remote with an Arduino Nano / Arduino Due.

Reason why I want to do this is because buying a new one costs 30$ where I live which is extremely expensive whereas buying an RF433 RX/TX module costs as cheap as 3$.

Anyhow, I tried researching on how to decrypt and see how the rolling codes work and ways to clone this specific remote.

I got nothing but this repo and another flipper zero repo.

How can I possibly clone my Nice Remote? Is it even possible to run a command with rtl_433 without a device connected to it to decrypt Rx data?

Example of Rx from the remote: 
0x01, 0xf2, 0x2e, 0xd3, 0x63, 0x1a, 0x7a
0x01, 0xc2, 0x2e, 0xd3, 0x63, 0x1a, 0x7a

merbanan, 2023-09-19:
You can use -Y and input raw bytes to test the decoder.

zuckschwerdt, 2023-09-19:
Also the Flipper Zero decoding code is in #2238 just copy that out.

closed 2023-09-19 by zuckschwerdt

Jev1337, 2023-09-19:
> You can use -Y and input raw bytes to test the decoder.

Hello, I successfully managed to execute it. However I received nothing in response. After messing with the first byte I managed to make it work (I assume that the first byte is the byte that specifies which button is used.).
Is it okay to assume that the rest is correct?
Please note that my remote is an old model of FLO-R (2 buttons and buttons are blue).

Bytes: 
0x01, 0xf2, 0x2e, 0xd3, 0x63, 0x1a, 0x7a

Here is what I executed (without modifications):
`
$ rtl_433 -R 169 -y "{52} 0x01f22ed3631a7a {0}" 
rtl_433 version 22.11 (2022-11-19) inputs file rtl_tcp RTL-SDR SoapySDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/malek/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 1 out of 223 device decoding protocols [ 169 ]
`
Here is what I changed to make it "work":
`$ rtl_433 -R 169 -y "{52} 0xe1f22ed3631a7a {0}"
rtl_433 version 22.11 (2022-11-19) inputs file rtl_tcp RTL-SDR SoapySDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/malek/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 1 out of 223 device decoding protocols [ 169 ]
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2023-09-19 14:52:42                                                                                                                                        
model     : Nice-FlorS   Button ID : 1             Serial (enc.): 0c9ce58    Code (enc.): dd12         count     : 16
`

Jev1337, 2023-09-26:
> Also the Flipper Zero decoding code is in #2238 just copy that out.

Thank you, after a week, I successfully paired my transceiver using the code in the issue #2238, both the C/C++ Code and the Python code. I now understand how rtl_433 didn't work as expected for me, which was because the values were encoded (can be decoded using a byte array).