merbanan/rtl_433 #2629: more robust MQTT connection

atrejut, 2023-09-20:
Hi,

sometimes I see
`Sep 18 08:14:36 XXX rtl_433[566]: [MQTT] MQTT Connection failed...`
and no further output, but it seems the process keeps running...

It would be awesome if it could be more robust, and just re-try the connection periodically? Alternatively having the process die (so systemctl can restart it and/or notify me) would also be ok.

I start the process as `/usr/local/bin/rtl_433 -R 121 -F kv -F mqtt://apollo retain 0 devices sensors/garten[/model][/channel]` so if there's some command line option that might be helpful and that I'm missing I'd love to hear about it :)

Thanks for all the good work!

zuckschwerdt, 2023-09-20:
It's just a warning and safe to ignore. Raise the log level (`-v`) to see the reconnect notice.
```
MQTT: MQTT Connection failed...
MQTT: MQTT Connected...
MQTT: MQTT Connection established.
```

closed 2023-09-20 by zuckschwerdt

atrejut, 2023-09-20:
@zuckschwerdt Thanks for the quick response. Unfortunately I didn't receive any messages after that line occured (I persist everything to a DB, so easy to check), so something must have gone wrong with the reconnect.
I'll raise the log-level and check if the issue re-occurs.

gdt, 2023-09-20:
I have not dug in to this -- I don't use the builtin mqtt -- but as a general principle I would say that connection established should be logged at the same level as connection failed, so that any given log show both or neither.

(I have found that the rtl_433_mqtt_relay ends up not connected, but that's a separate issue and is on my list to fix.)

commit be332c64 (2023-09-20): minor: Improve MQTT warning message (closes #2629)