merbanan/rtl_433 #2630: Unnamed command 12 in Chuango DWC-102 door sensor

LuccoJ, 2023-09-20:
I don't know if one report is enough to act on this, but unlike my previous Chuango door/window sensors (which likely had another model number but I cannot retrieve it), this DWC-102 additionally generates a cmd=12 signal, which gets transcribed as "?", but I can confirm that in the case of my sensor, it represents the door/windows being *closed*, something that the previous models didn't detect (in fact, my alarm system does nothing with it, but I presume newer Chuango systems use it to warn people they've left the door/window open when they arm the alarm, like other brands do).

So, to be clear, what I get what I open and close three times my door, with its sensor set to home mode, I get

```
{"time" : "2023-09-20 18:27:48", "model" : "Chuango-Security", "id" : [redacted], "cmd" : "Home Mode Zone", "cmd_id" : 6}
{"time" : "2023-09-20 18:27:48", "model" : "Chuango-Security", "id" : [redacted], "cmd" : "Home Mode Zone", "cmd_id" : 6}
{"time" : "2023-09-20 18:27:48", "model" : "Chuango-Security", "id" : [redacted], "cmd" : "Home Mode Zone", "cmd_id" : 6}
{"time" : "2023-09-20 18:27:48", "model" : "Chuango-Security", "id" : [redacted], "cmd" : "Home Mode Zone", "cmd_id" : 6}
{"time" : "2023-09-20 18:27:48", "model" : "Chuango-Security", "id" : [redacted], "cmd" : "Home Mode Zone", "cmd_id" : 6}
{"time" : "2023-09-20 18:27:48", "model" : "Chuango-Security", "id" : [redacted], "cmd" : "Home Mode Zone", "cmd_id" : 6}
{"time" : "2023-09-20 18:27:48", "model" : "Chuango-Security", "id" : [redacted], "cmd" : "Home Mode Zone", "cmd_id" : 6}
{"time" : "2023-09-20 18:27:48", "model" : "Chuango-Security", "id" : [redacted], "cmd" : "Home Mode Zone", "cmd_id" : 6}
{"time" : "2023-09-20 18:27:49", "model" : "Chuango-Security", "id" : [redacted], "cmd" : "Home Mode Zone", "cmd_id" : 6}
{"time" : "2023-09-20 18:27:49", "model" : "Chuango-Security", "id" : [redacted], "cmd" : "Home Mode Zone", "cmd_id" : 6}
{"time" : "2023-09-20 18:27:53", "model" : "Chuango-Security", "id" : [redacted], "cmd" : "Home Mode Zone", "cmd_id" : 6}
{"time" : "2023-09-20 18:27:53", "model" : "Chuango-Security", "id" : [redacted], "cmd" : "Home Mode Zone", "cmd_id" : 6}
{"time" : "2023-09-20 18:27:53", "model" : "Chuango-Security", "id" : [redacted], "cmd" : "Home Mode Zone", "cmd_id" : 6}
{"time" : "2023-09-20 18:27:53", "model" : "Chuango-Security", "id" : [redacted], "cmd" : "Home Mode Zone", "cmd_id" : 6}
{"time" : "2023-09-20 18:27:53", "model" : "Chuango-Security", "id" : [redacted], "cmd" : "Home Mode Zone", "cmd_id" : 6}
{"time" : "2023-09-20 18:27:53", "model" : "Chuango-Security", "id" : [redacted], "cmd" : "Home Mode Zone", "cmd_id" : 6}
{"time" : "2023-09-20 18:27:53", "model" : "Chuango-Security", "id" : [redacted], "cmd" : "Home Mode Zone", "cmd_id" : 6}
{"time" : "2023-09-20 18:27:53", "model" : "Chuango-Security", "id" : [redacted], "cmd" : "Home Mode Zone", "cmd_id" : 6}
{"time" : "2023-09-20 18:27:54", "model" : "Chuango-Security", "id" : [redacted], "cmd" : "Home Mode Zone", "cmd_id" : 6}
{"time" : "2023-09-20 18:27:54", "model" : "Chuango-Security", "id" : [redacted], "cmd" : "Home Mode Zone", "cmd_id" : 6}
{"time" : "2023-09-20 18:27:54", "model" : "Chuango-Security", "id" : [redacted], "cmd" : "Home Mode Zone", "cmd_id" : 6}
{"time" : "2023-09-20 18:27:59", "model" : "Chuango-Security", "id" : [redacted], "cmd" : "Home Mode Zone", "cmd_id" : 6}
{"time" : "2023-09-20 18:27:59", "model" : "Chuango-Security", "id" : [redacted], "cmd" : "Home Mode Zone", "cmd_id" : 6}
{"time" : "2023-09-20 18:27:59", "model" : "Chuango-Security", "id" : [redacted], "cmd" : "Home Mode Zone", "cmd_id" : 6}
{"time" : "2023-09-20 18:27:59", "model" : "Chuango-Security", "id" : [redacted], "cmd" : "Home Mode Zone", "cmd_id" : 6}
{"time" : "2023-09-20 18:27:59", "model" : "Chuango-Security", "id" : [redacted], "cmd" : "Home Mode Zone", "cmd_id" : 6}
{"time" : "2023-09-20 18:27:59", "model" : "Chuango-Security", "id" : [redacted], "cmd" : "Home Mode Zone", "cmd_id" : 6}
{"time" : "2023-09-20 18:27:59", "model" : "Chuango-Security", "id" : [redacted], "cmd" : "Home Mode Zone", "cmd_id" : 6}
{"time" : "2023-09-20 18:28:00", "model" : "Chuango-Security", "id" : [redacted], "cmd" : "Home Mode Zone", "cmd_id" : 6}
{"time" : "2023-09-20 18:28:00", "model" : "Chuango-Security", "id" : [redacted], "cmd" : "Home Mode Zone", "cmd_id" : 6}
{"time" : "2023-09-20 18:28:00", "model" : "Chuango-Security", "id" : [redacted], "cmd" : "Home Mode Zone", "cmd_id" : 6}
{"time" : "2023-09-20 18:28:00", "model" : "Chuango-Security", "id" : [redacted], "cmd" : "Home Mode Zone", "cmd_id" : 6}
{"time" : "2023-09-20 18:28:02", "model" : "Chuango-Security", "id" : [redacted], "cmd" : "?", "cmd_id" : 12}
{"time" : "2023-09-20 18:28:02", "model" : "Chuango-Security", "id" : [redacted], "cmd" : "?", "cmd_id" : 12}
{"time" : "2023-09-20 18:28:02", "model" : "Chuango-Security", "id" : [redacted], "cmd" : "?", "cmd_id" : 12}
{"time" : "2023-09-20 18:28:02", "model" : "Chuango-Security", "id" : [redacted], "cmd" : "?", "cmd_id" : 12}
{"time" : "2023-09-20 18:28:02", "model" : "Chuango-Security", "id" : [redacted], "cmd" : "?", "cmd_id" : 12}
{"time" : "2023-09-20 18:28:02", "model" : "Chuango-Security", "id" : [redacted], "cmd" : "?", "cmd_id" : 12}
{"time" : "2023-09-20 18:28:02", "model" : "Chuango-Security", "id" : [redacted], "cmd" : "?", "cmd_id" : 12}
{"time" : "2023-09-20 18:28:02", "model" : "Chuango-Security", "id" : [redacted], "cmd" : "?", "cmd_id" : 12}
{"time" : "2023-09-20 18:28:02", "model" : "Chuango-Security", "id" : [redacted], "cmd" : "?", "cmd_id" : 12}
{"time" : "2023-09-20 18:28:02", "model" : "Chuango-Security", "id" : [redacted], "cmd" : "?", "cmd_id" : 12}
{"time" : "2023-09-20 18:28:02", "model" : "Chuango-Security", "id" : [redacted], "cmd" : "?", "cmd_id" : 12}
{"time" : "2023-09-20 18:28:05", "model" : "Chuango-Security", "id" : [redacted], "cmd" : "Home Mode Zone", "cmd_id" : 6}
{"time" : "2023-09-20 18:28:05", "model" : "Chuango-Security", "id" : [redacted], "cmd" : "Home Mode Zone", "cmd_id" : 6}
{"time" : "2023-09-20 18:28:05", "model" : "Chuango-Security", "id" : [redacted], "cmd" : "Home Mode Zone", "cmd_id" : 6}
{"time" : "2023-09-20 18:28:05", "model" : "Chuango-Security", "id" : [redacted], "cmd" : "Home Mode Zone", "cmd_id" : 6}
{"time" : "2023-09-20 18:28:05", "model" : "Chuango-Security", "id" : [redacted], "cmd" : "Home Mode Zone", "cmd_id" : 6}
{"time" : "2023-09-20 18:28:05", "model" : "Chuango-Security", "id" : [redacted], "cmd" : "Home Mode Zone", "cmd_id" : 6}
{"time" : "2023-09-20 18:28:05", "model" : "Chuango-Security", "id" : [redacted], "cmd" : "Home Mode Zone", "cmd_id" : 6}
{"time" : "2023-09-20 18:28:05", "model" : "Chuango-Security", "id" : [redacted], "cmd" : "Home Mode Zone", "cmd_id" : 6}
{"time" : "2023-09-20 18:28:05", "model" : "Chuango-Security", "id" : [redacted], "cmd" : "Home Mode Zone", "cmd_id" : 6}
{"time" : "2023-09-20 18:28:05", "model" : "Chuango-Security", "id" : [redacted], "cmd" : "Home Mode Zone", "cmd_id" : 6}
{"time" : "2023-09-20 18:28:05", "model" : "Chuango-Security", "id" : [redacted], "cmd" : "Home Mode Zone", "cmd_id" : 6}
{"time" : "2023-09-20 18:28:07", "model" : "Chuango-Security", "id" : [redacted], "cmd" : "?", "cmd_id" : 12}
{"time" : "2023-09-20 18:28:07", "model" : "Chuango-Security", "id" : [redacted], "cmd" : "?", "cmd_id" : 12}
```

Some notes:
* Even if I change the sensor to normal (not home) mode, the cmd for closing still appears to be cmd=12.
* While multiple repetitions of the command are typical of these sensors, it would appear that the exact number of repetitions depends on how quickly I open or, especially, close the door, with it being closed very quickly sometimes resulting in *no* command being sent.
* Sometimes, the last cmd=12 is followed by ```"Smoke-GS558", "id" : [redacted], "unit" : 2, "learn" : 0, "code" : "3f1882"}``` which I do not believe to be a real sensor (certainly not mine, although I can't exclude a neighbor, but I've never seen it before making this test)
* Personally I would interpret these as quirks of this sensor, rather than an indication that it's not reporting "closing" but something else.

I would in fact propose "Closing" as the name for cmd=12, unless you want more confirmations that this is the case with other Chuango sensors.

zuckschwerdt, 2023-09-20:
Thanks for the report, the addition sounds good.

closed 2023-09-20 by zuckschwerdt

LuccoJ, 2023-09-20:
Thanks. As expected,

```
{u'model': u'Chuango-Security', u'cmd': u'Closing', u'id': 266639, u'cmd_id': 12, u'time': u'2023-09-20 19:02:06'}
```

(slightly different format merely because it's parsed from Python)

commit 76423bcc (2023-09-20): Add support for newer Chuango DWC-102 close command (closes #2630)