merbanan/rtl_433 #2631: Enable also Protocol 200 # Auriol temperature/rain sensor

MatejsLab, 2023-09-21:
I understand I can use `rtl_433 -R 200` to enable the * disabled protocol, but I would also like to listen in to other temperature sensors.

I need protocol 200 for the Auriol 4-LD5661 (actually LD5972) rain sensor, and it seems to work fine with `rtl_433 -R 200` in short-term testing.

How can I set the conf file properly as the [provided conf file](https://github.com/merbanan/rtl_433/blob/master/conf/rtl_433.example.conf) with removed # at protocol 200 seems to crash rtl_433

I put the file into `/home/m/.config/rtl_433/rtl_433.conf`

Is the an issue with my config, why protocol 200 is being disabled by default, and how to enable it properly?

In essence, I would like to enable all standard protocols and protocol 200, 

Thanks in advance
Matej

MatejsLab, 2023-09-21:
What is that bussines with so much Acurite stations found, they seem fake or **is there one that is changing id** all the time.

I could not belive that all my neighbours have Acurrite rain sensors, i have Auriol that luckily reads fine.

[image]

zuckschwerdt, 2023-09-21:
We have no reports of false positive with Acurite. Watch them for a longer time to see if the rain_mm looks good and distinct between all.

IIRC, the Fitipower FC0012 tuner you use has problems with gain and thus noise. Use a conf with `pulse_detect autolevel` (default) or use `-Y autolevel` on the command line.

MatejsLab, 2023-09-24:
I will have to try this **autolevel** feature, those Acurite stations that I don't have, seem to be bogus, there are over 20 of them on different IDs.

What is the best way to install RTL_433 on the Peppermint 10 Linux that does not have RTL in its repo?
I have openhab there on a physical machine that hopefully will be more reliable than VM.

I don't feel like building from a source. Is there a link I can get the compiled version and install manually?

gdt, 2023-09-26:
This looks like a support request, not a bug, and I think it has run its course and the issue can be closed.  If not, please explain the bug in rtl_433 :-)

MatejsLab, 2023-09-26:
The rain sensor on protocol 200 now runs fine, and I managed to get things working with MQTT, seems reliable for now.

Initially, there were issues with the default config file that I solved thanks to @zuckschwerdt.
Then with compiling, when solved it was not a bug anymore, I agree.

The only issue I have with reading my sensors, they change IDs occasionally, not sure how to map sensors in my software when they change ID, there are more sensors than channels, also neighbors have them.
Good sensors change ID only when I change battery, bad ones jump between a few IDs or never return to the same ID and change every few hours.

Where should I discuss those things, is there a forum or does it fit in some other category on this GitHub page? 
As soon as I get pointed to the solution or discussion place, I agree to close the issue.

gdt, 2023-09-26:
github discussions or the malinglist (google group) as described in the readme.b

closed 2023-09-26 by gdt