merbanan/rtl_433 #2634: Disable MQTT retain on spesific device

TomW1605, 2023-09-24:
i have a specific custom decoded for my doorbell and with mqtt retain it always triggers the automation when home assistant starts up. is there a way to turn off retain for that device specifically and not the rest?

gdt, 2023-09-24:
This really belongs in discussions or the mailinglist, because as is it is not a bug report.   It also lacks all the necessary background information, other than that you are using rtl_433, HA, and mqtt.   I would like to see us as a group sort things into issues vs discussions more strongly, and use the list for larger-scope discussions.

The real issue is almost certainly that these sort of devices decoded by rtl_433 are a blend of (to use HA terms) binary sensors and events, and those two should be treated differently.   This is compounded by the collision of (sensible) rtl_433 doctrine to produce a json line per reception, and the current lack of a stateful middleware that processes those to the semantics needed by consumers.  Happy to discuss on the mailinglist.

TomW1605, 2023-09-24:
sorry, i keep forgetting github discussions exist (and i didnt even know there was a mailinglist or really how they work). i think there is a way to convert an issue to a discussion but i cant find it (maybe only owners/members can do it) so ill just open a new on there and reference this

gdt, 2023-09-24:
The mailinglist is a google group and described in the README.

gdt, 2023-09-24:
One can convert, but I think it's just as well to start over, since you hadn't actually explained what you were doing precisely enough for anyone to help you.  (which programs, how do you have them configured, etc.)

closed 2023-09-24 by gdt