merbanan/rtl_433 #2637: bug in honeywell_cm921.c

aadvanstegeren, 2023-09-25:
hello,

I'm using honeywell evohome and I receive strange temperatures with 30C9 messages. I only receive distinct temperatures 10.28 , 12.85 , 15.42, 17.99 and 20.56. In hex those temperatures are 0x2828 , 0x3232, 0x3C3C, 0x4646 and 0x5050. It seems that the first and the second byte of the temperature have the same value.

If I look in the code of honeywell_cm921.c at line 238:

233        case 0x30C9: {
234          size_t num_zones = msg->payload_length/3;
235         for (size_t i=0; i < num_zones; i++) {
236            char name[256];
237            snprintf(name, sizeof(name), "temperature (zone %u)", msg->payload[3*i]);
238            int16_t temp = msg->payload[3*i+1] << 8 | msg->payload[3*i+1];
239           data = data_append(data, name, "", DATA_DOUBLE, temp/100.0, NULL);
240          }
241         break;
242        }

The first and the second byte of temp are filled with the same byte. payload[3*i+2] should be used for the second byte :

238            int16_t temp = msg->payload[3*i+1] << 8 | msg->payload[3*i+2];
                                                              
regards,

Aad

zuckschwerdt, 2023-09-25:
Oh, thanks! Very good find! This has must have been a typo in #1336

closed 2023-09-25 by zuckschwerdt

zuckschwerdt, 2023-09-25:
Fixed, thanks again for testing, finding that bug and providing a fix!

commit 07368e73 (2023-09-25): Fix Honeywell-CM921 temperature msg (closes #2637)