merbanan/rtl_433 #2645: Add way to use classic and minmax in parallel

gdt, 2023-09-26:
The basic idea is that if either decodes, you have the data, and there's no reason to have a global choice.

ondrej-zary, 2024-10-05:
ELV KS200/KS300 (also sold as a part of Welltech 40616/TW69) has the same problem - the new default minmax decoder misidentifies the packages as FSK. With -Y classic, the packages are identified as OOK and decoded properly.

zuckschwerdt, 2024-10-05:
Yes, there are ideas to run multiple optimized demods to get better results. This will be a major architecture change and an estimate for this to land is ~half a year.

merbanan, 2026-11-08:
Done: the default FSK detection mode `-Y auto` now runs all FSK detectors (classic, minmax, hysteresis, median) on every package in parallel, see #3623, #3681, #3696 and #3699 (merged in b019df9a and the commits before it). Use `-Y classic` etc. to force a single detector.

closed 2026-11-08 by merbanan