merbanan/rtl_433 #2647: Error: Parameter "parentChannel" needs to be of type "string" but type "undefined" has been passed

Vetinari1977, 2023-09-28:
System:
Raspi 4
Software : IObroker
Node-Js Version: v16.18.0
NPM Version: 8.19.2

Problem:
When new Nexus TH sensors (Bresser tempreture sensors) will be registered, then only the Info and Meta folders will be created and no sensor datas like tempreture and so on are avaylable. In the protokoll of the IOBroker the message from the headline is prompted.  All existing sensors works well. Maybe it has started after an update to a newer Js or NPM version, but not sure because it's long ago after I've inserted a new sensor.

Vetinari1977, 2023-09-28:
Done something wrong by my own / Closed.

closed 2023-09-28 by Vetinari1977

randomname32, 2023-12-06:
@Vetinari1977 I have the same issue. Could you please share what your solution was to fix it?

Vetinari1977, 2023-12-06:
@randomname32 At least my fault has been, that I've postet my report at the wrong repo, but at the correct one I also didn't get an answer. Anyway,  I've created an Blockly script which create an own object and also update those with the datas of the mqtt database. I've uploaded my Blockly and JS as txt files so you can use it to create your own. 

[Blockly.txt](https://github.com/merbanan/rtl_433/files/13582131/Blockly.txt)
[JS.txt](https://github.com/merbanan/rtl_433/files/13582133/JS.txt)