merbanan/rtl_433 #2653: WS90 battery voltage and several others are in mV, while supercap is in V

gdt, 2023-10-01:
It seems obvious that we should regularize quantities into SI units, or at least all battery-like things should be in the same units.
This issue has two steps:
  1) Confirm or fix docs to state norms (I couldn't find this quickly).
  2) Adjust drivers to meet norms

Split out from #2072 where it is off topic.
@jpochmara @sdalu

gdt, 2023-10-01:
Data format says `battery_V (battery_mV)`.   I propose to drop the mV form from the docs, so that docs will say V is the right approach, to solve point 1.

sdalu, 2023-10-01:
The only benefit I see to have `battery_mV` is to avoid dealing with float and rounding errors. Not sure if it is really an issue here.

gdt, 2023-10-01:
In general, values are floats, and there are not rounding errors that matter due to floating point representation.   That argument would lead to temperatures  in milli-degrees as ints ;-) 

The hard question is where to draw the line between "represent what is in the packet" and "use standard units and in general regularize".  However, we enthusiastically convert numbers that are in units of 0.25psi back to psi, so I think unit conversion is on the "be straightforward" side of the line.

jpochmara, 2023-10-03:
The point I was trying to make in the other thread was that we should not change the resolution of data when converting between units.  Changing 3.3V to 3300mV makes a statement about the data that we don't know to be true.  The 3.3V could be 3300mV or it could 3399mV or anything in between.  The original data does not have have resolution down to 0.001, only to 0.1.

gdt, 2023-10-03:
These are simply numbers.  There is absolutely no implied resolution in V, mV, uV, kV or anything else.   So  3300 mV does not make any kind of statement about resolution or accuracy, and is entirely equivalent to 3.3V.   rtl_433 does not have a data type which involves a value and an uncertainty.  If we wanted to do this right, each value would be a tuple containing the value itself and the standard deviation, and people could propagate that.  But this is logging cheap sensors, not a science lab, and the sensors lack the required documentation about their errors -- so I  don't see this as remotely making sense.

I thiink you may be seeing the printed output as honoring what we know about resolution, but I see that as more of a "avoid goofy extra digits that are just rounding noise" and not an attempt to represent precision.

What I have said is that we should steer to only using SI units, and to have everything that is sort of the same thing in a consistent unit.   To me that means V for all voltage measurments, and that doesn't seem to run afoul of what you are worried about, so we don't need to resolve this.

zuckschwerdt, 2023-10-03:
As far as I can see the battery voltages are all in `_mV` (which feels like the right range of scaling) except for Ambientweather WH31 and the WS90 "Supercap" which could reasonable be `_mV` too (but have 0.1 V steps).
The other two instances of `_V` are line voltages (Revolt NC5462, Esic EMT7110).
But @jpochmara makes a good argument about adding digits that are not there.

zuckschwerdt, 2023-10-03:
Maybe take wind speeds as an additional example: Should we convert to `km/h` or `m/s`? There are instances of sensors reporting in either. The scaling factor of 3.6 would produce unnecessary digits. The SI unit is `m/s` but the `km/h` unit would show a finer grained resolution (a below 1 m/s value might still read 1 – 3 km/h). This is very difficult to get right.

gdt, 2023-10-03:
The only two positions that make sense intellectually to me are
  - we are not conveying precision with digits; they are just numbers.  We do avoid printing goofy digits when going from a float to a string of digits for output.
  - we are going to carry value and uncertainty both (and this is way too much work for the gain)

All of  these outputs get procesed by some number of downstream consumers, and we do not expect any of them to store a set of printed digits as a string and analyze precision.  I expect them all to read them and store  a float (double).  If you care about knowing the precision of your device, you have to know it.  Is there an actual example?

An example of why this is impossible is TPMS sensors that have 0.25 psi resolution, e.g. Ford.   2 digits, as we print, over-hints resolution, and 1 is displeasing because it rounds values that otherwise seem sensible.  Nobody thinks these have 0.01 psi resolution.

Also, battery sensors in the real world surely have a huge array of resolutions, and many do not cleanly line up with digits.

gdt, 2023-10-03:
I just remembered we have a unit switch with values native/SI/customary.     I see it as more reasonable to stay in device-native units if native, and less if SI.   However, I don't think we can claim to know that a device is in V vs mV unless there is documentation, as we are inferrring things anyway.

gdt, 2023-10-03:
For wind speed, I don't think "extra digits" should be a concern.  These are just text representations of floats.   If we want to output a tuple with the bits and the value of 1, that's something else.  For example, the position that 1 unit represents 0.1 km/h and represnets (0.1/3.6) m/s are equally valid.

To me the real question is what the proper unit is.  m/s is SI, but one could look up the WMO handbook for reporting and see what their standard units are.   It would not surprise me if it's km/h.   If so, I would be ok with deciding the km/h is the standard version of windspeed, and convert to that, instead of m/s.

But Volts, unless one gets into cgs and statvolts -- which only a few physicists do - are just the way the world is.

jpochmara, 2023-10-03:
I guess I do not see this data as simply numbers, like numbers if math problem in a text book.  All of these numbers started life in an analog sensor, went through a A to D convert, and massaged by firmware.  The sensor and ADC have limits to their precision.   And the only clue we have to that precision is looking at the numbers produce by the devices.

I am all for standardizing the units used, I just think we should not be adding significance digits to the  data during the unit conversion process.  Appropriate rounding/truncating needs to be done, which also mean units need to be picked that this can be done with.  I don't know of anyway to  indicate the the "00" in 3300mV are and "unknown" value.

gdt, 2023-10-03:
Certainly the path to the data is complicated; no disagreement there.    The only clues we have to the precision are 1) knowing the encoding, which is a can't-be-better-than bound and 2) looking at histograms over time to reverse engineer the behavior, which is agreeing with you basically. 

Each device is different.  Just because something is in V and prints from an internal floating-point representation (from a data_make call) as 1.2 doesn't mean at all that it has 1.2V precision.   It could be 0.01V, 0.2V, 0.4V, or 1.2V.   Changing that to mV because that's how (making this up) ISO says batteries should be measured and making it 1200 does not change *the number of significant digits*.   It is merely a change of units.   Information about significant digits -- or more generally some kind of precision and accuracy estimates -- has to be carried as a side channel.   rtl_433 processes all sorts of values, with all sorts of precision (and likely wildly varying accuracy), and they get transformed to some unit and stuck in a C double, and then printf with %.Nf to avoid printing 8 digits when we all know that is silly *for any value that might be present in such a field*.   None of this represents precision to a user, and anybody who sees a value and interprets it as precision is concluding things from fumes.

There's a related issue in Home Assistant, where I ingest temperature data that has a precision of 0.01 degree (by inspection of histogram and maybe even spec, but actually it is 0.01C and in F it is often 0.02).  I see values like 72.52 F.   I also see Home Assistant show 75.6 F, which is jarring as I would like to see the same number of digits always -- but that requires configuring and carrying precision with a sensor.

My real point is that while you can talk about what makes sense in a double inside rtl_433, the next step is printf, and then there are who knows what steps, and that idea that consumers of rtl_433 data can somehow infer precision from seeing a single value and looking at the number of non-zero digits, just does not make sense.   The only way to know precision is to look at a long enough series of data and histogram so that you are confident that you are seeing the device's binning behavior.   That's not affected by multiplying or dividing by 1000 at all.

I think you're assuming too much about the values that come out of printf, and that in the general case, that sort of conclusion is not sound.